maximkulkin/esp-homekit-demo #200: reset wifi config

oremic, 2019-04-11:
hi

void wifi_config_reset() {
    sysparam_set_string("wifi_ssid", "");
    sysparam_set_string("wifi_password", "");
}

I can't find what parameter to reset to delete the saved wifi configuration.
besides passwords and ssid that use this function.
Who can help me thanks

maximkulkin, 2019-04-11:
The way sysparams work is that if you write same key, old values associated with key are replaced. Although if new value is empty (takes zero bytes), it won't be stored at all, thus it will look like the key is not even there.
If you're concerned about security, then it's a different story. The way flash storage works, you can not just overwrite existing data with new one, you need to reformat the whole sector. Thus old data just marked as "dead" (with a single bit reset) and new value is stored in yet unused area. Thus, your passwords might still be there. You could do erase_flash to fully erase all data (or erase particular sector through `esptool.py erase_region <address> <size>`) or programmatically by calling `sysparam_compact()`.

oremic, 2019-04-11:
thanks
it is not a security problem.
The password and ssid are written to "" but the wifi configuration after the software reset, remains saved, so I can overwrite ""

closed 2019-04-27 by maximkulkin