maximkulkin/esp-homekit-demo #203: Reverse Behavior

BillionSurf, 2019-04-18:
I used the example LED. However, the board LED lights up when I activate the LED in HomeKit and not the external one. I suspect that I need to change to gpio_write(led_gpio, on ? 1 : 0), but I don’t know how to with vagrant.

maximkulkin, 2019-04-18:
There are two ways to connect an LED: 1) pin, resistor, LED, ground; 2) power, resistor, LED, pin. In first case you need to set pin high to light up LED, in second - low. Usually pins can sink more current than they can source, so second option is used to get LED brighter.

Not sure I understand the problem with Vagrant.

maximkulkin, 2019-04-18:
Ok, maybe I missread the wuestion. If what you were saying is you want to blink with a different LED, you need to change led_gpio to pin number your other LED is connected to.

BillionSurf, 2019-04-19:
It's working.  My mistake was that I wired the LED like this:

Pin, resistor, led, ground.

maximkulkin, 2019-04-19:
Make sure you preserve polarity on LED, shorter leg goes to ground.

closed 2019-04-19 by maximkulkin