maximkulkin/esp-homekit-demo #205: button

AchimPieters, 2019-04-26:
Hi Maxim,

I have I little problem with your button demo.

My hardware is build like this:
[image]

unfortunately I can't get it to work properly, because when I push the button I says long press or sometimes double press.

I tried changing the button.c part:

```
// times in milliseconds
    button->debounce_time = 50;
    button->long_press_time = 1000;
    button->double_press_time = 500;
```

to other milliseconds, but not with the right results. can you help me?

maximkulkin, 2019-04-26:
I will look into it. When I wrote it more than a year ago it worked perfectly for me, but I've been getting reports about long press trigger before. Some people even stripped long press functionality from some button library copies (e.g. in sonoff_basic example). I'm certainly interested in keeping full functionality, so I will look into fixing it and extracting as a separate library (to finally stop it being copy-pasted everywhere).

maximkulkin, 2019-04-27:
Ok, I have figured that out. You're using button in "active low" configuration (meaning signal goes low when button is pressed). Original button example was written for "active high" and thus everything is inverted in your case and you get long presses all the time.

You can fix it by changing [this line](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/button/button.c#L48) to
```c
if (gpio_read(button->gpio_num) == 0) {
```

I have extracted button code into a separate library which allows configuring which active mode your button uses: [esp-button](https://github.com/maximkulkin/esp-button).

PS you do not need pullup resistors on buttons (shown on your wiring diagram), it will use built-in pullup resistors.

AchimPieters, 2019-04-27:
Awesome :bangbang: you are the 👍 

I have a look into it tomorrow!

closed 2019-04-28 by maximkulkin