maximkulkin/esp-homekit-demo #209: Adding more than 2 lights

Yousefff1, 2019-04-29:
Hi Maxim,

I have purchased the tutorial, and I was wondering if there is an example for adding more than 2 relays using nodemcu to homekit.

I am planning to add more than 2 and under 10.

Thank you,

Yousefff1, 2019-04-29:
I've paid a programmer to code this, but it doesn't work

/*
 * Example of using Sonoff Dual R2 for controlling floor lamp
 * with two individual lights (top and middle one). Also it
 * utilizes existing floor lamp rotary switch to allow switching
 * lights on and off. Existing switch has two output contacts and
 * 4 positions which alternate between contacts as in following table:
 *
 *    position     output1    output2
 *       0           off        off
 *       1           on         off
 *       2           off        on
 *       3           on         on
 *
 * Output 1 is connected to button1 pin on the board so when you
 * rotate switch the signal on button1 alternates every time. Change
 * of signal is used as trigger to change light bulb state.
 *
 * Lamp identify procedure blinks with bottom light 3 times.
 */
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <espressif/esp_common.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
// #include <wifi_config.h>

#include "toggle.h"
#include "wifi.h"

static void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
    sdk_wifi_station_connect();
}

// The GPIO pin that is connected to the relay on the Sonoff Dual R2
const int relay0_gpio = 12;
const int relay1_gpio = 5;

//  Extra Relays
const int relay2 = 16;      //   D0 on NodeMCU
const int relay3 = 4;       //   D2 on NodeMCU
const int relay4 = 0;       //   D3 on NodeMCU
const int relay5 = 2;       //   D4 on NodeMCU
const int relay6 = 14;      //   D5 on NodeMCU
const int relay7 = 15;      //   D8 on NodeMCU
const int relay8 = 3;       //   Rx on NodeMCU
const int relay9 = 1;       //   Tx on NodeMCU
const int relay10 = 10;     //   SD2 on NodeMCU

// The GPIO pin that is connected to the LED on the Sonoff Dual R2
const int led_gpio = 13;
// The GPIO pin that is oconnected to the button on the Sonoff Dual R2
const int button_gpio = 9;

void relay_write(int relay, bool on) {
    gpio_write(relay, on ? 1 : 0);
[... 401 lines cut ...]

    int name_len = snprintf(NULL, 0, "Dual Lamp-%02X%02X%02X",
                            macaddr[3], macaddr[4], macaddr[5]);
    char *name_value = malloc(name_len+1);
    snprintf(name_value, name_len+1, "Dual Lamp-%02X%02X%02X",
             macaddr[3], macaddr[4], macaddr[5]);

    name.value = HOMEKIT_STRING(name_value);
}

void user_init(void) {
    uart_set_baud(0, 115200);

    create_accessory_name();

    gpio_init();

    // wifi_config_init("dual lamp", NULL, on_wifi_ready);
    wifi_init();
    on_wifi_ready();

    if (toggle_create(button_gpio, toggle_callback)) {
        printf("Failed to initialize button\n");
    }
}

maximkulkin, 2019-04-29:
> I've paid a programmer to code this, but it doesn't work
I kind of find it hilarious.

So, what exactly does not work?

Yousefff1, 2019-04-29:
haha yeah, I am helping him with homekit now hahaha

this code doesn't even compile!

Would you mind checking it or do you have a solution to adding more than two lights (relays) ?

I am using a nodemcu

I am willing to tip you to make it worth your time

Yousefff1, 2019-04-29:
Using your own code, I was able to add 3 relays, but the state is wrong, and they wont turn on or off, here is that code 👍 

`
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <espressif/esp_common.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
// #include <wifi_config.h>

#include "toggle.h"
#include "wifi.h"

static void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
    sdk_wifi_station_connect();
}

// The GPIO pin that is connected to the relay on the Sonoff Dual R2
const int relay0_gpio = 12;
const int relay1_gpio = 5;
const int relay2_gpio = 14;
// The GPIO pin that is connected to the LED on the Sonoff Dual R2
const int led_gpio = 13;
// The GPIO pin that is oconnected to the button on the Sonoff Dual R2
const int button_gpio = 9;

void relay_write(int relay, bool on) {
    gpio_write(relay, on ? 1 : 0);
}

void led_write(bool on) {
    gpio_write(led_gpio, on ? 0 : 1);
}

void reset_configuration_task() {
    //Flash the LED first before we start the reset
    for (int i=0; i<3; i++) {
        led_write(true);
        vTaskDelay(100 / portTICK_PERIOD_MS);
        led_write(false);
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }

    // printf("Resetting Wifi Config\n");

    // wifi_config_reset();

    vTaskDelay(1000 / portTICK_PERIOD_MS);

    printf("Resetting HomeKit Config\n");

    homekit_server_reset();

    vTaskDelay(1000 / portTICK_PERIOD_MS);

    printf("Restarting\n");

[... 177 lines cut ...]
    int name_len = snprintf(NULL, 0, "Triple Lamp-%02X%02X%02X",
                            macaddr[3], macaddr[4], macaddr[5]);
    char *name_value = malloc(name_len+1);
    snprintf(name_value, name_len+1, "Triple Lamp-%02X%02X%02X",
             macaddr[3], macaddr[4], macaddr[5]);

    name.value = HOMEKIT_STRING(name_value);
}

void user_init(void) {
    uart_set_baud(0, 115200);

    create_accessory_name();

    gpio_init();

    // wifi_config_init("dual lamp", NULL, on_wifi_ready);
    wifi_init();
    on_wifi_ready();

    if (toggle_create(button_gpio, toggle_callback)) {
        printf("Failed to initialize button\n");
    }
}
`

outsiderz, 2019-04-29:
I did on 8 relays and everything works fine

Yousefff1, 2019-04-29:
@outsiderz would you mind sharing your code please ?

Yousefff1, 2019-04-30:
@outsiderz please ? : )

outsiderz, 2019-04-30:
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"

static void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
    sdk_wifi_station_connect();
}

const int relay1 = 2;
const int relay2 = 4;
const int relay3 = 5;
const int relay4 = 14;
const int relay5 = 1;
const int relay6 = 13;
const int relay7 = 12;
const int relay8 = 3;
bool relay1_on = false;
bool relay2_on = false;
bool relay3_on = false;
bool relay4_on = false;
bool relay5_on = false;
bool relay6_on = false;
bool relay7_on = false;
bool relay8_on = false;

void led_write(bool on) {
    gpio_write(relay1, on ? 0 : 1);
}

void led_write2(bool on) {
    gpio_write(relay2, on ? 0 : 1);
}

void led_write3(bool on) {
    gpio_write(relay3, on ? 0 : 1);
}

void led_write4(bool on) {
    gpio_write(relay4, on ? 0 : 1);
}

void led_write5(bool on) {
    gpio_write(relay5, on ? 0 : 1);
}

void led_write6(bool on) {
    gpio_write(relay6, on ? 0 : 1);
}

void led_write7(bool on) {
    gpio_write(relay7, on ? 0 : 1);
}

void led_write8(bool on) {
    gpio_write(relay8, on ? 0 : 1);
[... 230 lines cut ...]
  HOMEKIT_CHARACTERISTIC(NAME, "Relay8"),
  HOMEKIT_CHARACTERISTIC(
  ON, false,
  .getter=relay8_on_get,
  .setter=relay8_on_set
  ),
  NULL
  }),
  NULL
  }),
  NULL
  };

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void user_init(void) {
    uart_set_baud(0, 115200);

    wifi_init();
    led_init();
    homekit_server_init(&config);
}

outsiderz, 2019-04-30:
`#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"

static void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
    sdk_wifi_station_connect();
}

const int relay1 = 2;
const int relay2 = 4;
const int relay3 = 5;
const int relay4 = 14;
const int relay5 = 1;
const int relay6 = 13;
const int relay7 = 12;
const int relay8 = 3;
bool relay1_on = false;
bool relay2_on = false;
bool relay3_on = false;
bool relay4_on = false;
bool relay5_on = false;
bool relay6_on = false;
bool relay7_on = false;
bool relay8_on = false;

void led_write(bool on) {
    gpio_write(relay1, on ? 0 : 1);
}

void led_write2(bool on) {
    gpio_write(relay2, on ? 0 : 1);
}

void led_write3(bool on) {
    gpio_write(relay3, on ? 0 : 1);
}

void led_write4(bool on) {
    gpio_write(relay4, on ? 0 : 1);
}

void led_write5(bool on) {
    gpio_write(relay5, on ? 0 : 1);
}

void led_write6(bool on) {
    gpio_write(relay6, on ? 0 : 1);
}

void led_write7(bool on) {
    gpio_write(relay7, on ? 0 : 1);
}

void led_write8(bool on) {
    gpio_write(relay8, on ? 0 : 1);
[... 231 lines cut ...]
  HOMEKIT_CHARACTERISTIC(
  ON, false,
  .getter=relay8_on_get,
  .setter=relay8_on_set
  ),
  NULL
  }),
  NULL
  }),
  NULL
  };

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void user_init(void) {
    uart_set_baud(0, 115200);

    wifi_init();
    led_init();
    homekit_server_init(&config);
}
`

outsiderz, 2019-04-30:
`#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"

static void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
    sdk_wifi_station_connect();
}

const int relay1 = 2;
const int relay2 = 4;
const int relay3 = 5;
const int relay4 = 14;
const int relay5 = 1;
const int relay6 = 13;
const int relay7 = 12;
const int relay8 = 3;
bool relay1_on = false;
bool relay2_on = false;
bool relay3_on = false;
bool relay4_on = false;
bool relay5_on = false;
bool relay6_on = false;
bool relay7_on = false;
bool relay8_on = false;

void led_write(bool on) {
    gpio_write(relay1, on ? 0 : 1);
}

void led_write2(bool on) {
    gpio_write(relay2, on ? 0 : 1);
}

void led_write3(bool on) {
    gpio_write(relay3, on ? 0 : 1);
}

void led_write4(bool on) {
    gpio_write(relay4, on ? 0 : 1);
}

void led_write5(bool on) {
    gpio_write(relay5, on ? 0 : 1);
}

void led_write6(bool on) {
    gpio_write(relay6, on ? 0 : 1);
}

void led_write7(bool on) {
    gpio_write(relay7, on ? 0 : 1);
}

void led_write8(bool on) {
    gpio_write(relay8, on ? 0 : 1);
[... 230 lines cut ...]
  HOMEKIT_CHARACTERISTIC(NAME, "Relay8"),
  HOMEKIT_CHARACTERISTIC(
  ON, false,
  .getter=relay8_on_get,
  .setter=relay8_on_set
  ),
  NULL
  }),
  NULL
  }),
  NULL
  };

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void user_init(void) {
    uart_set_baud(0, 115200);

    wifi_init();
    led_init();
    homekit_server_init(&config);
}

outsiderz, 2019-04-30:
```
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"

static void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
    sdk_wifi_station_connect();
}

const int relay1 = 2;
const int relay2 = 4;
const int relay3 = 5;
const int relay4 = 14;
const int relay5 = 1;
const int relay6 = 13;
const int relay7 = 12;
const int relay8 = 3;
bool relay1_on = false;
bool relay2_on = false;
bool relay3_on = false;
bool relay4_on = false;
bool relay5_on = false;
bool relay6_on = false;
bool relay7_on = false;
bool relay8_on = false;

void led_write(bool on) {
    gpio_write(relay1, on ? 0 : 1);
}

void led_write2(bool on) {
    gpio_write(relay2, on ? 0 : 1);
}

void led_write3(bool on) {
    gpio_write(relay3, on ? 0 : 1);
}

void led_write4(bool on) {
    gpio_write(relay4, on ? 0 : 1);
}

void led_write5(bool on) {
    gpio_write(relay5, on ? 0 : 1);
}

void led_write6(bool on) {
    gpio_write(relay6, on ? 0 : 1);
}

void led_write7(bool on) {
    gpio_write(relay7, on ? 0 : 1);
}

void led_write8(bool on) {
[... 233 lines cut ...]
  ON, false,
  .getter=relay8_on_get,
  .setter=relay8_on_set
  ),
  NULL
  }),
  NULL
  }),
  NULL
  };

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void user_init(void) {
    uart_set_baud(0, 115200);

    wifi_init();
    led_init();
    homekit_server_init(&config);
}

```

Yousefff1, 2019-04-30:
@outsiderz You are amazing man!!!
so two things,
I wanna tip you if you are okay with that.
And can I use this code for commercial use in the future ?

Yousefff1, 2019-04-30:
@outsiderz do you also have buttons to control the relay along with the homekit app ?
if not, is it possible ?

maximkulkin, 2019-04-30:
I have added example of a slightly nicer way to add a lot of services - [dynamic_services](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/dynamic_services/main.c). Keep in mind: if you decide to change services in runtime, you need to increase accessory->config_number or your changes won't be noticed by controllers.

Yousefff1, 2019-04-30:
@maximkulkin wow! this is super nice! and works like a charm too!

one question if you may, I am trying to make this the brains for my 8 channel relay to control lights, so I need to incorporate buttons (momentary wall switches) too. Just like the sonoff basic, if you press on the onboard button, the relay will switch, and the homekit will recognize that the button has been clicked.

How can I incorporate that into this code, pretty please :)

maximkulkin, 2019-04-30:
You're not gonna have enough pins to support 8 relays + 8 buttons. You might want to use either a [shift register](https://www.sparkfun.com/products/13699) to control relays or a [GPIO extender](https://www.sparkfun.com/products/13601) or some other smart trickery.

Yousefff1, 2019-04-30:
Thank you for the reply! I really appreciate it.
Let's say for this example ( 4 relays with 4 buttons )
How can we make it happen?

maximkulkin, 2019-04-30:
It's super easy with my just released [esp-button](https://github.com/maximkulkin/esp-button) library. Just create a button as you create relay services and pass ON characteristic as button context:
```c
typedef struct {
    uint8_t relay_gpio;
    uint8_t button_gpio;
    homekit_characteristic_t* ch_on;
} relay_info_t;

relay_info_t relays[] = {
    { 12, 14 },
    { 5, 13 },
};

void button_callback(button_event_t event, void* context) {
    relay_info_t* info = context;
    homekit_characteristic_t *ch = info->ch_on;
    ch->value.bool_value = !ch->value.bool_value;
    homekit_characteristic_notify(ch, ch->value);
    relay_callback(ch, ch->value, info);
}

void relay_callback(homekit_characteristic_t *ch, homekit_value_t value, void* context) {
    relay_info_t* info = context;
    relay_write(info->relay_gpio, value.bool_value);
}

for (int i=0; i < relay_count; i++) {
    int relay_name_len = snprintf(NULL, 0, "Relay %d", i + 1);
    char *relay_name_value = malloc(name_len+1);
    snprintf(relay_name_value, relay_name_len+1, "Relay %d", i + 1);

    relays[i].ch_on = NEW_HOMEKIT_CHARACTERISTIC(
        ON, true,
        .callback=NEW_HOMEKIT_CHARACTERISTIC_CALLBACK(relay_callback, .context=(void*)&relays[i]),
    );
    button_create(relays[i].button_gpio, BUTTON_CONFIG(), button_callback, &relays[i]);
    c = characteristics;
    *(c++) = NEW_HOMEKIT_CHARACTERISTIC(NAME, relay_name_value);
    *(c++) = relays[i].ch_on;
    *(c++) = NULL;

    *(s++) = NEW_HOMEKIT_SERVICE(LIGHTBULB, .characteristics=MEMDUP_RANGE(characteristics, c));
}
```

outsiderz, 2019-04-30:
> У вас не будет достаточно контактов для поддержки 8 реле + 8 кнопок. Возможно, вы захотите использовать [сдвиговый регистр](https://www.sparkfun.com/products/13699) для управления реле, [расширитель GPIO](https://www.sparkfun.com/products/13601) или какой-то другой хитрый прием.

Максим а есть возможность сделать так что бы 4 реле можно было разнести в разные комнаты в приложении home?

outsiderz, 2019-04-30:
> @outsiderz You are amazing man!!!
> so two things,
> I wanna tip you if you are okay with that.
> And can I use this code for commercial use in the future ?

write me a telegram @outsiderzEnik

maximkulkin, 2019-04-30:
> Is there a way to put relays in different rooms in Home app

@outsiderz Haven't tried that, it might be that you can not, since the idea of multiple accessories grouped together is that they are physically reside together..
There is a concept of bridges (which supposedly are devices that bridge one technology accessories into HomeKit and thus they might not reside in same place physically), but I haven't explored it.

Yousefff1, 2019-05-04:
@maximkulkin I've tried to place this code everywhere in the original sketch, I keep getting errors,

would you be interested in writing the code I need, and I'll buy you 3 coffees ?

I am trying to do this : 
Connect buttons (4 buttons for example) to the Nodemcu, and control those via homekit too.

The Nodemcu outputs the pressed buttons via serial to an Arduino mega, the Arduino will control relays.

Would making this be a possibility sir ?

Yousefff1, 2019-05-04:
@maximkulkin @outsiderz please ? :):)

maximkulkin, 2019-05-11:
I believe, this can be closed

closed 2019-05-11 by maximkulkin