maximkulkin/esp-homekit-demo #212: can't connect to my wifi

swatu86, 2019-05-05:
I want to make a multi sensor, after i flashed my esp8266 i found a wifi called motion-sensor-xxxxxx, after i connect it pull up screen to choose a network, so i choose my wifi network and tap a pass but after nothing happened. can you help me

maximkulkin, 2019-05-06:
What did you expect to happen then?

swatu86, 2019-05-07:
Hi,

Normally I should be able to pair multi sensor with my home app but after I connect to multi sensor Wifi it shows pop up window where I choose network and tap a password and its stack on that moment only white screen on pop up window, and it’s looks like multi sensor didn’t get a password. Can you help me to fix that issue?

BR
Krzysztof Dobosz

Wysłane z iPhone'a

maximkulkin, 2019-05-07:
Sometimes it happens. This captive portal thing sometimes behave weirdly from iOS side, just try again.
If trying again does not work for you, please send me logs from ESP.

closed 2019-05-12 by maximkulkin