maximkulkin/esp-homekit-demo #214: Zero crossing pin

Yousefff1, 2019-05-16:
I am trying to use sonoff-basic-pwm with an AC dimming module, on the module there are 4 pins, VCC, GND, Zero Crossing and PWM

Using the sketch and flashing it to a nodemcu, I was able to dim the onboard LED, my question is:

Where to connect the Zero Crossing pin from the module to the Nodemcu?

The link for the AC dimming module 
 https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html

Thank you,

Yousefff1, 2019-05-19:
Any luck with this guys ?

Yousefff1, 2019-06-10:
did anybody try this ?

jukorp, 2019-06-10:
You need difference driver. 
Example ne555 timer or mcu ”microchip, atmel...” 
http://sound.whsites.net/project157.htm

Esp8266 it is not possible to drive time critical process”zero cross detect”

This is my project: https://github.com/jukorp?tab=projects
Time critical mcu is pic16lf15313
mcu convert uart data and zero cross to mosfet.

555 timer is simplest

closed 2019-07-09 by maximkulkin