maximkulkin/esp-homekit-demo #215: ESP8266 Arduino Mega

Yousefff1, 2019-05-16:
Hello,

Will this code work with an Arduino Mega that has ESP8266 built-in ?

Here is a link for the board

https://www.aliexpress.com/store/product/Mega-WiFi-R3-ATmega2560-ESP8266-32Mb-memory-USB-TTL-CH340G-Compatible-for-Arduino-Mega-NodeMCU-WeMos/1950989_32806021806.html?spm=2114.12010608.0.0.7ff36377CukE0n

maximkulkin, 2019-05-16:
From [description](https://robotdyn.com/mega-wifi-r3-atmega2560-esp8266-flash-32mb-usb-ttl-ch340g-micro-usb.html) it seems that you can flash ESP8266, but according to [schematic](https://robotdyn.com/pub/media/0G-00005806==MEGA+WiFi-R3-AT2560-ESP8266-32MB-CH340G/DOCS/Schematic==0G-00005806==MEGA+WiFi-R3-AT2560-ESP8266-32MB-CH340G.pdf) almost no GPIO pins are exposed, so it kind of useless.. Unless you flash Arduino with some kind of serial communication protocol that will allow ESP8266 control pins through some serial protocol.

Yousefff1, 2019-05-16:
@maximkulkin I see, so this board only combines the ESP8266 with an Arduino and I have to manually connect the wires between the two.

Thank you for the reply.

maximkulkin, 2019-05-16:
No, the board does allow to connect ESP8266 to Arduino, but it seems that you can not debug Arduino with WiFi (because that switch allows you to connect Arduino either to USB or to ESP8266, but not to both).
[image]
[image]

Yousefff1, 2019-05-16:
@maximkulkin Oh I think I understand, they hardwired the tx rx traces and put a mode selection pin to chose. I thought this could replace the multiplexer easily, but looks like it will need lots of effort to get it to function with your esp-homekit examples, specifically the dynamic relays.

maximkulkin, 2019-05-16:
Yes, seems like it. It's mostly meant to provide wifi communication to Mega chip.

closed 2019-05-16 by maximkulkin