maximkulkin/esp-homekit-demo #216: Example for a PIR sensor ?

Yousefff1, 2019-05-16:
Would it be possible to provide a demo code for a PIR sensor please ?

Yousefff1, 2019-05-19:
Any updates guys ?

maccoylton, 2019-05-20:
I created one which can be found here:- https://github.com/maccoylton/esp-homekit-motion-sensor

Yousefff1, 2019-05-20:
Thank you @maccoylton 
After using vagrant to upload the firmware to the Nodemcu, the nodemcu doesn't show up in my wifi list, and the onboard blue led is blinking.

maccoylton, 2019-05-21:
>Thank you @maccoylton  ...

@Yousefff1 please provide a log in my github repository 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2019-05-21 by maximkulkin

Yousefff1, 2019-05-22:
Hi @maximkulkin,

The solution provided is completely different than your solutions, it provides a prebuilt bin file and ota updates, it kind of complicates it.

I really like your examples as they are all opensource, and allows for easy manual editing.

can this please stay open till we get a similar code ?

maximkulkin, 2019-05-22:
That one is also open source and you can edit everything... If you do not like process of updating over the air, you can flash it by normal means.

Yousefff1, 2019-05-22:
@maximkulkin OTA is embedded within the code, and it will check for the latest bin file, download and install it. This is why I prefer your method. It's completely isolated.

maximkulkin, 2019-05-22:
It shouldn't be hard to strip that away. See no reason to duplicate it here.

This project assumes some skill level and is intended for intermediate level developers/makers. It's not commercial so I do not see reason to bore myself providing every combination of features for every possible accessory.

Yousefff1, 2019-05-23:
I wish I had the skills to do that, believe me @maximkulkin, I spend allot of time trying before contacting you guys, the last thing I want is to bug you.

Yousefff1, 2019-05-23:
I think I have worked my way around this, I have used the door-sensor example, but can the door sensor get pulled high instead of low ?

The PIR sensor I am using outputs 3.3V when it detects motion.

jwktje, 2020-01-24:
@Yousefff1 did you manage to adapt the code from @maccoylton in the end?
I am about to start working on this myself. I agree with @maximkulkin . It should be pretty straight forward to implement strip out the OTA parts and get something that looks more like the other examples. 

If you managed to do this, would you mind sharing the code? Or, if you didn't, do you want me to share my code if I get it to work?

maccoylton, 2020-01-24:
Curious . .why do you want to strip this out?

jwktje, 2020-01-24:
Well to be honest, for me personally it's because I'm not a 100% sure how Life-Cycle-Manager works and if it's ESP32 compatible. 

Obviously I'd prefer to have easy Wifi Provisioning and OTA updates in my setup. But as I'm prototyping it's easier to stick with flashing over USB. 

I found [this documentation ](https://github.com/HomeACcessoryKid/life-cycle-manager/blob/master/homekit%20integration/esp-homekit-instructions.md) about integrating it but it has some info about possibly bricking/rejecting the device if you mess it up. And I'm not sure how Wifi provisioning works here, if it even is supported with this approach.

maccoylton, 2020-01-24:
LCM is targeted at esp8266

LCM includes the ability to configure WIF settings via a terminal  or via your iPhone, iPad etc.

LCM does't stop you  flashing over USB as you develop and test. 

Use of LCM should not in any way increase the risk of briking a device

We have recently been working on a new bootloader that works in conjunction with LCM to  support more options to recover an unresponsive device that utilises LCM https://github.com/HomeACcessoryKid/rboot4lcm. This particularly useful for devices were physical connection is not possible and there are no physical buttons own the device for example a lightbulb.

jwktje, 2020-01-24:
Thanks for clearing that up! Sounds very promising. 
Are there plans for ESP32 compatibility for LCM?

maccoylton, 2020-01-24:
Not aware of any, ESP8266 is cheaper and  built into many commercial devices so has been our focus. If you want further help testing, using or modify my example, feel free to post on that repository

jwktje, 2020-01-24:
Yep I agree and understand. I just happen to have a bunch of ESP32 boards which is why I'm testing with that. But yeah, ESP8266 is more cost effective for sure. 

If I get anything useful working I'll contribute to the repo.

maximkulkin, 2020-01-24:
I have pushed an example of occupancy sensor firmware (based on PIR sensor): https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/occupancy. Check it out

jwktje, 2020-01-24:
That looks awesome. Which version of the ESP-IDF is this compatible with? I'll test it out tonight!

maximkulkin, 2020-01-24:
I only compiled it for esp-open-rtos

jwktje, 2020-01-26:
Tried to compile it with esp-homekit with a couple versions but can’t get it to work. I’ll do some more testing and report back with exact errors if I fail to make it work