maximkulkin/esp-homekit-demo #218: Not compile a project

ShelukhoAlexander, 2019-05-21:
Hello, Maxim. Thanks for your work! 
when clone from git (with "--recursive") i can't compile a project, errors like:
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/23sky/esp/esp-idf/requirements.txt are s                                                                                                                atisfied.
Building partitions from /home/23sky/esp/esp-idf/components/partition_table/part                                                                                                                itions_singleapp.csv...
CC build/driver/spi_master.o
CC build/esp_http_client/esp_http_client.o
CC build/esp_http_server/src/httpd_sess.o
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c: In function 'sp                                                                                                                i_bus_free':
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:331:57: error: i                                                                                                                nvalid operands to binary == (have 'struct <anonymous>' and 'void *')
         SPI_CHECK(atomic_load(&spihost[host]->device[x])==NULL, "not all CSses                                                                                                                 freed", ESP_ERR_INVALID_STATE);
                                                         ^
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:212:11: note: in                                                                                                                 definition of macro 'SPI_CHECK'
     if (!(a)) { \
           ^
In file included from C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_mas                                                                                                                ter.c:145:0:
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c: In function 'sp                                                                                                                i_bus_add_device':
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:384:13: error: i                                                                                                                nvalid initializer
         if (atomic_compare_exchange_strong(&spihost[host]->device[freecs], &nul                                                                                                                l, (spi_device_t *)1)) break;
             ^
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:414:5: error: in                                                                                                                valid initializer
     atomic_store(&spihost[host]->device[freecs], dev);
     ^
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c: In function 'sp                                                                                                                i_bus_remove_device':
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:459:104: error:                                                                                                                 invalid operands to binary != (have 'struct <anonymous>' and 'spi_device_handle_                                                                                                                t {aka struct spi_device_t *}')
     SPI_CHECK(handle->host->cur_cs == NO_CS || atomic_load(&handle->host->devic                                                                                                                e[handle->host->cur_cs])!=handle, "Have unfinished transactions", ESP_ERR_INVALI                                                                                                                D_STATE);
                                                                                                                                                                                                                        ^
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:212:11: note: in                                                                                                                 definition of macro 'SPI_CHECK'
     if (!(a)) { \
           ^
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:472:51: error: i                                                                                                                nvalid operands to binary == (have 'struct <anonymous>' and 'spi_device_handle_t                                                                                                                 {aka struct spi_device_t *}')
         if (atomic_load(&handle->host->device[x]) == handle){
                                                   ^
In file included from C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_mas                                                                                                                ter.c:145:0:
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:473:13: error: i                                                                                                                nvalid initializer
             atomic_store(&handle->host->device[x], NULL);
             ^
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c: In function 'sp                                                                                                                i_setup_device':
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:502:25: error: i                                                                                                                ncompatible types when initializing type 'spi_device_t * {aka struct spi_device_                                                                                                                t *}' using type 'struct <anonymous>'
     spi_device_t *dev = atomic_load(&host->device[dev_id]);
                         ^
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c: In function 'de                                                                                                                vice_release_bus_internal':
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:600:29: error: i                                                                                                                ncompatible types when initializing type 'spi_device_t * {aka struct spi_device_                                                                                                                t *}' using type 'struct <anonymous>'
         spi_device_t* dev = atomic_load(&host->device[i]);
                             ^
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:621:29: error: i                                                                                                                ncompatible types when initializing type 'spi_device_t * {aka struct spi_device_                                                                                                                t *}' using type 'struct <anonymous>'
         spi_device_t *dev = atomic_load(&host->device[i]);
                             ^
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c: In function 'sp                                                                                                                i_post_trans':
C:/msys32/home/23sky/esp/esp-idf/components/driver/spi_master.c:704:25: error: i                                                                                                                ncompatible types when initializing type 'spi_device_t * {aka struct spi_device_                                                                                                                t *}' using type 'struct <anonymous>'
     spi_device_t* dev = atomic_load(&host->device[host->cur_cs]);
                         ^
In file included from C:/msys32/home/23sky/esp/esp-idf/components/freertos/inclu                                                                                                                de/freertos/semphr.h:77:0,
[... cut ...]
CC build/bootloader/soc/src/hal/spi_hal.o
CC build/bootloader/soc/src/hal/spi_hal_iram.o
AR build/bootloader/soc/libsoc.a
make: *** [/home/23sky/esp/esp-idf/components/bootloader/Makefile.projbuild:41:                                                                                                                 /home/23sky/esp/esp-homekit/examples/esp32/switch/build/bootloader/bootloader.bi                                                                                                                n] Ошибка 2

Can you help me?

maximkulkin, 2019-05-23:
Looks like you did not do `--recursive` when you were cloning esp-idf repository. That repository also has a lot of submodules and you need to check them out in order to build projects.

maximkulkin, 2019-05-23:
Also, it seems that your esp-idf repository is rather old... micro-ecc component is no longer in components/ directory. You need to update esp-idf project to latest

ShelukhoAlexander, 2019-05-24:
Thank you very much!

closed 2019-05-24 by ShelukhoAlexander