maximkulkin/esp-homekit-demo #220: Lock Mechanism Accessory

AchimPieters, 2019-05-28:
Hi maxim,

While making you lock mechanism accessory I stumble upon one problem: The Button Part.
Everting works The lock, the LED. unfortunately the button doesn't. You used a different approach for the button part as in your default Button.c files.

I have tried:

active high - signal goes from low to high when button is pressed.
active low - signal connects to ground when button is pressed.

But noting happens, It even doesn't show in the terminal? Do you have a Idea?

maximkulkin, 2019-05-28:
How's your wiring look like?

maximkulkin, 2019-05-28:
Also, why do you put "Lock Mechanism Accessory" in the title while the actual problem is with button?

AchimPieters, 2019-05-28:
Here's one of my wiring examples, I have tried several setup's but unfortunately without succes.

[image]

I just changed the GPIO's:

```
// The GPIO pin that is connected to a relay
const int relay_gpio = 12;
// The GPIO pin that is connected to a LED
// const int led_gpio = 13;
const int led_gpio = 2;
// The GPIO pin that is connected to a button
// const int button_gpio = 0;
const int button_gpio = 14;
```

Title: changed! :thumbsup:

maximkulkin, 2019-05-28:
Ok, I'm not sure about whether you need an external pullup resistor for reset and program buttons (haven't tried it on bare esp-12 module), but for your other button you do not need pullup resistor, esp-button library (I assume you use that one) will do that for you. Although this should not cause issues you're experiencing. What's your code look like?

AchimPieters, 2019-05-28:
Here's My Code:

```
#include <stdio.h>
//#include <espressif/esp_wifi.h>
//#include <espressif/esp_sta.h>>
#include <espressif/esp_common.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>
#include <etstimer.h>
#include <esplibs/libmain.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
//#include "wifi.h"
#include <wifi_config.h>
#include "button.h"

// The GPIO pin that is connected to a relay
const int relay_gpio = 12;
// The GPIO pin that is connected to a LED
// const int led_gpio = 13;
const int led_gpio = 2;
// The GPIO pin that is connected to a button
// const int button_gpio = 0;
const int button_gpio = 14;

// Timeout in seconds to open lock for
const int unlock_period = 5;  // 5 seconds
// Which signal to send to relay to open the lock (0 or 1)
const int relay_open_signal = 1;

void lock_lock();
void lock_unlock();

void relay_write(int value) {
    gpio_write(relay_gpio, value ? 1 : 0);
}

void led_write(bool on) {
    gpio_write(led_gpio, on ? 0 : 1);
}

void reset_configuration_task() {
    //Flash the LED first before we start the reset
    for (int i=0; i<3; i++) {
        led_write(true);
        vTaskDelay(100 / portTICK_PERIOD_MS);
        led_write(false);
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }

    printf("Resetting Wifi Config\n");

    wifi_config_reset();

    vTaskDelay(1000 / portTICK_PERIOD_MS);

    printf("Resetting HomeKit Config\n");

    homekit_server_reset();

    vTaskDelay(1000 / portTICK_PERIOD_MS);

    printf("Restarting\n");

    sdk_system_restart();

[... 174 lines cut ...]
    sdk_wifi_get_macaddr(STATION_IF, macaddr);

    int name_len = snprintf(NULL, 0, "Lock-%02X%02X%02X",
                            macaddr[3], macaddr[4], macaddr[5]);
    char *name_value = malloc(name_len+1);
    snprintf(name_value, name_len+1, "Lock-%02X%02X%02X",
             macaddr[3], macaddr[4], macaddr[5]);

    name.value = HOMEKIT_STRING(name_value);
}

void user_init(void) {
    uart_set_baud(0, 115200);

    create_accessory_name();

    wifi_config_init("lock", NULL, on_wifi_ready);
    gpio_init();
    lock_init();

    if (button_create(button_gpio, 0, 4000, button_callback)) {
        printf("Failed to initialize button\n");
    }
}
```

maximkulkin, 2019-05-28:
You're still using old button library which only supports "active high" configuration.

AchimPieters, 2019-05-29:
Yes that's right. But What I have done is changed the GPIO to the original numbers (eg 2,4 and 12).
Then I also changed the hardware so I matches the code. And voila it works... or not...

When I press the button a few times the ESP crashes and doest cam back See error below. When You open the lock trough Homekit and the press the button it crashed also see error below.

```
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
Toggling relay
>>> HomeKit: [Client 7] Update Characteristics
>>> HomeKit: [Client 7] Get Characteristics
>>> HomeKit: [Client 7] Update Characteristics
Resetting configuration
Resetting Wifi Config
Resetting HomeKit Config
Restarting
rm match
del if0
usl
sul 0 0
!!! HomeKit: [Client 7] Error reading data from socket (code 113). Disconnecting
!!! HomeKit: [Client 6] Error reading data from socket (code 113). Disconnecting
!!! HomeKit: [Client 4] Error reading data from socket (code 113). Disconnecting
!!! HomeKit: [Client 5] Error reading data from socket (code 113). Disconnecting
>>> HomeKit: [Client 7] Closing client connection
>>> HomeKit: [Client 6] Closing client connection
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: [Client 5] Closing client connection

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3fff00a0, prio:14, stack:512
pm_task_hdl : 3ffef990, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4028, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ May 29 2019 16:59:53
phy ver: 273, pp ver: 8.3

>>> wifi_config: Initializing WiFi config
>>> wifi_config: wifi_config_station_connect: No configuration found
>>> wifi_config: Starting AP mode
>>> wifi_config: wifi_config_softap_start: Starting AP SSID=lock-B8D61D
>>> wifi_config: Starting DHCP server
mode : sta(5c:cf:7f:b8:d6:1d) + softAP(5e:cf:7f:b8:d6:1d)
add if0
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
scandone
>>> wifi_config: Starting DNS server
>>> wifi_config: Staring HTTP server
scandone
reconnect
scandone
add 0
aid 2
cnt 

connected with AirPort Network, channel 6
dhcp client start...
ip:192.168.178.26,mask:255.255.255.0,gw:192.168.178.1
scandone
```
 and then it stops when I give a reset:

```
ESP-Open-SDK ver: 0.0.1 compiled @ May 29 2019 16:59:53
phy ver: 273, pp ver: 8.3

>>> wifi_config: Initializing WiFi config
>>> wifi_config: wifi_config_station_connect: No configuration found
>>> wifi_config: Starting AP mode
>>> wifi_config: wifi_config_softap_start: Starting AP SSID=lock-B8D61D
>>> wifi_config: Starting DHCP server
mode : sta(5c:cf:7f:b8:d6:1d) + softAP(5e:cf:7f:b8:d6:1d)
add if0
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
scandone
>>> wifi_config: Starting DNS server
>>> wifi_config: Staring HTTP server
scandone
reconnect
scandone
add 0
aid 4
cnt 

connected with AirPort Network, channel 6
dhcp client start...
ip:192.168.178.26,mask:255.255.255.0,gw:192.168.178.1
scandone
```
 I can only do a clean install?

AchimPieters, 2019-05-30:
So I have tested some more. but first my realise setup:
[image]

As I said above, the setup crashes. When pushing the button as the lock is activated with homekit.
Then I took a look at the other examples there ore more based upon this setting for example `sonoff_basic` The only thing I have to do is change te GPIO's and I can use my existing setup above. But also here it crashes? How can I solve this?

AchimPieters, 2019-05-31:
Next Test:

I changed the button library to the new one. Here's the code:

```
#include <stdio.h>
#include <stdlib.h>

//#include <espressif/esp_wifi.h>
//#include <espressif/esp_sta.h>>
#include <espressif/esp_common.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>
#include <etstimer.h>
#include <esplibs/libmain.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
//#include "wifi.h"
#include <wifi_config.h>
#include "button.h"

const int BUTTON_GPIO = 4;

// The GPIO pin that is connected to a relay
const int relay_gpio = 12;
// The GPIO pin that is connected to a LED
// const int led_gpio = 13;
const int led_gpio = 2;

// Timeout in seconds to open lock for
const int unlock_period = 5;  // 5 seconds
// Which signal to send to relay to open the lock (0 or 1)
const int relay_open_signal = 1;

void lock_lock();
void lock_unlock();

void relay_write(int value) {
    gpio_write(relay_gpio, value ? 1 : 0);
}

void led_write(bool on) {
    gpio_write(led_gpio, on ? 0 : 1);
}

void reset_configuration_task() {
    //Flash the LED first before we start the reset
    for (int i=0; i<3; i++) {
        led_write(true);
        vTaskDelay(100 / portTICK_PERIOD_MS);
        led_write(false);
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }

    printf("Resetting Wifi Config\n");

    wifi_config_reset();

    vTaskDelay(1000 / portTICK_PERIOD_MS);

    printf("Resetting HomeKit Config\n");

    homekit_server_reset();

    vTaskDelay(1000 / portTICK_PERIOD_MS);

    printf("Restarting\n");
[... 306 lines cut ...]
scandone
scandone
scandone
scandone
no buf for probe, ie len 0
scandone
scandone
scandone
scandone
no buf for probe, ie len 0
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
no buf for probe, ie len 0
no buf for probe, ie len 0
scandone
scandone
scandone
```
So now I don't know any other solution? Do you?

AchimPieters, 2019-05-31:
Update:

Okay solved it by changing this line:

```
case button_event_long_press:
            reset_configuration();
            break;
```
to

```
case button_event_long_press:
            lock_lock();
            break;
```

Now it works! but I still have a question: what does the `reset_configuration();` should do? beside reset configuration? Because it does't seems to work?

closed 2019-06-02 by AchimPieters