maximkulkin/esp-homekit-demo #221: Implicit declaration of 'gpio_set_level'

KuhlTime, 2019-05-28:
When I'm compiling the led example for the esp32 I always get the following error. I can only imagine that I have missed installing something.

```
Toolchain path: /Users/kuhltime/.esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/kuhltime/.esp/esp-idf/requirements.txt are satisfied.

App "led" version: 240120f
CC examples/esp32/led/build/main/led.o
/Users/kuhltime/.esp/esp-homekit-demo/examples/esp32/led/main/led.c: In function 'led_write':
/Users/kuhltime/.esp/esp-homekit-demo/examples/esp32/led/main/led.c:62:5: error: implicit declaration of function 'gpio_set_level' [-Werror=implicit-function-declaration]
     gpio_set_level(led_gpio, on ? 1 : 0);
     ^
/Users/kuhltime/.esp/esp-homekit-demo/examples/esp32/led/main/led.c: In function 'led_init':
/Users/kuhltime/.esp/esp-homekit-demo/examples/esp32/led/main/led.c:66:5: error: implicit declaration of function 'gpio_set_direction' [-Werror=implicit-function-declaration]
     gpio_set_direction(led_gpio, GPIO_MODE_OUTPUT);
     ^
/Users/kuhltime/.esp/esp-homekit-demo/examples/esp32/led/main/led.c:66:34: error: 'GPIO_MODE_OUTPUT' undeclared (first use in this function)
     gpio_set_direction(led_gpio, GPIO_MODE_OUTPUT);
                                  ^
/Users/kuhltime/.esp/esp-homekit-demo/examples/esp32/led/main/led.c:66:34: note: each undeclared identifier is reported only once for each function it appears in
In file included from /Users/kuhltime/.esp/esp-homekit-demo/components/common/homekit/include/homekit/homekit.h:4:0,
                 from /Users/kuhltime/.esp/esp-homekit-demo/examples/esp32/led/main/led.c:10:
/Users/kuhltime/.esp/esp-homekit-demo/examples/esp32/led/main/led.c: At top level:
/Users/kuhltime/.esp/esp-homekit-demo/examples/esp32/led/main/led.c:108:40: warning: initialized field overwritten [-Woverride-init]
     HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
                                        ^
/Users/kuhltime/.esp/esp-homekit-demo/components/common/homekit/include/homekit/types.h:236:11: note: in definition of macro 'HOMEKIT_ACCESSORY'
         ##__VA_ARGS__ \
           ^
/Users/kuhltime/.esp/esp-homekit-demo/examples/esp32/led/main/led.c:108:40: note: (near initialization for '(anonymous).category')
     HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
                                        ^
/Users/kuhltime/.esp/esp-homekit-demo/components/common/homekit/include/homekit/types.h:236:11: note: in definition of macro 'HOMEKIT_ACCESSORY'
         ##__VA_ARGS__ \
           ^
cc1: warning: unrecognized command line option '-Wno-frame-address'
cc1: some warnings being treated as errors
make[1]: *** [led.o] Error 1
make: *** [component-main-build] Error 2
make: *** Waiting for unfinished jobs....
```

AchimPieters, 2019-05-30:
Hi @KuhlTime 

If your using a MAC you can use these bash files to install the ESP32 toolchain.

part one:

#!/usr/bin/env bash

#Install Prerequisites
sudo easy_install pip
touch ~/.bash_profile

# Setup Toolchain
mkdir -p ~/esp32
cd ~/esp32

curl -O https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
tar -xzf xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz

#update PATH environment variable
echo "export PATH=$HOME/esp32/xtensa-esp32-elf/bin:$PATH" >> ~/.bash_profile

#Get ESP-IDF
cd ~/esp32
git clone -b v3.2 --recursive https://github.com/espressif/esp-idf.git

#update PATH environment variable
echo "export IDF_PATH=~/esp32/esp-idf" >> ~/.bash_profile
Restart your Mac en the run part two:

#!/usr/bin/env bash

#Install the Required Python Packages
python -m pip install --user -r $IDF_PATH/requirements.txt

KuhlTime, 2019-05-31:
Thanks for your quick replay @AchimPieters. After reinstalling everything it worked fine. I think I might have missed updating my `esp-idf` for a while.

closed 2019-05-31 by KuhlTime