maximkulkin/esp-homekit-demo #222: Example 'dynamic_services' (ESP32!): Exception was unhandled

GermanSheepDog, 2019-05-31:
Hello Maxim,

I ported the example 'dynamic_services' to ESP32 and I changed all ESP8266 procedure for example:
```
    #if WITH_ESP32 == 1
      gpio_set_level(relay, on ? 1 : 0);
    #else
      gpio_write(relay, on ? 1 : 0);
    #endif
```
It worked fine with the last version of homekit. Now I updated homekit to current version 'Add functions & macros to allocate objects at runtime'!

Now I got this error if I connect iPhone with my ESP32:
```
>>> HomeKit: Got new client connection: 55
>>> HomeKit: [Client 55] Pair Setup Step 1/3
>>> HomeKit: [Client 55] Pair Setup Step 2/3
>>> HomeKit: [Client 55] Pair Setup Step 3/3
>>> HomeKit: Added pairing with A6836AB0-B4AF-4F15-9AB4-E3FE1FE823DF
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 55] Successfully paired
>>> HomeKit: [Client 55] Closing client connection
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4011e774  PS      : 0x00060830  A0      : 0x8011e7af  A1      : 0x3ffc96e0  
0x4011e774: homekit_characteristic_remove_notify_callback at /Users/myname/esp/esp-homekit-demo/components/common/homekit/src/accessories.c:477

A2      : 0x3ffc6028  A3      : 0x4011e7c8  A4      : 0x3ffbd640  A5      : 0x3ffbaf00  
0x4011e7c8: client_notify_characteristic at /Users/myname/esp/esp-homekit-demo/components/common/homekit/src/server.c:3115

A6      : 0x3ffc93d0  A7      : 0x00000001  A8      : 0x36304337  A9      : 0x3ffc96c0  
A10     : 0xfb003234  A11     : 0x00000037  A12     : 0x3ffc95f4  A13     : 0x00000000  
A14     : 0x00000002  A15     : 0xff000000  SAR     : 0x00000009  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfb003234  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4011e774:0x3ffc96e0 0x4011e7ac:0x3ffc9700 0x4012093a:0x3ffc9720 0x40120b20:0x3ffc9740 0x40120bfa:0x3ffc9760 0x40121c3a:0x3ffc97a0 0x4008dc21:0x3ffc97c0
0x4011e774: homekit_characteristic_remove_notify_callback at /Users/myname/esp/esp-homekit-demo/components/common/homekit/src/accessories.c:477

0x4011e7ac: homekit_accessories_clear_notify_callbacks at /Users/myname/esp/esp-homekit-demo/components/common/homekit/src/accessories.c:477

0x4012093a: homekit_server_close_client at /Users/myname/esp/esp-homekit-demo/components/common/homekit/src/server.c:3115

0x40120b20: homekit_server_close_clients at /Users/myname/esp/esp-homekit-demo/components/common/homekit/src/server.c:3184

0x40120bfa: homekit_run_server at /Users/myname/esp/esp-homekit-demo/components/common/homekit/src/server.c:3230

0x40121c3a: homekit_server_task at /Users/myname/esp/esp-homekit-demo/components/common/homekit/src/server.c:3391

0x4008dc21: vPortTaskWrapper at /Users/myname/esp/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57
```
Do you have any idea for the reason?

maximkulkin, 2019-05-31:
Did you update your code as well? Can you post part of the code that generates services/characteristics?

GermanSheepDog, 2019-05-31:
Here is the main procedure:
```
void init_accessory() {
    uint8_t macaddr[6];
    #if WITH_ESP32 == 1
       esp_wifi_get_mac(ESP_IF_WIFI_STA, macaddr);
    #else
       sdk_wifi_get_macaddr(STATION_IF, macaddr);
    #endif
    int name_len = snprintf(NULL, 0, "Relays-%02X%02X%02X",
                            macaddr[3], macaddr[4], macaddr[5]);
    char *name_value = malloc(name_len+1);
    snprintf(name_value, name_len+1, "Relays-%02X%02X%02X",
             macaddr[3], macaddr[4], macaddr[5]);

    homekit_service_t* services[MAX_SERVICES + 1];
    homekit_service_t** s = services;

     *(s++) = NEW_HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
         NEW_HOMEKIT_CHARACTERISTIC(NAME, name_value),
         NEW_HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
         NEW_HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0"),
         NEW_HOMEKIT_CHARACTERISTIC(MODEL, "Relays"),
         NEW_HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
         NEW_HOMEKIT_CHARACTERISTIC(IDENTIFY, lamp_identify),
         NULL
     });

    for (int i=0; i < relay_count; i++) {
        int relay_name_len = snprintf(NULL, 0, "Relay %d", i + 1);
        char *relay_name_value = malloc(name_len+1);
        snprintf(relay_name_value, relay_name_len+1, "Relay %d", i + 1);

        *(s++) = NEW_HOMEKIT_SERVICE(LIGHTBULB, .characteristics=(homekit_characteristic_t*[]) {
            NEW_HOMEKIT_CHARACTERISTIC(NAME, relay_name_value),
            NEW_HOMEKIT_CHARACTERISTIC(
                ON, true,
                .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(
                    relay_callback, .context=(void*)&relay_gpios[i]
                ),
            ),
            NULL
        });
    }

    *(s++) = NULL;

    accessories[0] = NEW_HOMEKIT_ACCESSORY(.category=homekit_accessory_category_other, .services=services);
}
```

kiralikbeyin, 2019-06-22:
@maximkulkin  I was trying dynamic example. Please can you check?
 
````

>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
Fatal exception (9):
epc1=0x4025ba4e

0x4025ba4e: homekit_characteristic_has_notify_callback at /Volumes/case-sensitive4/maxim/components/common/homekit/src/accessories.c:581
 
....
....

>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Closing client connection
Fatal exception (9):
epc1=0x4023880e

0x4023880e: homekit_characteristic_remove_notify_callback at /Volumes/case-sensitive4/hk/components/common/homekit/src/accessories.c:477

````

kiralikbeyin, 2019-07-05:
@maximkulkin any news?

maximkulkin, 2019-07-08:
Quick update: I've found the problem and will push fix soon.

kiralikbeyin, 2019-07-08:
@maximkulkin i hope it works with more than 8 accessory.. https://github.com/maximkulkin/esp-homekit-demo/issues/230

GermanSheepDog, 2019-07-11:
At my environment it did not work. I used latest version of esp_homekit:
```
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: BE:E3:E1:2C:CB:9F
>>> HomeKit: Configuring mDNS
>>> HomeKit: Got new client connection: 51
>>> HomeKit: [Client 51] Pair Setup Step 1/3
>>> HomeKit: [Client 51] Pair Setup Step 2/3
>>> HomeKit: [Client 51] Pair Setup Step 3/3
>>> HomeKit: Added pairing with A6836AB0-B4AF-4F15-9AB4-E3FE1FE823DF
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 51] Successfully paired
>>> HomeKit: [Client 51] Closing client connection
>>> HomeKit: Got new client connection: 52
>>> HomeKit: [Client 52] Pair Verify Step 1/2
>>> HomeKit: [Client 52] Pair Verify Step 2/2
>>> HomeKit: [Client 52] Found pairing with A6836AB0-B4AF-4F15-9AB4-E3FE1FE823DF
>>> HomeKit: [Client 52] Verification successful, secure session established
>>> HomeKit: [Client 52] Get Accessories
>>> HomeKit: [Client 52] Update Characteristics
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e8c7e  PS      : 0x00060430  A0      : 0x800e9318  A1      : 0x3ffcb1f0  
0x400e8c7e: homekit_characteristic_add_notify_callback at /Users/myname/esp/M5Stack_20_Dyn_Homekit/components/esp-homekit/src/accessories.c:477

A2      : 0x3ffcca94  A3      : 0x400e8d18  A4      : 0x3ffbd8f4  A5      : 0x3ffbcd38  
0x400e8d18: client_notify_characteristic at /Users/myname/esp/M5Stack_20_Dyn_Homekit/components/esp-homekit/src/server.c:3115

A6      : 0x3ffcaf10  A7      : 0x00000001  A8      : 0x003ffbd1  A9      : 0x3ffbd46d  
A10     : 0x3ffb9ad0  A11     : 0x003ffbd4  A12     : 0x00000000  A13     : 0x3ffbd394  
A14     : 0x3ffbd185  A15     : 0x3ffbd395  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x003ffbd1  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

ELF file SHA256: 522fbf438fe01081ab363a974922143c7bb5f4ab0b0c7024f93a432a90cc8f63

Backtrace: 0x400e8c7e:0x3ffcb1f0 0x400e9315:0x3ffcb210 0x400eabe5:0x3ffcb260 0x400ec1b1:0x3ffcb2a0 0x40125475:0x3ffcb2c0 0x400e9cca:0x3ffcb320 0x400eb003:0x3ffcb350 0x400ebfba:0x3ffcb390 0x4008b26d:0x3ffcb3b0
0x400e8c7e: homekit_characteristic_add_notify_callback at /Users/myname/esp/M5Stack_20_Dyn_Homekit/components/esp-homekit/src/accessories.c:477

0x400e9315: process_characteristics_update at /Users/myname/esp/M5Stack_20_Dyn_Homekit/components/esp-homekit/src/server.c:3115

0x400eabe5: homekit_server_on_update_characteristics at /Users/myname/esp/M5Stack_20_Dyn_Homekit/components/esp-homekit/src/server.c:3115

0x400ec1b1: homekit_server_on_message_complete at /Users/myname/esp/M5Stack_20_Dyn_Homekit/components/esp-homekit/src/server.c:2911

0x40125475: http_parser_execute at /Users/myname/esp/M5Stack_20_Dyn_Homekit/components/http-parser/http-parser/http_parser.c:2335

0x400e9cca: homekit_client_process at /Users/myname/esp/M5Stack_20_Dyn_Homekit/components/esp-homekit/src/server.c:3115

0x400eb003: homekit_run_server at /Users/myname/esp/M5Stack_20_Dyn_Homekit/components/esp-homekit/src/server.c:3223

0x400ebfba: homekit_server_task at /Users/myname/esp/M5Stack_20_Dyn_Homekit/components/esp-homekit/src/server.c:3391

0x4008b26d: vPortTaskWrapper at /Users/myname/esp/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57
```
Here are my complete example:
[M5Stack_20_Dyn_Homekit.zip](https://github.com/maximkulkin/esp-homekit-demo/files/3381512/M5Stack_20_Dyn_Homekit.zip)

kiralikbeyin, 2019-07-11:
Quick solution maybe changing HOMEKIT_SPI_FLASH_BASE_ADDR 

HOMEKIT_SPI_FLASH_BASE_ADDR = 0xb0000 # 720.896
make clean and make erse_flash

#HOMEKIT_SPI_FLASH_BASE_ADDR ?= 0x7A000 orginal
#HOMEKIT_SPI_FLASH_BASE_ADDR = 0xcb000 # 831.488

GermanSheepDog, 2019-07-11:
Hi kiralikbeyin,
thank you for trying to help quickly. In past I used the default address 0x200000.
I did this:
1. make menuconfig
2. Change HOMEKIT_SPI_FLASH_BASE_ADDR to 0xb0000
3. make clean
4. make erase_flash flash monitor
I got:
```
WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0xb0000
!!! HomeKit: Failed to erase flash
>>> HomeKit: Generated new accessory ID: 7A:D9:50:0C:D4:26
!!! HomeKit: Failed to write accessory ID to flash
>>> HomeKit: Generated new accessory key
!!! HomeKit: Failed to write accessory key to flash
>>> HomeKit: Configuring mDNS
```
Here I used an ESP32 and I think the correct address is 0x200000. I looked at Kconfig:
```
config HOMEKIT_SPI_FLASH_BASE_ADDR
    string "SPI flash address for storing HomeKit data"
    default 0x200000
    help
        SPI flash address to store HomeKit related persisted data
```
The address is 0xb0000 is to small.
I think my problem is at the environment of 'homekit_characteristic_add_notify_callback'!
The HomeKit pair verify worked fine: 'Verification successful, secure session established'
At 'get accessories' my problems started.

kiralikbeyin, 2019-07-11:
@GermanSheepDog  i dont have any experince with esp32 but if your uncompressed firmware.bin size is less than 700k you can try 0xb0000 or 0xcb000 (831.488)  so HomeKit setup will be at the veri end of your flash.

I am not sure if esp32 accepts more than 1MB binary..

GermanSheepDog, 2019-07-11:
@kiralikbeyin: I am sure that 0x200000 worked at an ESP32. I have some experience and I have a static example that worked fine. I got some problems with the dynamic example and at an other program with the heap size:  [Issue 98](https://github.com/maximkulkin/esp-homekit/issues/98)

kiralikbeyin, 2019-07-11:
Please can you send a link of your HTTPS client lib?

I am using https://github.com/SuperHouse/esp-open-rtos/tree/master/extras/mbedtls and i am chaning config.h to get all data (4096 heap direclty used by lib)

```
#define MBEDTLS_SSL_MAX_CONTENT_LEN              4096 /**< Maxium fragment length in bytes, determines the size of each of the two internal I/O buffers */
```

I tried to get data with wolfssl which is already in this Homekit lib but couldnt succeed..

Maybe dynamic example has a special situation with ESP32. Did you try with ESP8266?

GermanSheepDog, 2019-07-11:
@kiralikbeyin: I used https://github.com/loboris/ESP32_curl_example.
No, I dianot try with ESP8266 because the memory is to small for my application.

GermanSheepDog, 2019-07-11:
@kiralikbeyin + @maximkulkin:
I found two challenges and solved them:
1. ESP32 need more Stack Size:
```
void lamp_identify(homekit_value_t _value) {
    printf("Lamp identify\n");
    #if ESP32 != 1
       xTaskCreate(lamp_identify_task, "Lamp identify", 256, NULL, 2, NULL);
    #else
       xTaskCreate(lamp_identify_task, "Lamp identify", 2048, NULL, 2, NULL);
    #endif
}
```
2. HOMEKIT_CHARACTERISTIC_CALLBACK did not work (only at ESP32?)
```
*(s++) = NEW_HOMEKIT_SERVICE(LIGHTBULB, .characteristics=(homekit_characteristic_t*[]) {
            NEW_HOMEKIT_CHARACTERISTIC(NAME, relay_name_value),
            NEW_HOMEKIT_CHARACTERISTIC(
                ON, true,
                .setter = relay_callback,
                .context = (void*)&relay_gpios[i],
                //.callback=HOMEKIT_CHARACTERISTIC_CALLBACK(
                //  relay_callback, .context=(void*)&relay_gpios[i]
                //),
            ),
            NULL
        });
```
Now relay_callback was called, but the value of context is not correct in procedure relay_callback. Did it work at ESP8266?

kiralikbeyin, 2019-07-11:
HOMEKIT_CHARACTERISTIC_CALLBACK words fine with current commit for ESP8266

GermanSheepDog, 2019-07-11:
@kiralikbeyin: Thank you for the quick answer. On ESP32 it did not work (or I made a big mistake).

GermanSheepDog, 2019-07-11:
@kiralikbeyin + @maximkulkin: I found a workaround:
```
        *(s++) = NEW_HOMEKIT_SERVICE(LIGHTBULB, .characteristics=(homekit_characteristic_t*[]) {
            NEW_HOMEKIT_CHARACTERISTIC(NAME, relay_name_value),
            NEW_HOMEKIT_CHARACTERISTIC(
                ON, true,
                #if ESP32 != 1
                   .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(
                    relay_callback, .context=(void*)&relay_gpios[i]
                  ),
                #else
                   .setter_ex = relay_callback,
                   .context = (void*) relay_gpios[i],                
                #endif
            ),
            NULL
        });
```
and:
```
#if ESP32 != 1
  void relay_callback(homekit_characteristic_t *ch, homekit_value_t value, void *_context) {
     uint8_t *gpio = _context;
     relay_write(*gpio, value.bool_value);
  }
#else
  void relay_callback(homekit_characteristic_t *ch, const homekit_value_t value) {
      uint8_t gpio = (uint8_t) ch->context;
      relay_write(gpio, value.bool_value);
  }
#endif
```
Now I am sure that HOMEKIT_CHARACTERISTIC_CALLBACK is a problem at ESP32.

GermanSheepDog, 2019-07-12:
@kiralikbeyin + @maximkulkin: I updated all software components to last version:
- esp-homekit
- esp-cjson
- esp-http-parser
- esp-wolfssl
- esp-idf

Summary: I got the same error in homekit_characteristic_has_notify_callback.

During this change I found two little challenges:
1. In file CMakeListst.txt in esp-homekit there are used the names without "esp-". Please look here:
set(COMPONENT_REQUIRES wolfssl cJSON http-parser mdns)
I would have expected:
set(COMPONENT_REQUIRES esp-wolfssl esp-cjson esp-http-parser mdns)
2. In esp-homekit-demo/examples/dynamic_services/main.c there was used an wrong variable in malloc:
```
for (int i=0; i < relay_count; i++) {
        int relay_name_len = snprintf(NULL, 0, "Relay %d", i + 1);
        char *relay_name_value = malloc(relay_name_len+1);  // <- not name_len!!!
        snprintf(relay_name_value, relay_name_len+1, "Relay %d", i + 1);
```

maximkulkin, 2019-07-12:
I wonder, why do you use CMake? I'm using regular Make, works fine.

Good catch with that variable, fixed it, although that should not have affected you as name_len is larger anyways (unless you're creating a 10 million of services).

GermanSheepDog, 2019-07-13:
I didn‘t use CMake. I load all modules separatly, got errors, renamed all diretories and it worked. Then I search for the filenames and it was an mistake of me. 
Did you have an idea why  HOMEKIT_CHARACTERISTIC_CALLBACK or (!) the callback-function-call was a problem at ESP32? I can send you the whole example.

maximkulkin, 2019-07-13:
@GermanSheepDog I plan to test that example on ESP32 In near future.

> I can send you the whole example.

I thought you just used dynamic_servies example on ESP32. If not, yeah, send me your code in full.

GermanSheepDog, 2019-07-13:
@maximkulkin Thanks for planning the activity in near future.

Yes, the base was the dynamic_service example. But I changed the system calls from ESP8266 to ESP32. You do not have to do this a second time.
You can use my example 'M5Stack_20_Dyn_Homekit.zip' above.

skanico, 2019-07-13:
@GermanSheepDog 
Hello,

Here is an example that works for me on ESP32 with HOMEKIT_CHARACTERISTIC_CALLBACK

[Archive.zip](https://github.com/maximkulkin/esp-homekit-demo/files/3388924/Archive.zip)

But I tried to optimize my code to remove redundancy (based on dynamic_services) and it fails like you say.
I don't know why... maybe because of NEW_HOMEKIT_..., or the context pointer?

I'll tell you if I find something.

maximkulkin, 2019-07-14:
I just have converted dynamic_services example to ESP32, tested it and it worked fine. See [examples/esp32/dynamic_services](https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/esp32/dynamic_services).

I checked your code, it looks correct, but I guess (unless what I was seeing was not the latest code) is that you run not the latest version of esp-homekit: you miss commit 9ae79f4 "Fix cloning characteristic callbacks" and that commit fixes exactly your problem.

Also, I should say that this is an awful way to maintain cross platform code: you have #if statements everywhere which mess up code a lot. E.g. to abstract GPIO API, you could

a. Use #defines to alias ESP8266 APIs to translate into ESP32 APIs:

```c
#if ESP32
#define gpio_enable gpio_set_direction
#define gpio_write gpio_set_level
#define GPIO_OUTPUT GPIO_MODE_OUTPUT
#endif
```
and continue using ESP8266 versions in the code (or vise versa)

b. Abstract away those APIs and create your own which would call platform specific APIs:

```c
#if ESP32
#define MY_GPIO_OUTPUT GPIO_MODE_OUTPUT
void my_gpio_init(uint8_t gpio, uint8_t mode) {
    gpio_set_direction(gpio, mode);
}
void my_gpio_write(uint8_t gpio, bool level) {
    gpio_set_level(gpio, level);
}
#elif ESP8266
#define MY_GPIO_OUTPUT GPIO_OUTPUT
void my_gpio_init(uint8_t gpio, uint8_t mode) {
    gpio_enable(gpio, mode);
}
void my_gpio_write(uint8_t gpio, bool level) {
    gpio_write(gpio, level);
}
#endif
```
and use this abstract API in the code.

closed 2019-07-14 by maximkulkin