maximkulkin/esp-homekit-demo #223: Dimmable LED

Partizan7676, 2019-06-06:
Maxim, can you create an example of dimmable led via PWM ? I try sonoff_basic_pwm but it seems it is not for NodeMCU and has all pins inverted.

maximkulkin, 2019-06-06:
Not sure I understand which pins are inverted

AchimPieters, 2019-06-07:
You can use the `sonoff_basic_pwm` demo. you onlu have to change these lines so the matc the GPIO's you want to use:

```
// The GPIO pin that is connected to the relay on the Sonoff Basic.
const int relay_gpio = 12;
// The GPIO pin that is connected to the LED on the Sonoff Basic.
const int led_gpio = 13;
// The GPIO pin that is connected to the button on the Sonoff Basic.
const int button_gpio = 0;
// The GPIO pin that is connected to the header on the Sonoff Basic (external switch).
const int toggle_gpio = 14;

#include <pwm.h>
// The PWM pin that is connected to the PWM daughter board.
const int pwm_gpio = 13;
```

Partizan7676, 2019-06-08:
AchimPieters i know that, but all pins for sonoff are inverted. if i connect pwm board with led - when its off - i see minimal light, when brightness is set to 100% led is off. btw when i increase brightness led is fading out.

Yousefff1, 2019-06-10:
I am trying to use sonoff-basic-pwm example with a Nodemcu and an AC dimming module, on the module there are 4 pins, VCC, GND, Zero Crossing and PWM

Using the sketch and flashing it to a nodemcu, I was able to dim the onboard LED, my question is:

Where to connect the Zero Crossing pin from the module to the Nodemcu?

The link for the AC dimming module
https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html

Thank you,

lrusnac, 2019-07-08:
@Partizan7676  you can change the code easily:
https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/sonoff_basic_pwm/main.c#L63
this becomes: `gpio_write(led_gpio, on);`
and 
https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/sonoff_basic_pwm/main.c#L108
instead of `false` use `true`, that should do the trick, cheers

okanduzyel, 2019-09-25:
Hi, 

I investigated your example folder and unfortunately, I couldn't find and RGB led example. Is there? or not? If it is not, Can you create a RGB led example?

lrusnac, 2019-09-25:
There is some RGB stuff in the led_strip example https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/led_strip/led_strip.c <https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/led_strip/led_strip.c>

okanduzyel, 2019-09-25:
I was expecting to drive RGB led via 3 pins and mosfets. I guess, your application drives i2c chip, right? Is there any example like what I want?

maximkulkin, 2019-09-25:
> Can you create a RGB led example?

@okanduzyel This is not how it works. I'm not here to fulfill each and every your homekit accessory fantasy. Most examples I wrote is to show how FRAMEWORK works/should be used. I have no interest in writing custom LED strip controllers for every device you have over and over again. The way it does work is: if you've used this framework to implement a very new and interesting accessory, you can send pull request to share it with the world. I emphasize, a new or interesting accessory. Having just another LED controller, because someone might not have skill/30 minutes to customize existing example, is not the purpose of this repo.

okanduzyel, 2019-09-25:
Hi, 

Can I download your files from GitHub and compile on my computer via any compiler? I guess, vagrant doesn't work on my MacBook Pro. Maybe, I miss some steps. You have shown it on windows 10.

lrusnac, 2019-09-25:
@okanduzyel it's in the main README file on the repository: https://github.com/maximkulkin/esp-homekit-demo#usage

okanduzyel, 2019-09-25:
> @okanduzyel it's in the main README file on the repository: https://github.com/maximkulkin/esp-homekit-demo#usage
> 
> points to: https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions

I am freshman and couldn't build the instructions correctly.

lrusnac, 2019-09-25:
@okanduzyel I don't think this is the best place to get help on questions like this.
I followed the docker instructions and works flawlessly I don't know where you got vagrant stuff from. I also run it on mac os

kenthinson, 2019-10-09:
@okanduzyel 
3 misfit rgb pwn project is here. 
https://github.com/DominikHorn/esp8266-rgb-led-strip
I am currently modifying it to single mosfit (white only) strip. I will post the code tomorrow.

AchimPieters, 2019-10-15:
@okanduzyel Use the Docker script [esp-sdk-dockerfile](https://github.com/AchimPieters/HomeKit-Core/blob/master/esp-sdk-dockerfile) and [esp-rtos-dockerfile](https://github.com/AchimPieters/HomeKit-Core/blob/master/esp-rtos-dockerfile). and everything works just fine. Maybe [this ](https://www.studiopieters.nl/esp-homekit-sdk-full-installation/)will help?

closed 2019-11-05 by maximkulkin