maximkulkin/esp-homekit-demo #226: History graph like EVE homekit products

Martvvliet, 2019-06-12:
Hi,

I was wondering how to enable the history in the homekit devices. WIth the EVE homekit app you can read out the history. This works with EVE product, but also with my flora sensor that I connected by Homebridge. I'm building an air quality sensor and I was wondering if it is possible to enable this with this software?

maximkulkin, 2019-06-12:
You need to accumulate history on accessory end and implement EVE's custom characteristics that will expose that history to EVE app.

Martvvliet, 2019-06-13:
Thanks for the response, are you familiar with these characteristics?

maximkulkin, 2019-06-13:
I'm not, but a quick google search gave this: https://gist.github.com/simont77/3f4d4330fa55b83f8ca96388d9004e7d

dezdeepblue, 2019-06-15:
Duplicate of #121

closed 2019-09-04 by maximkulkin