maximkulkin/esp-homekit-demo #227: Unable to complete pairing

kunjal83, 2019-06-16:
Hi,
While trying to pair the device, program all of a sudden closes the connection after "Get Accessories". 

Can someone please help on how do i troubleshoot this? 

[image]

Thanks,
Kunjal

toshibochan, 2019-11-23:
@kunjal83 I having this same problem. did you fix this problem?

kunjal83, 2019-11-24:
@toshibochan, that was happening due to Pin2 going low. 
Was some hardware related issue.
Please do check the voltage at the pins.

toshibochan, 2019-11-24:
@kunjal83 what you did to fix?

toshibochan, 2019-11-24:
pin 2 mean gpio2?

kunjal83, 2019-11-24:
Yes sorry... Yes i meant GPIO2.
Ensure it is high always

Cheers,
Kunjal
Cell: +91 9987274445

toshibochan, 2019-11-26:
high is 0 or 1?

toshibochan, 2019-11-26:
#include <stdio.h>
#include <stdlib.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>
#include <math.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"

#include "WS2812FX/WS2812FX.h"

#define LED_RGB_SCALE 255       // this is the scaling factor used for color conversion
#define LED_COUNT 50            // this is the number of WS2812B leds on the strip
#define LED_INBUILT_GPIO 2      // this is the onboard LED used to show on/off only

// Global variables
float led_hue = 0;              // hue is scaled 0 to 360
float led_saturation = 100;      // saturation is scaled 0 to 100
float led_brightness = 33;     // brightness is scaled 0 to 100
bool led_on = false;            // on is boolean on or off
bool led_on_value = (bool)0;                // this is the value to write to GPIO for led on (0 = GPIO low)

float fx_hue = 64;              // hue is scaled 0 to 360
float fx_saturation = 50;      // saturation is scaled 0 to 100
float fx_brightness = 50;     // brightness is scaled 0 to 100
bool fx_on = true;

//http://blog.saikoled.com/post/44677718712/how-to-convert-from-hsi-to-rgb-white
static void hsi2rgb(float h, float s, float i, ws2812_pixel_t* rgb) {
    int r, g, b;

    while (h < 0) { h += 360.0F; };     // cycle h around to 0-360 degrees
    while (h >= 360) { h -= 360.0F; };
    h = 3.14159F*h / 180.0F;            // convert to radians.
    s /= 100.0F;                        // from percentage to ratio
    i /= 100.0F;                        // from percentage to ratio
    s = s > 0 ? (s < 1 ? s : 1) : 0;    // clamp s and i to interval [0,1]
    i = i > 0 ? (i < 1 ? i : 1) : 0;    // clamp s and i to interval [0,1]
    i = i * sqrt(i);                    // shape intensity to have finer granularity near 0

    if (h < 2.09439) {
        r = LED_RGB_SCALE * i / 3 * (1 + s * cos(h) / cos(1.047196667 - h));
        g = LED_RGB_SCALE * i / 3 * (1 + s * (1 - cos(h) / cos(1.047196667 - h)));
        b = LED_RGB_SCALE * i / 3 * (1 - s);
    }
    else if (h < 4.188787) {
        h = h - 2.09439;
        g = LED_RGB_SCALE * i / 3 * (1 + s * cos(h) / cos(1.047196667 - h));
        b = LED_RGB_SCALE * i / 3 * (1 + s * (1 - cos(h) / cos(1.047196667 - h)));
        r = LED_RGB_SCALE * i / 3 * (1 - s);
    }
    else {
        h = h - 4.188787;
        b = LED_RGB_SCALE * i / 3 * (1 + s * cos(h) / cos(1.047196667 - h));
        r = LED_RGB_SCALE * i / 3 * (1 + s * (1 - cos(h) / cos(1.047196667 - h)));
        g = LED_RGB_SCALE * i / 3 * (1 - s);
    }

    rgb->red = (uint8_t) r;
    rgb->green = (uint8_t) g;
    rgb->blue = (uint8_t) b;
    rgb->white = (uint8_t) 0;           // white channel is not used
}

static void wifi_init() {
[... 233 lines cut ...]

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void user_init(void) {
    // uart_set_baud(0, 115200);

    // This example shows how to use same firmware for multiple similar accessories
    // without name conflicts. It uses the last 3 bytes of accessory's MAC address as
    // accessory name suffix.
    uint8_t macaddr[6];
    sdk_wifi_get_macaddr(STATION_IF, macaddr);
    int name_len = snprintf(NULL, 0, "Chihiro-%02X%02X%02X", macaddr[3], macaddr[4], macaddr[5]);
    char *name_value = malloc(name_len + 1);
    snprintf(name_value, name_len + 1, "Chihiro-%02X%02X%02X", macaddr[3], macaddr[4], macaddr[5]);
    name.value = HOMEKIT_STRING(name_value);

    wifi_init();
    WS2812FX_init(LED_COUNT);
    homekit_server_init(&config);
    
    led_identify(HOMEKIT_INT(led_brightness));
}

toshibochan, 2019-11-26:
I can see this is set to 1

for (int i = 0; i < 3; i++) {
    for (int j = 0; j < 3; j++) {
        gpio_write(LED_INBUILT_GPIO, (int)led_on_value);
        vTaskDelay(100 / portTICK_PERIOD_MS);
        gpio_write(LED_INBUILT_GPIO, 1 - (int)led_on_value);
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }
    vTaskDelay(250 / portTICK_PERIOD_MS);
}

gpio_write(LED_INBUILT_GPIO, 1 - (int)led_on_value);

vTaskDelete(NULL);

toshibochan, 2019-11-26:
im trying to pair Wemos D1 mini but >>> HomeKit: [Client 5] Closing client connection

>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: [Client 5] Pair Setup Step 1/3
>>> HomeKit: [Client 5] Closing client connection

maccoylton, 2019-11-27:
Can you pull the latest version of esp-homekit, build and test again, it will  give some extra info to potentially help diagnose

d4rkmen, 2020-01-13:
Here is detailed log of pairing process on ESP8266. Connection were closed a few times before it become paired
```
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: E8:EC:BC:93:02:CF
>>> HomeKit: Configuring mDNS
>>> homekit_run_server: Staring HTTP server
>>> HomeKit: Got new client connection: 1 from 192.168.1.33
>>> homekit_client_process: [Client 1] Got 131 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 48900
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 48816
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 1] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 48144
>>> homekit_server_on_pair_setup: [Client 1] Using user-specified password: 111-11-111
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> send_tlv_response: [Client 1] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \xF3>\\\x0C\x06%\xA7\xB0\xBC>\xBB+\x83\xCAXE\xF5\xEC\x9AZ\x1F\xD6"\x19\xEF\x0AL\x1F\xE5=<+mJ3&\x
11\x08\xEBG\x07\xBC\x1A"!\x9C\xB7\x89u\x95\xBCp\x03\xC0\xEA\xDCC\x92\x985\x0D5/\x9A\x9EDg\xDE\x1F\xBD\xCA7\xC3\x17\x03RW\xD7\xBF\xAA\xD6\
xEB\xE53\x0C\xA1\xC5k\xD3\xA2\x98N\xB0\xDE\x12\xCD\xC7?\x17\x1E\x823V\xDB\xDE\xC6\xE2\x9E\xBC\x07\x99\x84\xFC\xB27W\x1A|\xFD\xE0\xB7\xBB\
xC1\x1A\xD1\xDF{AN\x1B\x94\xCB\xCA\x1B`\xAA\xA2\xCEU!\xF7~\x9C\xB7\xCF\xE2C\x86\xA1\x11=\xC8-,\xA7\xE9>\x05\x90\xE2\x04;xv\x82E\xFA}[\xEE
\x84fc\x9A\x97\xB9\x8C\xC9\x97\xFF\xBC\xDB\x8D\xFB\xCFB\x09c\xE2\x95\x84\\@!~v\x1C\xE9\xED\x9C\xE3b\x15y"\x00\x02' \xF28\xEB\x93\x0D\x1C\
xD7\xE9\x06\xC4'\x06\x03\x9D\x12\x9A\xDE\xCE1\xF6Q-\x91\xF5\xDD\xB8q\xEDUcL\xB3Z\xF9\xB5\x1C\xE16\xA1P*x\x86\x8C\xB9\xF2\xFE\x8E\xCA\xD7*
6\x1D\x01]\xCF\xCFR\xD48`HaQG7\x91Q\xF7\xB5\xAE5D\xDE\xA3sQ\xBE\xF8\xCE},u\x13\xC0HjbC\xE6\xEFS\xF7\xBCL\xB62\xFB\x81\\{{R\xCF\xC9\xEE\xE
5\xD5jU\x9D\xD0\x03\xE1?\xA5\xFA\x83\xC8\xDF\xEFN:\xFB{\xECv\xCE\x91\x07\x9C\x02\xFE.>\x8F!K\x8E\x82\xDBN\xEF\xFF\x0F1e\x96\xC7\xCC\x0B\x
1BB+g\xF0\xF7\x93\x8F\xF1(Yz\xF3\xF8R\xCC\xD8\xC91\xE2\xE8\xF3\x96\xF9
>>> tlv_debug: Type 2 value (16 bytes): \x05\x1D\x9D5\x83y\xD4l*4xa\x9B\xE1\xB2\xD4
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D
\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\xF3>\\\x0C\x06%\xA7\xB0\xBC>\xBB+\x83\xCAXE\xF5\xEC\x9AZ\x1F\xD6"\x19\xEF\x0AL\x1F\xE5
=<+mJ3&\x11\x08\xEBG\x07\xBC\x1A"!\x9C\xB7\x89u\x95\xBCp\x03\xC0\xEA\xDCC\x92\x985\x0D5/\x9A\x9EDg\xDE\x1F\xBD\xCA7\xC3\x17\x03RW\xD7\xBF
\xAA\xD6\xEB\xE53\x0C\xA1\xC5k\xD3\xA2\x98N\xB0\xDE\x12\xCD\xC7?\x17\x1E\x823V\xDB\xDE\xC6\xE2\x9E\xBC\x07\x99\x84\xFC\xB27W\x1A|\xFD\xE0
\xB7\xBB\xC1\x1A\xD1\xDF{AN\x1B\x94\xCB\xCA\x1B`\xAA\xA2\xCEU!\xF7~\x9C\xB7\xCF\xE2C\x86\xA1\x11=\xC8-,\xA7\xE9>\x05\x90\xE2\x04;xv\x82E\
xFA}[\xEE\x84fc\x9A\x97\xB9\x8C\xC9\x97\xFF\xBC\xDB\x8D\xFB\xCFB\x09c\xE2\x95\x84\\@!~v\x1C\xE9\xED\x9C\xE3b\x15y"\x00\x02' \xF28\xEB\x93
\x0D\x1C\xD7\xE9\x06\xC4'\x06\x03\x9D\x12\x9A\xDE\xCE1\xF6Q-\x91\xF5\xDD\xB8q\xEDUcL\xB3Z\xF9\xB5\x1C\xE16\xA1P*x\x86\x8C\xB9\xF2\x03\x81
\xFE\x8E\xCA\xD7*6\x1D\x01]\xCF\xCFR\xD48`HaQG7\x91Q\xF7\xB5\xAE5D\xDE\xA3sQ\xBE\xF8\xCE},u\x13\xC0HjbC\xE6\xEFS\xF7\xBCL\xB62\xFB\x81\\{
{R\xCF\xC9\xEE\xE5\xD5jU\x9D\xD0\x03\xE1?\xA5\xFA\x83\xC8\xDF\xEFN:\xFB{\xECv\xCE\x91\x07\x9C\x02\xFE.>\x8F!K\x8E\x82\xDBN\xEF\xFF\x0F1e\
x96\xC7\xCC\x0B\x1BB+g\xF0\xF7\x93\x8F\xF1(Yz\xF3\xF8R\xCC\xD8\xC91\xE2\xE8\xF3\x96\xF9\x02\x10\x05\x1D\x9D5\x83y\xD4l*4xa\x9B\xE1\xB2\xD
4\x06\x01\x02
>>> homekit_client_process: [Client 1] Finished processing
>>> HomeKit: Got new client connection: 2 from 192.168.1.33
>>> HomeKit: [Client 1] Closing client connection
>>> HomeKit: Got new client connection: 1 from 192.168.1.33
>>> homekit_client_process: [Client 2] Got 131 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 46532
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 2] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 46448
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 2] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 45784
>>> homekit_server_on_pair_setup: [Client 2] Using user-specified password: 111-11-111
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
[... 134 lines cut ...]
>>> tlv_debug: Type 4 value (64 bytes): 5\xA0Qd\xCB\xAA\xD5\x08\x97\x0FJ"u\x85\x13\xBAL\x88\xDEk\x1D2'\xACm\xB2\xEF\xBF\x07\xE0\xD9V\xFE\
xEF\xA3\xD9\x17\x92?\xA5\x84\x86g \xF8\xB1\xB3\xF4\x0C\x1D!r%hs|\xED\xC8h\xB4qU`\xA6
>>> client_send: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 69\x0D\
[... cut ...]
>>> HomeKit: [Client 1] Pair Setup Step 3/3
>>> homekit_server_on_pair_setup: Free heap: 46160
>>> homekit_server_on_pair_setup: [Client 1] Calculating shared secret
>>> homekit_server_on_pair_setup: [Client 1] Decrypting payload
>>> homekit_server_on_pair_setup: [Client 1] Importing device public key
>>> homekit_server_on_pair_setup: [Client 1] Calculating DeviceX
>>> homekit_server_on_pair_setup: [Client 1] Verifying device signature
>>> HomeKit: Added pairing with 8F39E629-B67E-44CD-81FD-E7D729CB0B77

```

toshibochan, 2020-01-13:
how to start log?

d4rkmen, 2020-01-13:
> how to start log?

in sdk config set debug for homekit component

> Can you pull the latest version of esp-homekit, build and test again, it will give some extra info to potentially help diagnose

i have used the latest sources and the log above shows there is a problem - connection drop after first pairing request

RavenSystem, 2020-01-13:
With esp-open-rtos, can you try again using my modified mdnsresponder?

https://github.com/RavenSystem/esp-homekit-devices/blob/master/external_libs/homekit/src/mdnsresponder.h

https://github.com/RavenSystem/esp-homekit-devices/blob/master/external_libs/homekit/src/mdnsresponder.c

d4rkmen, 2020-01-13:
@RavenSystem are you sure its related to pairing? Or its about remote access?

RavenSystem, 2020-01-13:
Pairing.

d4rkmen, 2020-01-13:
Solved (!!!) by sending header and payload separate in send_tlv_response()
```c
    static char *http_headers =
        "HTTP/1.1 200 OK\r\n"
        "Content-Type: application/pairing+tlv8\r\n"
        "Content-Length: %d\r\n"
        "Connection: keep-alive\r\n\r\n";
    int response_size = strlen(http_headers) + 32;
    char *response = malloc(response_size);
    int header_len = snprintf(response, response_size, http_headers, payload_size);
    client_send(context, (byte *)response, header_len);
    free(response);
    client_send(context, (byte *)payload, payload_size);
    free(payload);
```
Now pairing is always @ the first atempt

d4rkmen, 2020-01-15:
Located the problem: controller closes connection before got response due to long timeout. Fixed with sending http header before long-time crypto procedures, then body when data is ready
Thanks for good library

toshibochan, 2020-01-15:
@d4rkmen you can send request to fix the issue?

maccoylton, 2020-01-15:
@d4rkmen did you submit a pull request for your proposed changes?

d4rkmen, 2020-01-16:
Suming up a few days of tries: splitting request to parts not gives any effort. Problem seems different kind of. And seems its related to mdns. After some time (~2 mintures) the accessory stops discovering in home app. If pairind happen at the same time - controller makes retry (closes current connection, and starts new pairing context)
I have no idea why this happens. I have tested other mdns library on mongoose-os and there is no such problems: accessory stays discoverable all the time (before paitring done) Any thoughts?

maximkulkin, 2020-01-16:
@d4rkmen I wonder what firmware are you running so that you're getting pairing timeouts? Are you doing funny stuff in other threads?

d4rkmen, 2020-01-16:
@maximkulkin its the latest ESP8266_RTOS_SDK + led example from this repo.
Found temporary workaround: mdns setup before pairing. Will push PR for you, guys, to test if you like. Its now 10 / 10 success pairings for me.

maximkulkin, 2020-01-16:
Try esp-open-rtos and see if the problem reproduces

d4rkmen, 2020-01-16:
@maximkulkin i will (i saw there were changes on mdns last days)
But I move from esp-open-rtos because ot this issue.

maximkulkin, 2020-01-16:
I generally do not use ESP8266_RTOS_SDK, have only done one example as proof of concept and haven't tested it for long period of time, but esp-open-rtos works for me all the time, and if it doesn't work for you, you might be doing something wrong.

d4rkmen, 2020-01-17:
Not sure why, but there were pairing and remote access problems on esp-open-rtos. Then I decided to check official SDK and I like it better. Just tested now on esp-open-rtos:
```
[MAC] >>> HomeKit: Got new client connection: 5 from 192.168.1.33
[MAC] >>> HomeKit: [Client 5] Pair Setup Step 1/3
[MAC] >>> HomeKit: Got new client connection: 6 from 192.168.1.33
[MAC] >>> HomeKit: [Client 5] Closing client connection
[MAC] >>> HomeKit: [Client 6] Pair Setup Step 1/3
[MAC] >>> HomeKit: Got new client connection: 5 from 192.168.1.33
[MAC] >>> HomeKit: [Client 5] Pair Setup Step 1/3
[MAC] >>> HomeKit: [Client 5] Refusing to pair: another pairing in progress
[MAC] >>> HomeKit: [Client 5] Closing client connection
[MAC] >>> HomeKit: [Client 6] Closing client connection
```
result: failed to pair (one of 3-4 pairings) - with workaround works ok, remote access not works :(

maximkulkin, 2020-01-17:
What is your firmware?

closed 2020-02-13 by maximkulkin