maximkulkin/esp-homekit-demo #229: Garage door opener - sliding gate opener

carneirobeatriz, 2019-07-01:
Hello everybody. First I’d like to congratulate for the very good job. I am amazed with the library. 

I have a slide gate that open automatic with a remote control. I was wondering if I could use the Garage example to make it work with my hardware. (https://www.amazon.com/CO-Z-Automatic-Hardware-Driveway-Security/dp/B0197V3OD2?ref_=fsclp_pl_dp_3)

In its board, have 2 relays that when triggered open and close the gate and use 2 reed switches to manage when to stop.

My doubt is that in the code, I only see one relay and one reed switch. On the other hand, I saw YouTube video that it was able to make it work with a Sonoff.

Have you guys stumbled in a similar situation or know an alternative?
Thank you very much

maximkulkin, 2019-07-14:
Ok, I agree, current garage door opener is a bit weird. Looks like all it does is rely on existing door opener controller to do opening/closing and ESP just flicks relay that is connected to a button that controls stock door controller (which kind of surprising, it might have been just a pin, but who knows, maybe it's a 12V button). Anyways, I agree, it would be nice to have an another controller, that controls up/down commands individually.

xfce, 2019-07-21:
I can design the controller, but can not code :(

xd94402002, 2019-08-22:
This is the circuit I design for using one relay to control two switches with state sensor. I original design for homebridge plugin homebridge-gpio-garage-door. It also prevent bounce when relay switch its state. Using 2 or 4-channel relay module. I test it on bread broad with no issue but I'm not sure it can work perfect with real door.  

[image]

closed 2019-09-04 by maximkulkin