maximkulkin/esp-homekit-demo #230: 16 Accessories - Question

kiralikbeyin, 2019-07-06:
@maximkulkin 
I cant add more than 8 accessories. How is it possible?

kiralikbeyin, 2019-07-08:
@maximkulkin 

I always see iid 7 when setupping on homekit app. Accessories disapperad if there are more than 8.

````
Processing element {
"aid":	2,
"iid":	7,
"value":	true
}
````

maximkulkin, 2019-07-08:
I never have tried that, can you provide your accessory database code (that HOMEKIT_ACCESSORY macro construct)

kiralikbeyin, 2019-07-08:
I tried a lot of combinations latest here:

````
homekit_accessory_t *accessories[] = {
  HOMEKIT_ACCESSORY(.id = 1, .category = homekit_accessory_category_switch, .services = (homekit_service_t*[]) {
    HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics = (homekit_characteristic_t*[]) {
      &name,
      &manufacturer,
      //&serial,
      HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "xxx"),
      HOMEKIT_CHARACTERISTIC(MODEL, "xxx"),
      HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.7.9"),
      HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
      //            HOMEKIT_CHARACTERISTIC(IDENTIFY, switch_identify),
      NULL
    }),

    HOMEKIT_SERVICE(SWITCH, .primary = true, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 1"),
                             &switch_on_1,
                             NULL
    }),
    HOMEKIT_SERVICE(SWITCH, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 2"),
                             &switch_on_2,
                             NULL
    }),
    HOMEKIT_SERVICE(SWITCH,  .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 3"),
                             &switch_on_3,
                             NULL
    }),
    HOMEKIT_SERVICE(SWITCH, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 4"),
                             &switch_on_4,
                             NULL
    }),
    HOMEKIT_SERVICE(OUTLET,   .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 5"),
                             &switch_on_5,
                             HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
                             NULL
    }),
    HOMEKIT_SERVICE(OUTLET, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 6"),
                             &switch_on_6,
                             HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
                             NULL
    }),

    NULL
  }),

  HOMEKIT_ACCESSORY(.id = 2, .category = homekit_accessory_category_outlet, .services = (homekit_service_t*[]) {
    HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics = (homekit_characteristic_t*[]) {
      &name,
      &manufacturer,
      //&serial,
      HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "xxx"),
      HOMEKIT_CHARACTERISTIC(MODEL, "xxx"),
      HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.7.9"),
      HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
      //            HOMEKIT_CHARACTERISTIC(IDENTIFY, switch_identify),
      NULL
    }),

    HOMEKIT_SERVICE(OUTLET,  .primary = true, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 7"),
                             &switch_on_7,
                             HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
                             NULL
    }),
    HOMEKIT_SERVICE(OUTLET, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 8"),
                             &switch_on_8,
                             HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
                             NULL
    }),
    HOMEKIT_SERVICE(OUTLET, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 9"),
                             &switch_on_9,
                             HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
                             NULL
    }),

    HOMEKIT_SERVICE(OUTLET, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 10"),
                             &switch_on_10,
                             HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
                             NULL
    }),
    HOMEKIT_SERVICE(OUTLET, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 11"),
                             &switch_on_11,
                             HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
                             NULL
    }),
    HOMEKIT_SERVICE(OUTLET, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 12"),
                             &switch_on_12,
                             HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
                             NULL
    }),

    NULL
  }),

  NULL
};

````

maximkulkin, 2019-07-08:
Ok, one thing that I can immediately spot is that you reuse characteristics in multiple places. Thing is that each characteristic should be assigned unique ID and should be associated with exactly one accessory. You are supposed to set .id property for each characteristic, but since it's so common and tedious operation that esp-homekit automates that for you. It seems that HomeKit on iOS side can detect inconsistencies and refuse to pair.

Bottom line: you should create separate characteristics and not reuse them (in your case, you're using same `name` and `manufacturer` characteristics for different accessories.

Also, your code formatting is inconsistent: instead of
```c
    HOMEKIT_SERVICE(OUTLET, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 12"),
                             &switch_on_12,
                             HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
                             NULL
    }),
```
you should have
```c
    HOMEKIT_SERVICE(OUTLET, .characteristics = (homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Switch 12"),
      &switch_on_12,
      HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
      NULL
    }),
```

kiralikbeyin, 2019-07-08:
i tried this code but it use too much heap..

````
HOMEKIT_ACCESSORY(.id = 1, .category = homekit_accessory_category_outlet, .services = (homekit_service_t*[]) {
  HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics = (homekit_characteristic_t*[]) {
    &name,
    &manufacturer,
    HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "get_my_ip"),
    HOMEKIT_CHARACTERISTIC(MODEL, "xxx"),
    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.7.9"),
    HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
    //            HOMEKIT_CHARACTERISTIC(IDENTIFY, switch_identify),
    NULL
  }),
  HOMEKIT_SERVICE(OUTLET, .primary = true, .characteristics = (homekit_characteristic_t*[]) {
    HOMEKIT_CHARACTERISTIC(NAME, "Switch 9"),
    &switch_on_9,
    HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
    NULL
  }),
  NULL
}),
HOMEKIT_ACCESSORY(.id = 2, .category = homekit_accessory_category_outlet, .services = (homekit_service_t*[]) {
  HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics = (homekit_characteristic_t*[]) {
    &name,
    &manufacturer,
    HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "get_my_ip"),
    HOMEKIT_CHARACTERISTIC(MODEL, "xxx"),
    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.7.9"),
    HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
    //            HOMEKIT_CHARACTERISTIC(IDENTIFY, switch_identify),
    NULL
  }),
  HOMEKIT_SERVICE(OUTLET, .primary = true, .characteristics = (homekit_characteristic_t*[]) {
    HOMEKIT_CHARACTERISTIC(NAME, "Switch 10"),
    &switch_on_10,
    HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
    NULL
  }),
  NULL
}),
HOMEKIT_ACCESSORY(.id = 3, .category = homekit_accessory_category_outlet, .services = (homekit_service_t*[]) {
  HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics = (homekit_characteristic_t*[]) {
    &name,
    &manufacturer,
    HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "get_my_ip"),
    HOMEKIT_CHARACTERISTIC(MODEL, "xxx"),
    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.7.9"),
    HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
    //            HOMEKIT_CHARACTERISTIC(IDENTIFY, switch_identify),
    NULL
  }),
  HOMEKIT_SERVICE(OUTLET, .primary = true, .characteristics = (homekit_characteristic_t*[]) {
    HOMEKIT_CHARACTERISTIC(NAME, "Switch 11"),
    &switch_on_11,
    HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
    NULL
  }),
  NULL
}),
HOMEKIT_ACCESSORY(.id = 4, .category = homekit_accessory_category_outlet, .services = (homekit_service_t*[]) {
  HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics = (homekit_characteristic_t*[]) {
    &name,
    &manufacturer,
    HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "get_my_ip"),
    HOMEKIT_CHARACTERISTIC(MODEL, "xxx"),
    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.7.9"),
    HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
    //            HOMEKIT_CHARACTERISTIC(IDENTIFY, switch_identify),
    NULL
  }),
  HOMEKIT_SERVICE(OUTLET, .primary = true, .characteristics = (homekit_characteristic_t*[]) {
    HOMEKIT_CHARACTERISTIC(NAME, "Switch 12"),
    &switch_on_12,
    HOMEKIT_CHARACTERISTIC(OUTLET_IN_USE, true),
    NULL
  }),
  NULL
}),
`````

kiralikbeyin, 2019-07-09:
impossible to add 10-16 accesories

maximkulkin, 2019-07-09:
It's not about putting every service into a separate accessory, it's about every accessory referencing the same NAME and MANUFACTURER characteristics:

```c
HOMEKIT_ACCESSORY(.id = 4, .category = homekit_accessory_category_outlet, .services = (homekit_service_t*[]) {
  HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics = (homekit_characteristic_t*[]) {
    &name,           // <=== This is a problem
    &manufacturer,   // <=== And this is a problem
    HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "get_my_ip"),
    HOMEKIT_CHARACTERISTIC(MODEL, "xxx"),
    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.7.9"),
    HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
    //            HOMEKIT_CHARACTERISTIC(IDENTIFY, switch_identify),
    NULL
  }),
  ...
```

maximkulkin, 2019-07-09:
Ok, just for giggles I flashed a dynamic_services example with 12 pins (although they were repeated as it does not matter here) so I've got an accessory with 12 relays. It paired fine, I was able to switch them on and off and see their status changing in serial monitor.

[image]

maximkulkin, 2019-07-09:
Be sure to update to latest code version

kiralikbeyin, 2019-07-09:
@maximkulkin Thank you.

I missed something very stupid. When adding more than 8 accessories, HomeKit can not add more than 8 FAVORITE devices. 
This means i didnt checked rooms which shows all accessories.

Dynamic services example works and everything is clear.

closed 2019-07-09 by kiralikbeyin

spxnr, 2020-05-18:
> It's not about putting every service into a separate accessory, it's about every accessory referencing the same NAME and MANUFACTURER characteristics:

Hi @maximkulkin , I was hoping to pick your brain on this. Say I started putting my design in production, the name and manufacturer would be fixed. Why would Home app not want to connect to two of my (identical but incremented serial number) devices in the same home?

maximkulkin, 2020-05-18:
Manufacturer or serial number values are really not important at all. But the problem there is with NAME (characteristic inside ACCESSORY INFO service) of first accessory - it is used as hostname and as service name in mDNS announcements. HomeKit spec requires accessories to do conflict resolution on mDNS names, but unfortunately it is not implemented currently, so you need to manually make sure those names are unique. In a lot of accessory examples that name is suffixed with first 6 digits of device MAC address in runtime, so that gives enough flexibility to avoid those conflicts, so this is a recommended workaround.

I was looking into improving mDNS recently, but haven't implemented it yet and it is not on top of my priority list.