maximkulkin/esp-homekit-demo #231: Build Instruction problem

helgard999, 2019-07-08:
Hello 
I have a problem building everything on my Mac
docker is installed ...

Used this link:
ESP8266 (Docker) (recommended for macOS users)

step 4 Build esp-sdk Docker container:
docker build . -f esp-sdk-dockerfile -t esp-sdk
 It runs for quite a while then this message comes up:
 
Step 3/4 : COPY --from=builder /build/esp-open-sdk/xtensa-lx106-elf /opt/xtensa-lx106-elf
invalid from flag value builder: pull access denied for builder, repository does not exist or may require 'docker login'

I don’t know how to solve this, any ideas? I am logged in to docker, didn’t seem to make a difference.

Attached is the complete build log.
[buildlog.txt](https://github.com/maximkulkin/esp-homekit-demo/files/3369542/buildlog.txt)

Kind regards
Helgard

maximkulkin, 2019-07-08:
Tried on my Linux machine (don't have access to Mac at the moment, but result should be the same), worked fine for me:
```
... [skipped] ...
Setting up python-serial (3.0.1-1) ...
Processing triggers for libc-bin (2.23-0ubuntu11) ...
Removing intermediate container 54f326147a34
 ---> 571f025c38ac
Step 8/9 : COPY --from=builder /build/esp-open-sdk/xtensa-lx106-elf /opt/xtensa-lx106-elf
 ---> d2ea00761e79
Step 9/9 : ENV PATH /opt/xtensa-lx106-elf/bin:$PATH
 ---> Running in 7a1974293720
Removing intermediate container 7a1974293720
 ---> 296bbf5c18ae
Successfully built 296bbf5c18ae
Successfully tagged esp-rtos:latest
```

It looks like you Docker is completely unaware of multistage builds. What Docker version are you running (`docker --version`)? My Linux Docker version is 18.09.7 (build 2d0083d), latest stable [Mac Docker version](https://docs.docker.com/docker-for-mac/release-notes/) is 18.09.2, so it should be fine.

helgard999, 2019-07-09:
Hi Thanks for having a look...
Yes it was running on Docker 18.09.2 macOS Mojave 10.14.5.
I know how to program small embedded microcontrollers,  docker and linux is foreign to me.
Your instructions are for someone that knows a bit more about these things :-)
Where do you normally put your installation on the Mac, does it matter? 
Isn't there special paths or environmental variables or that need to be set before?
Regards HR

helgard999, 2019-07-09:
What do I do to retry the step? do  I re-issue the command from the start (docker build . -f esp-sdk-dockerfile -t esp-sdk) or can I continue with the last part?

maximkulkin, 2019-07-09:
You definitely need to finish first image before proceeding so rerun the command.

helgard999, 2019-07-10:
Hi 
Nope, rebooted and restarted everything..... same message.
 esp-hr docker build . -f esp-sdk-dockerfile -t esp-sdk
Sending build context to Docker daemon    171kB
Step 1/8 : FROM ubuntu:16.04 as builder
 ---> 13c9f1285025
Step 2/8 : RUN groupadd -g 1000 docker && useradd docker -u 1000 -g 1000 -s /bin/bash --no-create-home
 ---> Using cache
 ---> 04412df99deb
Step 3/8 : RUN mkdir /build && chown docker:docker /build
 ---> Using cache
 ---> 026e65fcfad1
Step 4/8 : RUN apt-get update && apt-get install -y   make unrar-free autoconf automake libtool gcc g++ gperf   flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial   sed git unzip bash help2man wget bzip2 libtool-bin
 ---> Using cache
 ---> 5b8cb581c5ad
Step 5/8 : RUN su docker -c "     git clone --recursive https://github.com/pfalcon/esp-open-sdk.git /build/esp-open-sdk ;     cd /build/esp-open-sdk ;     make STANDALONE=n ; "
 ---> Using cache
 ---> aed6552d0f62
Step 6/8 : RUN apt-get update && apt-get install -y make python python-serial
 ---> Using cache
 ---> 333bb30a6d43
Step 7/8 : COPY --from=builder /build/esp-open-sdk/xtensa-lx106-elf /opt/xtensa-lx106-elf
invalid from flag value builder: pull access denied for builder, repository does not exist or may require 'docker login'

helgard999, 2019-07-10:
Hi found my problem..
"FROM ubuntu:16.04 as builder" was not in the docker files....
thanks for the help

closed 2019-07-10 by helgard999