maximkulkin/esp-homekit-demo #232: Window curtain Core Panic

konifer44, 2019-07-09:
Hey, Im running Window Curtain program on ESP32.
Everything seems to work until I push open/close/choose position on slider on my iPhone. Then:

>>> HomeKit: [Client 56] Update Characteristics
Update target position to: 100
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr 
Core 1 register dump:
PC      : 0x40081e24  PS      : 0x00000016  A0      : 0x40080306  A1      : 0x3ffc6140  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x4008c40b  A5      : 0x3ffbc9c0  
A6      : 0x00050023  A7      : 0x3ffb40d4  A8      : 0x00000000  A9      : 0x3ffb0c60  
A10     : 0x00000001  A11     : 0x00060f23  A12     : 0x8008aa94  A13     : 0x3ffbc9b0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001f  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 37e5580a31a43eb64f3edd1b7adbc38d27ac18c051101f530d599770fe9b7e80

Backtrace: 0x40081e21:0x3ffc6140 0x40080303:0x03649c50 |<-CORRUPTED

Rebooting...

Any ideas ?

maximkulkin, 2019-07-10:
Feels like either memory corruption or hardware problem. Can you post your code?

konifer44, 2019-07-11:
I have two cases
Case 1

This is Your version of code, I modify only things to run it on ESP32. When I click open on iPhone everything seems to be working, Im printing to console status opening, but after few seconds when the curtain accessory is not responding to iPhone. Its working but not responding. When reach 100 - full open its happening this:

"
>>> HomeKit: [Client 56] Update Characteristics
Update target position to: 100
position 1, target 100
.
.
position 99, target 100
position 100, target 100
reached destination 100
Guru Meditation Error: Core  1 panic'ed (Unhandled kernel exception)
Core 1 register dump:
PC      : 0x40090077  PS      : 0x00000d1c  A0      : 0x40080000  A1      : 0x3ffc62f0  
A2      : 0x00000000  A3      : 0x40105d8c  A4      : 0x40090057  A5      : 0x3ffbc980  
A6      : 0x00050023  A7      : 0x3ffb40d8  A8      : 0x00000000  A9      : 0x3ffb0c60  
A10     : 0x00000001  A11     : 0x00060f23  A12     : 0x8008e6e0  A13     : 0x3ffbc970  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001f  EXCCAUSE: 0x00000003  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 2065a2c7b05747b965a9547d42f7b1de77731d34ca258a8d6fed47b5a5d44d02

Backtrace: 0x40090074:0x3ffc62f0 0x4007fffd:0x00000000 |<-CORRUPTED

Rebooting..."

The code: 

`#include "string.h"
#include <esp_event_loop.h>
#include <esp_log.h>
#include <esp_wifi.h>
#include <nvs_flash.h>
#include <stdio.h>
#include <stdlib.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

#include "wifi.h"
#include <homekit/characteristics.h>
#include <homekit/homekit.h>

#include "driver/mcpwm.h"
#include "driver/pcnt.h"
#include "esp_attr.h"
#include "freertos/queue.h"
#include "soc/mcpwm_periph.h"
#include "soc/rtc.h"

#define MAX(x, y) (((x) > (y)) ? (x) : (y))
#define MIN(x, y) (((x) < (y)) ? (x) : (y))
#define POSITION_OPEN 100
#define POSITION_CLOSED 0
#define POSITION_STATE_CLOSING 0
#define POSITION_STATE_OPENING 1
#define POSITION_STATE_STOPPED 2
#define ESP_INTR_FLAG_DEFAULT 0
TaskHandle_t updateStateTask;
homekit_characteristic_t current_position;
homekit_characteristic_t target_position;
homekit_characteristic_t position_state;
homekit_accessory_t *accessories[];

void update_state() {
     while (true) {
        uint8_t position = current_position.value.int_value;
        int8_t direction = position_state.value.int_value == POSITION_STATE_OPENING ? 1 : -1;
[... 122 lines cut ...]
        update_state();
        vTaskResume(updateStateTask);
    }
}

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};
void on_wifi_ready() { homekit_server_init(&config); }

void app_main(void)
{
    // Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    wifi_init();
    update_state_init();
}`

konifer44, 2019-07-11:
I have two cases.

1. I modify code to run it on ESP32. I changed only part with init of WiFi. 
the rest of code is the same. 
Im thinking there is an issue with 'vTaskResume(updateStateTask);'

There is a code: 
'#include "string.h"
#include <esp_event_loop.h>
#include <esp_log.h>
#include <esp_wifi.h>
#include <nvs_flash.h>
#include <stdio.h>
#include <stdlib.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

#include "wifi.h"
#include <homekit/characteristics.h>
#include <homekit/homekit.h>

#include "driver/mcpwm.h"
#include "driver/pcnt.h"
#include "esp_attr.h"
#include "freertos/queue.h"
#include "soc/mcpwm_periph.h"
#include "soc/rtc.h"

#define MAX(x, y) (((x) > (y)) ? (x) : (y))
#define MIN(x, y) (((x) < (y)) ? (x) : (y))
#define POSITION_OPEN 100
#define POSITION_CLOSED 0
#define POSITION_STATE_CLOSING 0
#define POSITION_STATE_OPENING 1
#define POSITION_STATE_STOPPED 2
#define ESP_INTR_FLAG_DEFAULT 0
TaskHandle_t updateStateTask;
homekit_characteristic_t current_position;
homekit_characteristic_t target_position;
homekit_characteristic_t position_state;
homekit_accessory_t *accessories[];

void update_state() {
     while (true) {
        uint8_t position = current_position.value.int_value;
        int8_t direction = position_state.value.int_value == POSITION_STATE_OPENING ? 1 : -1;
        int16_t newPosition = position + direction;

        printf("position %u, target %u\n", newPosition, target_position.value.int_value);

        current_position.value.int_value = newPosition;
        homekit_characteristic_notify(&current_position, current_position.value);

        if (newPosition == target_position.value.int_value) {
            printf("reached destination %u\n", newPosition);
            position_state.value.int_value = POSITION_STATE_STOPPED;
            homekit_characteristic_notify(&position_state, position_state.value);
            vTaskSuspend(updateStateTask);
            return;
        }

        vTaskDelay(pdMS_TO_TICKS(100));
    }
    
}

void on_wifi_ready();
esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch (event->event_id)
[... 97 lines cut ...]
        homekit_characteristic_notify(&position_state, position_state.value);
        vTaskResume(updateStateTask);
    }
}

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};
void on_wifi_ready() { homekit_server_init(&config); }

void app_main(void)
{
    // Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    wifi_init();
    update_state_init();
}'

konifer44, 2019-07-11:
I suppose there is some error with "vTaskResume(updateStateTask);"
Now I Have this error: 

'>>> HomeKit: [Client 59] Update Characteristics
Update target position to: 100
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x3ffc6130  PS      : 0x00050033  A0      : 0x3ffc6130  A1      : 0x3ffc6120  
A2      : 0x40137630  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x400d296c  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000014  
EXCVADDR: 0x3ffc6130  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 89d1c312e8ee672d5c9c8a63a9eb0575921262ad963946d2649cb39aa461e9ef

Backtrace: 0x3ffc612d:0x3ffc6120 |<-CORRUPTED

Rebooting...
ets Jun  8 2016 00:22:57'

Code:

[Code1.pdf](https://github.com/maximkulkin/esp-homekit-demo/files/3383376/Code1.pdf)

konifer44, 2019-07-11:
I investigate and change some code to try something else like this:

Im not using
'vTaskResume(updateStateTask);'
but I just call function
'updateState()'

Like this:

'void on_update_target_position(homekit_characteristic_t *ch, homekit_value_t value, void *context)
{
    printf("Update target position to: %u\n", target_position.value.int_value);
    if (target_position.value.int_value == current_position.value.int_value)
    {
        printf("Current position equal to target. Stopping.\n");
        position_state.value.int_value = POSITION_STATE_STOPPED;
        homekit_characteristic_notify(&position_state, position_state.value);
        vTaskSuspend(updateStateTask);
    }
    else
    {
        position_state.value.int_value = target_position.value.int_value > current_position.value.int_value
                                             ? POSITION_STATE_OPENING
                                             : POSITION_STATE_CLOSING;

        homekit_characteristic_notify(&position_state, position_state.value);
        **update_state();
        //vTaskResume(updateStateTask);**
    }
}'

'
void update_state()
{
    while (true)
    {
        uint8_t position = current_position.value.int_value;
        int8_t direction = position_state.value.int_value == POSITION_STATE_OPENING ? 1 : -1;
        int16_t newPosition = position + direction;

        printf("position %u, target %u\n", newPosition, target_position.value.int_value);

        current_position.value.int_value = newPosition;
        homekit_characteristic_notify(&current_position, current_position.value);

        if (newPosition == target_position.value.int_value)
        {
            printf("reached destination %u\n", newPosition);
            position_state.value.int_value = POSITION_STATE_STOPPED;
            homekit_characteristic_notify(&position_state, position_state.value);
            **//vTaskSuspend(updateStateTask);
            return;**
        }

        vTaskDelay(pdMS_TO_TICKS(100));
    }
}
'

Everything seems to be working, position is updating, im printing everything into console but accessory starts to be  unresponsive
on iPhone:
'
>>> HomeKit: [Client 58] Update Characteristics
Update target position to: 100
position 1, target 100
.
.
.
position 99, target 100
reached destination 100
>>> HomeKit: [Client 58] Closing client connection
I (121540) wifi: state: run -> init (7c0)
I (121540) wifi: pm stop, total sleep time: 63011726 us / 78434865 us

I (121540) wifi: new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
STA disconnected
'

konifer44, 2019-07-11:
I investigate and change some code to try something else like this:

Im not using
`**vTaskResume(updateStateTask);**`
but I just call function
`**updateState()**`

Like this:

```
void on_update_target_position(homekit_characteristic_t *ch, homekit_value_t value, void *context)
{
    printf("Update target position to: %u\n", target_position.value.int_value);
    if (target_position.value.int_value == current_position.value.int_value)
    {
        printf("Current position equal to target. Stopping.\n");
        position_state.value.int_value = POSITION_STATE_STOPPED;
        homekit_characteristic_notify(&position_state, position_state.value);
        //vTaskSuspend(updateStateTask);
    }
    else
    {
        position_state.value.int_value = target_position.value.int_value > current_position.value.int_value
                                             ? POSITION_STATE_OPENING
                                             : POSITION_STATE_CLOSING;

        homekit_characteristic_notify(&position_state, position_state.value);
        **update_state();**
        //vTaskResume(updateStateTask);
    }
}

void update_state()
{
    while (true)
    {
        uint8_t position = current_position.value.int_value;
        int8_t direction = position_state.value.int_value == POSITION_STATE_OPENING ? 1 : -1;
        int16_t newPosition = position + direction;

        printf("position %u, target %u\n", newPosition, target_position.value.int_value);

        current_position.value.int_value = newPosition;
        homekit_characteristic_notify(&current_position, current_position.value);

        if (newPosition == target_position.value.int_value)
        {
            printf("reached destination %u\n", newPosition);
            position_state.value.int_value = POSITION_STATE_STOPPED;
            homekit_characteristic_notify(&position_state, position_state.value);
          //vTaskSuspend(updateStateTask);
            **return;**
        }

        vTaskDelay(pdMS_TO_TICKS(100));
    }
}

```

Everything seems to be working, position is updating, im printing everything into console but accessory starts to be  unresponsive
on iPhone:
```
>>> HomeKit: [Client 58] Update Characteristics
Update target position to: 100
position 1, target 100
.
.
.
position 99, target 100
reached destination 100
>>> HomeKit: [Client 58] Closing client connection
I (121540) wifi: state: run -> init (7c0)
I (121540) wifi: pm stop, total sleep time: 63011726 us / 78434865 us

I (121540) wifi: new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
STA disconnected

```

skanico, 2019-07-11:
Maybe have a look at [this](https://github.com/maximkulkin/esp-homekit-demo/pull/174#issuecomment-504002146)

skanico, 2019-07-11:
If you want, I made something for my 5 blinds with ESP32 (code is not nice, because I made that a few days before going on vacation... it needs some tuning).

It also works with physical buttons.

[Archive.zip](https://github.com/maximkulkin/esp-homekit-demo/files/3383883/Archive.zip)

skanico, 2019-07-11:
If you want to keep your code, you should replace 256 by a higher value (2048?) in xTaskCreate.
This is a difference between ESP32 and ESP8266

konifer44, 2019-07-11:
Hey guys I probably found one mistake why accessory is being unresponsive.  This code while update position is notify HomeKit every 100ms, HomeKit is starting to be unresponsive because is receiving notifications too fast. Changing time for example to 1000ms doesn't help.

```
void update_state()
{
    while (true)
    {
        uint8_t position = current_position.value.int_value;
        int8_t direction = position_state.value.int_value == POSITION_STATE_OPENING ? 1 : -1;
        int16_t newPosition = position + direction;

        printf("position %u, target %u\n", newPosition, target_position.value.int_value);

        current_position.value.int_value = newPosition;
        homekit_characteristic_notify(&current_position, current_position.value);

        if (newPosition == target_position.value.int_value)
        {
            printf("reached destination %u\n", newPosition);
            position_state.value.int_value = POSITION_STATE_STOPPED;
            homekit_characteristic_notify(&position_state, position_state.value);
          //vTaskSuspend(updateStateTask);
            return;
        }

        vTaskDelay(pdMS_TO_TICKS(100));
    }
}
```

This is working, but is not actually accurate because it notify only when reach destination.

```
void update_state()
{
    while (true)
    {
        uint8_t position = current_position.value.int_value;
        int8_t direction = position_state.value.int_value == POSITION_STATE_OPENING ? 1 : -1;
        int16_t newPosition = position + direction;

        printf("position %u, target %u\n", newPosition, target_position.value.int_value);

        current_position.value.int_value = newPosition;
        

        if (newPosition == target_position.value.int_value)
        {
            printf("reached destination %u\n", newPosition);
            position_state.value.int_value = POSITION_STATE_STOPPED;
            homekit_characteristic_notify(&position_state, position_state.value);
       

            homekit_characteristic_notify(&current_position, current_position.value);            return;
        }

        vTaskDelay(pdMS_TO_TICKS(100));
    }
}
```

konifer44, 2019-07-11:
> If you want to keep your code, you should replace 256 by a higher value (2048?) in xTaskCreate.
> This is a difference between ESP32 and ESP8266

YOU ARE GENIUS  :D it's working

konifer44, 2019-07-11:
Thanks a lot! Closing.

closed 2019-07-11 by konifer44