maximkulkin/esp-homekit-demo #233: Easy Notify

kiralikbeyin, 2019-07-09:
@maximkulkin 
I am trying dynamic_services example.

How can i use a function like 
```
void notify (int relayid, bool position) {
...?

 }
````
so i can call notfiy easliy?

maximkulkin, 2019-07-09:
Not sure I understand the question.

kiralikbeyin, 2019-07-10:
I was trying to manually trigger accessory to on-off with a function;

note that i added 
```
  c->value = new_value;
  c->value.bool_value = on;
  // I think both works
````
because ```homekit_characteristic_notify```is not enough if your iPad or iPhone is turned off. It only notfiy if your HomeKit screen is open.

````
void relay_write_fbin(int relay, bool on) {
  printf("relay_write_fbin %d %s\n", relay, on ? "ON" : "OFF");
  gpio_write(relay, on ? 1 : 0);
  homekit_value_t new_value = HOMEKIT_BOOL(on);
  printf("relay_write_fbin Notifying homekit that current door state is now '%s'\n", on ? "ON" : "OFF");
  homekit_accessory_t *accessory = accessories[0];
  homekit_service_t *service = accessory->services[1 + yeni];
  homekit_characteristic_t *c = service->characteristics[1];
  assert(c);
  c->value = new_value;
  c->value.bool_value = on;

  homekit_characteristic_notify(c, new_value);
}
````

Thank you for reply.

closed 2019-07-10 by kiralikbeyin

maximkulkin, 2019-07-10:
`homekit_cahracteristic_notify()` is to notify all connected controllers about the new state. If you will do just that, controllers will think that you have new state, while your internal state will remain the same. When controller gets disconnected and reconnected back, it will query full information and thus old state that you have internally.

Also, you do not need to reset it twice:
```c
c->value = HOMEKIT_BOOL_(on);
homekit_characteristic_notify(c, c->value);
```