maximkulkin/esp-homekit-demo #234: Garage Example Relay Config

jsmith79, 2019-07-09:
Hi Guys,

I am hoping that someone may be able to offer some advice.

I would like to be able to change the operation of the relay in the garage example.
The example has the RELAY_PIN pulled high (3.3v) all the time and when clicking on the garage icon in homekit the RELAY_PIN is pulled low for a second the high again.

I would like this to be the opposite. RELAY_PIN floating or pulled down (D8 on wemosD1mini is pulldown) and when triggered pulled high.

Any advice would be appreciated, i am quite new to this.

Thanks.

maximkulkin, 2019-07-09:
Change [this](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/garage/garage.c#L106-L108) to this
```c
void relay_write(bool on) {
    gpio_write(RELAY_PIN, on ? 1 : 0);
}
```
On the other hand, you could change the meaning of true and false for relay (swap all uses of true and false to set state for the relay)

jsmith79, 2019-07-10:
Thanks @maximkulkin , worked perfectly.

closed 2019-07-10 by maximkulkin