maximkulkin/esp-homekit-demo #236: Installation issue - serial port

helgard999, 2019-07-12:
Hi 
Please point me in the right direction.
For some reason the esptool.py does not see the correct serial port on my machine, when it is run from the docker container. The environment values are set as per installation instructions. 

env                                                   

TERM_SESSION_ID=w0t1p0:18B5F6D2-2ECE-496A-A1D2-722CEAB8E2E1
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.UyMEcsjGc2/Listeners
Apple_PubSub_Socket_Render=/private/tmp/com.apple.launchd.Mg46Ua2e0X/Render
COLORFGBG=15;0
ITERM_PROFILE=dark
XPC_FLAGS=0x0
PWD=/Users/hr/esp/esp-homekit-demo
SHELL=/bin/zsh
TERM_PROGRAM_VERSION=3.2.9
TERM_PROGRAM=iTerm.app
PATH=/Library/Frameworks/Python.framework/Versions/3.7/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
COLORTERM=truecolor
TERM=xterm-256color
HOME=/Users/hr
TMPDIR=/var/folders/0q/tnjlzdrn2pb9vxr364yt4fmr0000gn/T/
USER=hr
XPC_SERVICE_NAME=0
LOGNAME=hr
__CF_USER_TEXT_ENCODING=0x0:0:0
ITERM_SESSION_ID=w0t1p0:18B5F6D2-2ECE-496A-A1D2-722CEAB8E2E1
SHLVL=1
OLDPWD=/Users/hr
ZSH=/Users/hr/.oh-my-zsh
SDK_PATH=/Users/hr/esp/esp-open-rtos
ESPTOOL_PORT=/dev/tty.SLAB_USBtoUART
ESPTOOL_BAUDPORT=115200
ESPPORT=/dev/tty.SLAB_USBtoUART
PAGER=less
LESS=-R
LSCOLORS=Gxfxcxdxbxegedabagacad
LC_CTYPE=UTF-8
_=/usr/bin/env

The variables seems to be correct?
Now I run only the docker part, without any complaints.

 ~/esp/esp-homekit-demo   master ● ?  docker run -i -t --rm -v `pwd`:/project -w /project esp-rtos

The project compiles and a bin file is created

The next part is a problem

root@05b06632ae61:/project# make -C examples//sonoff_basic flash monitor
make: Entering directory '/project/examples/sonoff_basic'
esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash -fs 8m -fm dout -ff 40m \
	0x0 /opt/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /opt/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/sonoff_basic.bin
esptool.py v1.2
Traceback (most recent call last):
  File "/opt/xtensa-lx106-elf/bin/esptool.py", line 1270, in <module>
    main()
  File "/opt/xtensa-lx106-elf/bin/esptool.py", line 1179, in main
    esp = ESPROM(args.port, initial_baud)
  File "/opt/xtensa-lx106-elf/bin/esptool.py", line 71, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/lib/python2.7/dist-packages/serial/__init__.py", line 85, in serial_for_url
    instance.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 294, in open
    raise SerialException(msg.errno, "could not open port %s: %s" % (self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
/opt/esp-open-rtos/common.mk:247: recipe for target 'flash' failed
make: *** [flash] Error 1
make: Leaving directory '/project/examples/sonoff_basic'

Any ideas?

Regards
HR

maximkulkin, 2019-07-12:
On macOS Docker does not yet support passing devices into container (thing you would normally do with `--device=/dev/ttyUSB0`, so sadly that part should be done on host machine.
1. Make sure you have esptool.py installed
    ```shell
    pip install esptool
    ```
1. Export variables:
    ```shell
    export SDK_PATH=<path-to-esp-open-rtos>
    export ESPPORT=/dev/ttyUSB0
    ```
1. Run flashing/monitoring:
    ```shell
    make flash monitor
    ```

helgard999, 2019-07-14:
It make sense I was running your docker helper function...

but there is still a problem, must also be some variable that is not set correctly. 
how do I see which arguments is sent to "make"?
make -C examples/sonoff_basic flash   
esptool.py -p /dev/tty.SLAB_USBtoUART --baud 115200 write_flash -fs 8m -fm dout -ff 40m \
		0x0 /Users/hr/esp/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /Users/hr/esp/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/sonoff_basic.bin
make: esptool.py: No such file or directory

/Users/hr/esp/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin     <=this is correct 
c/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin <= this is also ./firmware/sonoff_basic.bin    <= this is not
there needs to be sone environment variable here
"/Users/hr/esp/esp-homekit-demo/examples/sonoff_basic"/firmware/sonoff_basic.bin
=`pwd` + `examples/sonoff_basic`

maximkulkin, 2019-07-14:
1. `pip install esptool`
2. The path is correct since `-C examples/sonoff_basic` changes Make directory to examples/sonoff_basic, so ./firmware/sonoff_basic.bin is relative to that directory, which is correct.

helgard999, 2019-07-24:
Thanks for your help and patience.

I upgraded to Python 3 and pip3 on the Mac, from there on everything worked as explained.

Don't you want to consider changing your note about the docker command since it does not work as explained in the macOS environment.

closed 2019-07-24 by helgard999

maximkulkin, 2019-07-24:
@helgard999 which note in particular does not work as explained?