maximkulkin/esp-homekit-demo #238: Task stack overflow name="Tmr Svc"

NN708, 2019-07-15:
I'm trying to create an air conditioner (category 21), but when the server is starting, the following error occurred:
```
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x100000
>>> HomeKit: Generated new accessory ID: xx:xx:xx:xx:xx:xx
Task stack overflow (high water mark=0 name="Tmr Svc")
Task stack overflow (high water mark=0 name="Tmr Svc")
>>> wifi_config: Stopping HTTP server
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
Task stack overflow (high water mark=0 name="Tmr Svc")
Task stack overflow (high water mark=0 name="Tmr Svc")
mDNS announcement: Name=xxxx-xxxx 
                                   md=xxxxxpv=1.0id=xx:xx:xx:xx:xx:xxc#
ets Jan  8 2013,rst cause:2, boot mode:(1,6)

ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

I tried to increase the "HomeKit Server" task stack size, but the error still occurred. Then I used a very similar code to create a thermostat, and it worked well.

Can anyone help me with this problem?

maximkulkin, 2019-07-15:
Well, there are no fortune tellers here, you have to show the code.

NN708, 2019-07-16:
Hi, @maximkulkin !

Here’s my code that create the air conditioner accessory: **[air_conditioner.c](https://github.com/NN708/ESP-Homekit-IRHub/blob/21e7bb1c9a398af3f270648f9716384e59d09aaa/ir_hub/accessories/air_conditioner.c)**

Two functions, homekit_add_service() and homekit_add_characteristic() are used to add services and characteristics dynamically. They are defined in **[homekit_utility.c](https://github.com/NN708/ESP-Homekit-IRHub/blob/21e7bb1c9a398af3f270648f9716384e59d09aaa/ir_hub/utilities/homekit_utility.c)**.

maximkulkin, 2019-07-16:
@NN708 Oh, man, I like your module design. I assume there are also LEDs pointing down (underneath the board). Also, I like the neatness of your code.

Ok, I've checked your code and I can see the problem: when you use HOMEKIT_* macros, what happens is a structure is allocated ON STACK and then you get a pointer to it.. Thus, if you store that pointer, it will be pointer to stack and thus the data will be messed up. These macros are very useful for statically initialized data, because in that case those structures are allocated in "static memory". That's why I added [homekit_*_clone() functions](https://github.com/maximkulkin/esp-homekit/blob/master/include/homekit/types.h#L282-L291) and [NEW_HOMEKIT_* macros](https://github.com/maximkulkin/esp-homekit/blob/master/include/homekit/types.h#L294-L309) - they take a pointer to a structure and create a copy that is allocated on the heap. You can find example of their usage in [examples/dynamic_services](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/dynamic_services/main.c#L114-L139). You should either use those to pass a pointer to homekit_add_*() functions or you clone them inside those functions.

maximkulkin, 2019-07-16:
Ok, my guess is: all structures that should be allocated on stack are allocated at once when you enter your function thus you need a lot of stack space. It might be a good idea to declare services in static memory and then just clone them into an accessory:
```c
static homekit_service_t *accessory_info = HOMEKIT_SERVICE_(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
    HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
    HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
    HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF19D"),
    HOMEKIT_CHARACTERISTIC(MODEL, "MyLED"),
    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
    HOMEKIT_CHARACTERISTIC(IDENTIFY, led_identify),
    NULL
});
// ...
void air_conditioner_init(homekit_accessory_t* accessory) {
    accessory->category = homekit_accessory_category_air_conditioner;
    homekit_add_service(accessory, homekit_service_clone(accessory_info));
    // ...
}
```

If you really wan't to preserve that beauty, you might want consider inside on reaction to wifi event spawning a task with your own stack size to configure everything.

NN708, 2019-07-17:
I defined the services globally and it worked. Thank you!

closed 2019-07-17 by NN708

kiralikbeyin, 2019-08-22:
for your info

if i use 
````
      vTaskDelay(1000 / portTICK_PERIOD_MS);
````
before ```homekit_server_init(&config);``` makes reset..

Timer Start Failed
Timer Stop Failed

remove vTaskDelay