maximkulkin/esp-homekit-demo #241: Blinds

AchimPieters, 2019-07-20:
Hi Maxim,

I have a problem with my blinds setup, I used the  [sonoff_dual_blinds](https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/sonoff_dual_blinds) example to build upon. So after compiling everything is working correctly.

After adding my blinds to HomeKit the motor (see scheme below) turns left and right when I open and close the blinds. ( from 0 to 100 in 15 seconds / 100 to 0 in 15 seconds).

* When the Blinds are closed and I press the Up button nothing happens.
* When the Blinds are closed and I press the Down button nothing happens.
* When the Blinds are open and I press the Up button nothing happens.
* When the Blinds are open and I press the Down button it closes the blinds.

I can't figure out why the Up button does not work? and the 'Button' button should it not stop the up and down movement when pressed?

Maybe I have looked at it for too long and can't see the solution anymore? :weary:

### Scheme
[image]

### Code

```
#include <stdio.h>
#include <stdlib.h>
//#include <espressif/esp_wifi.h>
//#include <espressif/esp_sta.h>>
#include <espressif/esp_common.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <wifi_config.h>

#include "button.h"

#define MAX(x, y) (((x) > (y)) ? (x) : (y))
#define MIN(x, y) (((x) < (y)) ? (x) : (y))

#define POSITION_OPEN 100
#define POSITION_CLOSED 0
#define POSITION_STATE_CLOSING 0
#define POSITION_STATE_OPENING 1
#define POSITION_STATE_STOPPED 2

// number of seconds the blinds take to move from fully open to fully closed position
#define SECONDS_FROM_CLOSED_TO_OPEN 15

TaskHandle_t updateStateTask;
homekit_characteristic_t current_position;
homekit_characteristic_t target_position;
homekit_characteristic_t position_state;
homekit_accessory_t *accessories[];

// The GPIO pin that is connected to the relay on the Sonoff Dual R2
const int relay0_gpio = 12;
const int relay1_gpio = 13;
// The GPIO pin that is connected to the LED on the Sonoff Dual R2
const int led_gpio = 2;
// The GPIO pin that is oconnected to the button on the Sonoff Dual R2
const int BUTTON_GPIO = 4;
// The GPIO pin that is oconnected to the BUTTON1 header pin
const int BUTTON1_GPIO = 16;
// The GPIO pin that is oconnected to the BUTTON0 header pin
const int BUTTON0_GPIO = 14;

const int button_up = 16;
const int button_down = 14;
const int relay_up = 12;
[... 235 lines cut ...]
    wifi_config_init("blinds", NULL, on_wifi_ready);
    update_state_init();

    button_config_t button_config = BUTTON_CONFIG(
        .active_level=button_active_low,
    );

    int r = button_create(BUTTON_GPIO, button_config, button_callback, NULL);
    if (r) {
        printf("Failed to initalize button (code %d)\n", r);
    }

    int u = button_create(BUTTON0_GPIO, button_config, button_down_callback, NULL);
    if (u) {
        printf("Failed to initalize button (code %d)\n", r);
    }

    int d = button_create(BUTTON1_GPIO, button_config, button_up_callback, NULL);
    if (d) {
        printf("Failed to initalize button (code %d)\n", r);
    }

}

```

maximkulkin, 2019-07-22:
According to [this invaluable reference](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/) it looks like GPIO16 does not have an interrupt support. That's why your UP button (which is connected to it) does not work. I recommend using the other unused pin - GPIO5 for that.

A few suggestions on the code:
1. You have button_up constant and BUTTON1_GPIO define both specifying pin number for UP button. In one place you use button_up, in other - BUTTON1_GPIO. I suggest getting rid of BUTTON1_GPIO and using a more descriptive button_up constant. Same applies to DOWN and BUTTON button.
2. You can simplify relay_write like this:
    ```c
    void relays_write(int state) {
        gpio_write(relay_up, state == POSITION_STATE_OPENING);
        gpio_write(relay_down, state == POSITION_STATE_CLOSING);
    }
    ```
3. Instead of doing this:
    ```c
    int8_t direction = position_state.value.int_value == POSITION_STATE_OPENING ? 1 : -1;
    ```
    I would do this:
    ```c
    int8_t direction = current_position.value.int_value < target_position.value.int_value ? 1 : -1;
    ```
    to prevent runaway. In general, I would just use current_position & target_position to control movements and would derive position_state based on the other two.

AchimPieters, 2019-07-23:
Snap, I forgot about the gpio16 interrupt support.
I will look in to your code suggestions.

Thank you so much!

AchimPieters, 2019-08-02:
@maximkulkin 

It works! thank you so much, now I have to figure out how to implement this in a real life setup for my curtains / Blinds. 

[image]

Here's my working code:
```
#include <stdio.h>
#include <stdlib.h>
#include <espressif/esp_common.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>
#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <wifi_config.h>
#include "button.h"

#define MAX(x, y) (((x) > (y)) ? (x) : (y))
#define MIN(x, y) (((x) < (y)) ? (x) : (y))

#define POSITION_OPEN 100
#define POSITION_CLOSED 0
#define POSITION_STATE_CLOSING 0
#define POSITION_STATE_OPENING 1
#define POSITION_STATE_STOPPED 2

// number of seconds the blinds take to move from fully open to fully closed position
#define SECONDS_FROM_CLOSED_TO_OPEN 15

TaskHandle_t updateStateTask;
homekit_characteristic_t current_position;
homekit_characteristic_t target_position;
homekit_characteristic_t position_state;
homekit_accessory_t *accessories[];

const int relay0_gpio = 12;
const int relay1_gpio = 13;
const int led_gpio = 2;
const int button_gpio = 4;
const int button1_gpio = 5;
const int button0_gpio = 14;

const int button_up = 5;
const int button_down = 14;
const int relay_up = 12;
const int relay_down = 13;

void target_position_changed();

void relay_write(int relay, bool on) {
        gpio_write(relay, on ? 1 : 0);
}

void led_write(bool on) {
        gpio_write(led_gpio, on ? 0 : 1);
}

void relays_write(int state) {
        gpio_write(relay_up, state == POSITION_STATE_OPENING);
        gpio_write(relay_down, state == POSITION_STATE_CLOSING);
}

void reset_configuration_task() {
        for (int i=0; i<3; i++) {
                led_write(true);
                vTaskDelay(100 / portTICK_PERIOD_MS);
                led_write(false);
[... 171 lines cut ...]
        .accessories = accessories,
        .password = "345-58-410",
        .setupId="0LK7",
};

void on_wifi_ready() {
        homekit_server_init(&config);
}

void user_init(void) {
        uart_set_baud(0, 115200);

        gpio_init();

        wifi_config_init("blinds", NULL, on_wifi_ready);
        update_state_init();

        if (button_create(button_up, 0, 1000, button_up_callback)) {
                printf("Failed to initialize button up\n");
        }
        if (button_create(button_down, 0, 1000, button_down_callback)) {
                printf("Failed to initialize button down\n");
        }
}
```

closed 2019-08-07 by AchimPieters