maximkulkin/esp-homekit-demo #242: No response after reboot ESP32

syu-source, 2019-07-23:
I am working on ESP32. 

First run, the homekit-demo works fine with Apple Homekit, I can find my LED and add, control LED with Siri. 
But after ESP32 reboot, iPhone homekit shows Device not responding, and I get log as below after #define HOMEKIT_DEBUG 1

>>> HomeKit: Starting server
>>> HomeKit: Set accessory ID: 49:80:E1:5A:50:B7
>>> HomeKit: Generated new accessory key
>>> HomeKit: Found admin pairing with 2A91DF28-B3BE-4D27-B74E-687BDAE0B891, disabling pair setup
>>> HomeKit: Configuring mDNS
>>> homekit_run_server: Staring HTTP server

========================Following log repeats when trying iphone homekit app=========

>>> HomeKit: Got new client connection: 57
>>> homekit_client_process: [Client 57] Got 159 incomming data
>>> homekit_server_on_pair_verify: HomeKit Pair Verify
>>> homekit_server_on_pair_verify: Free heap: 81256
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> tlv_debug: Type 3 value (32 bytes): \x80\x9Cdu\xFA(\xA8\xA4\x10s\xA5\x9F6\x12i\x12\xB2\x98\x82\xC7\x97\xD1\xD0Zbd+m\xC59\x02T
>>> HomeKit: [Client 57] Pair Verify Step 1/2
>>> homekit_server_on_pair_verify: [Client 57] Importing device Curve25519 public key
>>> homekit_server_on_pair_verify: [Client 57] Generating accessory Curve25519 key
>>> homekit_server_on_pair_verify: [Client 57] Exporting accessory Curve25519 public key
>>> homekit_server_on_pair_verify: [Client 57] Generating Curve25519 shared secret
>>> homekit_server_on_pair_verify: [Client 57] Generating signature
>>> homekit_server_on_pair_verify: [Client 57] Generating proof
>>> homekit_server_on_pair_verify: [Client 57] Encrypting response
>>> send_tlv_response: [Client 57] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> tlv_debug: Type 3 value (32 bytes): \x1F\xA9\xAE6\xD1\x88\xAE\xD3[x\xFD\x9FU\xEC\x96\x97\x18\xD9\xD7\x1B%p\xBCz\x96\xDE\x9E\xBAB7\x9C\x1A
>>> tlv_debug: Type 5 value (101 bytes): \x85\xC2n\x0A\xCF\xE5\x0C\xCFp\x0A\x1F0i\x17k\x15\xCF\x03\xF3\x11\xE6\x11\xFA\xD0U\xFF\xE7\x92UH\x8F\x9DSR[\x9Ejb\x9F\xB6Y\xA0\xF6p0\xB8\xCE3O\xE9\x0Cp)Ec\x83\xA4\x9Fx\xF9\x9E\xB7\xC5\x93-\x05\x1B\xBF\x8FP\x8F\x9F\x18\x1Dm\xC8\x14\xF9ls\x801h\xFD>\x0B\x04\x85U\xB1\xC4\xC9\xB1[S\xE7X\xD8lC\xAC
>>> client_send: [Client 57] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x03 \x1F\xA9\xAE6\xD1\x88\xAE\xD3[x\xFD\x9FU\xEC\x96\x97\x18\xD9\xD7\x1B%p\xBCz\x96\xDE\x9E\xBAB7\x9C\x1A\x05e\x85\xC2n\x0A\xCF\xE5\x0C\xCFp\x0A\x1F0i\x17k\x15\xCF\x03\xF3\x11\xE6\x11\xFA\xD0U\xFF\xE7\x92UH\x8F\x9DSR[\x9Ejb\x9F\xB6Y\xA0\xF6p0\xB8\xCE3O\xE9\x0Cp)Ec\x83\xA4\x9Fx\xF9\x9E\xB7\xC5\x93-\x05\x1B\xBF\x8FP\x8F\x9F\x18\x1Dm\xC8\x14\xF9ls\x801h\xFD>\x0B\x04\x85U\xB1\xC4\xC9\xB1[S\xE7X\xD8lC\xAC
>>> homekit_client_process: [Client 57] Finished processing
>>> HomeKit: [Client 57] Closing client connection

maximkulkin, 2019-07-23:
Can you try changing HOMEKIT_SPI_FLASH_BASE_ADDR to a different value, reflash and re-pair. It looks like pairing keys got messed up.

syu-source, 2019-07-31:
it works. Thanks.

closed 2019-07-31 by syu-source