maximkulkin/esp-homekit-demo #245: OTA question

kiralikbeyin, 2019-08-03:
https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/http_client_ota/http_get.c

http_client_ota Example works with local ip address..

I tried to put blink.bin file on a firebase hosting also to a github repo but couldnt update it..

Is it possible to update a very important release like http_client_ota (i can re-setup homekit settings its ok)

Tried these only local ip worked..

````
#define BINARY_PATH "/blink.bin"
#define BINARY_PATH "https://testxxx.firebaseapp.com/blink.bin"
#define BINARY_PATH "https://github.com/kiralikbeyin/html/raw/master/blink.bin"

#define SHA256_PATH NULL //bypass sha256

#define SERVER "192.168.43.107" //local ip works fine
#define SERVER "140.82.118.4" //github 
#define SERVER "testxxx.web.app"
#define SERVER "151.101.65.195" //firebase
````

AchimPieters, 2019-08-04:
Maybe this is a solution for you? https://github.com/HomeACcessoryKid/life-cycle-manager

kiralikbeyin, 2019-08-04:
@AchimPieters Firmware size makes problem..

https://github.com/HomeACcessoryKid/life-cycle-manager/issues/11

closed 2019-08-04 by kiralikbeyin