maximkulkin/esp-homekit-demo #249: ESP32 - Timout - Waiting for Header

kunjal83, 2019-09-04:
Hi all,

I am trying to work with ESP32 with the most basic header available [here](https://www.amazon.in/gp/product/B07CSJ5MXM/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) on amazon. 
However, I am constantly failing to do anything with it. 
Checking the voltage on pins: 
Pin 3V3 (2) and EN = 3.29. 
Pin GPIO0 and GND(1) = GND
Pin GPIO2 - Floating

I am not sure what I am missing as I have tried almost everything but still seems to fail. 
Can anyone please help with your hardware setup for successful communication with ESP32? 

Much appreciate any pointers. 

Thanks,
Kunjal

maximkulkin, 2019-09-05:
Just start with something [more straightforward](https://www.amazon.in/dp/B075YSNVPJ/ref=cm_sw_r_cp_api_i_YMgCDbAYXDRZ4)

kunjal83, 2019-09-05:
@maximkulkin , Thanks mate. 
I did try something very basic of my own on a perforated PCB and it worked out perfectly fine. 

Thanks ton. 
K

closed 2019-09-05 by kunjal83