maximkulkin/esp-homekit-demo #254: error with: git submodule update --init --recursive

renandw, 2019-10-05:
Hello guys, I'am having trouble when doing Initialize and sync all submodules (recursively):

~/projects/esp-homekit-demo$ git submodule update --init --recursive
fatal: index file smaller than expected
Failed to recurse into submodule path 'components/esp-8266/cJSON/cJSON'
fatal: index file smaller than expected
fatal: index file smaller than expected
fatal: index file smaller than expected
Failed to recurse into submodule path 'components/esp-8266/cJSON'
Failed to recurse into submodule path 'components/esp-8266/led-status'
Failed to recurse into submodule path 'components/esp-8266/qrcode'
Failed to recurse into submodule path 'components/esp-8266/wifi_config'

Any one having this trouble?
thanks

kenthinson, 2019-10-09:
Are you on Mac? I couldn’t get it working normally. I ended up using the docer instructions. Then it was all fine.

maximkulkin, 2019-10-09:
Try removing your working copy and clone whole repository again.

renandw, 2019-10-11:
Thanks. E redo all the process and it is working. Thanks

closed 2019-10-11 by renandw