maximkulkin/esp-homekit-demo #255: Buttons

Kristian8606, 2019-10-05:
hello I built two thermostats with a display and three buttons from the examples of Maxim When I plug a device into the socket, one thermostat resets my configuration and I have to add it from the beginning.
If I add bits through three tasks then I have no reset but I have too many tasks and I failed to pair.

AchimPieters, 2019-10-18:
Did you made two identical thermostats? I believe that you need to make them unique.

Kristian8606, 2019-10-21:
I set the GPIO_INTTYPE_EDGE_NEG interruptions  and the restart is gone and the thermostats are stable for now
 
[image]

maximkulkin, 2019-10-21:
You probably have your buttons active state different from what was set up in the code.

Kristian8606, 2019-10-22:
I use this connection scheme

[image]

Kristian8606, 2019-10-22:
previously i used your example for buttons with the same connection scheme
but i got strange esp behavior and reset.
 am i doing something wrong?

AchimPieters, 2019-10-22:
@Kristian8606 look at this: https://github.com/maximkulkin/esp-button

maximkulkin, 2019-10-22:
Yeah, look at that. Obviously your case is “active low”, but that example was designed for “active high” case.

closed 2019-11-05 by maximkulkin