maximkulkin/esp-homekit-demo #258: Watchdog Timer request

okanduzyel, 2019-10-07:
Hi,

I am testing your examples which are very good but I found some issues. 

First, even simple led example, all of them crashes after hours or days. I mean that, esp8266 gets under hard fault. It is stuck and doesn't any response. (its program counter is stuck). Then, requires reset by hand.

I am testing them on Lolin v3 esp8266 (I have tested two different lolin board and I don't think my hardwares have any problem). Unfortunately, both of them are stuck. 

So, I think, examples need watchdog timer. For example, If esp doesn't respond within 30 second or 1 minute, It must be automatically reseted by watchdog timer. 

I said, hard fault causes after hours or after days. But it happens. Can you add watchdog timer for your examples?

maximkulkin, 2019-10-07:
I have a few of my accessories at home running for months and I had never had any problems with them. I have a few accessories that I bought (a smart socket, a dimmable smart light bulb) which surprisingly are quite often unresponsive and I need to manually power cycle them.

I'm not a fan of lame fixes and would prefer to trace down the problem and fix the problem in the code, rather than advising everybody that "this is normal, just throw more hardware workarounds on it".

okanduzyel, 2019-10-07:
You are right but it is necessary for HomeKit devices. If it will be stuck and required reset by hand, Its idea is going to opposite way. If we always reset it by hand, we can also open the lights, open the windows etc. So what does HomeKit do for us?

> surprisingly are quite often unresponsive and I need to manually power cycle them.

Esp8266 must handle that problem itself and must work correctly by years. Please, just think about it.

RavenSystem, 2019-10-07:
I have many custom devices running esp-homekit for months without any issue.

maximkulkin, 2019-10-07:
@okanduzyel What I'm saying is: you can definitely add watchdogs and reboot devices every now and then, but this is not normal behavior and probably has something to do with your setup.

Kristian8606, 2019-10-07:
The only problem I find is resetting a device that uses an example with more than one button when plugging an appliance into a socket that is near the device.  If the buttons are performed as tasks then the device is stable.  I understand that the problem is caused by EMI but I do not understand how emi causes device reset.  I added 2 amps of power and for a few days the device looks stable.  Try a more powerful power supply and monitor the devices.  All other devices I have no buttons on have been stable for months, unlike my bulbs  IKEA

okanduzyel, 2019-10-08:
> I have many custom devices running esp-homekit for months without any issue.

Please test that,

some examples have wifi configuration page. After first program of esp, you need to open your settings->wifi and find esp's wifi name. Connect it, select your wifi name and enter your wifi password. Phone will be reconnect you home's wifi and you are able to find your device in HomeKit accessories. Everyone knows that. 

Now the case for you, Cut home's electricity(whole electricity), wait two minutes and give it back. Modem will be boot up 30-90 seconds but esp will be ready less ten 5 seconds. When it can not find your modem, it opens configuration wifi and waits like that. Your modem boots up and your devices never connect your wifi. You need to reset them by hand. After hard reset, It can find modem's wifi and works correctly. 

Haven't your home's electricity ever gone for months? Just a little timer solves all problems. But forget about it. Everyone will find their solutions.

Kristian8606, 2019-10-08:
I think I understand.
Are you sure you have updated your homekit components?
When my esp starts and does not detect wifi it scans for 5 minutes and when wifi returns it connects.

okanduzyel, 2019-10-08:
You can check sonoff examples to understand what I mean.

I am currently working to merge of temperature and motion sensor without configuration page. I have done it but I need to check it for a week. If issues were solved, I will share the code here.

peros550, 2019-10-08:
The wifi behavior your are describing has been fixed with one of Maxim latest updates in this [library](https://github.com/maximkulkin/esp-wifi-config) a while ago. It’s very strange that you still facing this issue. Please make sure you are using latest updates on your repository. 

And here is a similar [issue](https://github.com/maximkulkin/esp-wifi-config/issues/13) that you may want to read further.

peros550, 2019-10-08:
> > I have many custom devices running esp-homekit for months without any issue.
> 
> Please test that,
> 
> Some examples have wifi configuration page. I mean that after first program of your esp, you need to open your settings->wifi and find esp's wifi name. Connect it, select your wifi name and enter your wifi password. Phone will be reconnect your home's wifi and then, you are able to find your device in HomeKit accessories. Everyone knows that.
> 
> Now the case for you, Cut home's electricity(whole electricity), wait two minutes and give it back. Modem will be boot up 30-90 seconds but esp will be ready less ten 5 seconds. When it can not find your modem, it opens configuration wifi and waits like that. Your modem boots up and your devices never connect your wifi. Because, there are under configuration mode. You need to reset them by hand. After hard reset, It can find modem's wifi and works correctly.
> 
> Haven't your home's electricity ever gone in months? Just a little timer solves all problems. But forget about it. Everyone will find own solutions.

The wifi behavior your are describing has been fixed with one of Maxim latest updates in this [library](https://github.com/maximkulkin/esp-wifi-config) a while ago. It’s very strange that you still facing this issue. Please make sure you are using latest updates on your repository. 

And here is a similar [issue](https://github.com/maximkulkin/esp-wifi-config/issues/13) that you may want to read further.

okanduzyel, 2019-10-12:
> The wifi behavior your are describing has been fixed with one of Maxim latest updates in this [library](https://github.com/maximkulkin/esp-wifi-config) a while ago. It’s very strange that you still [...]

I had deleted vagrant and installed again. Now, It works what you mention like. It is so strange 😕 I am currently dealing with firewall problems with my new modem. I need to test it for a long time to understand its stability.

okanduzyel, 2019-10-23:
After 11 days, 

I have had an issue about hard fault. Now, I have temperature + motion sensor (one device combined) and three sonoff basic toggle which are connected to lamps.

Today, temperature + motion sensor's esp8266 has fallen hard fault which means that, Its program counter stuck and can not run a code. Moreover, It causes that it cancels all devices' signals. Each device gives "no response".

I suspected temperature + motion sensor and just pulled out the plug and plugged in again. I didn't touch any devices. Then, every devices became normal working. I don't think I have hardware issue. This is just lolin v3 board and I tried it two different boards. 

We are taking remote control, where is outside of home, via hub. One device can cancel all devices. Some devices, which are like sonoff basic toggle, are located a place, where can not be reachable always. Inside of lamb, inside of wall etc. 

It is simple idea. If esp8266 has fallen hard fault for 30 seconds or 1 minute. Watchdog resets the device. So, We don't deal with this kind of super annoying problems.

maximkulkin, 2019-11-05:
Ok, to end this conversation: if existing examples do not work for you, as well as if you need an extra functionality, you're advised to create your own project and implement whatever you feel necessary. I personally rarely use stock examples, I build my own custom firmwares for every project I do. E.g. recently I have updated all my firmwares to include OTA update functionality and blink codes, just to name a few.

closed 2019-11-05 by maximkulkin