maximkulkin/esp-homekit-demo #260: door-sensor requires hub

okanduzyel, 2019-10-12:
Hi, 

Door-sensor requires a hub. It works after programming esp, however, cutting cable connection between pc it doesn't work. Am I doing anything wrong or it works like that? 🙁

okanduzyel, 2019-10-14:
Addition, 

I have tested two different board(lolin v3, esp-12e) It unfortunately works like what I mention. Is there any way that It can work standalone? (without required hub)

maximkulkin, 2019-10-14:
I am not sure what weird experiments you are conducting here. Can you explain what are you trying to achieve and what do you call “a hub”?

maximkulkin, 2019-10-14:
If I read that correctly, yes, if you want to do something in reaction to something else, you need a HomeKit hub (e.g. an Apple TV or an iPad) which will monitor states of triggers and update states of other devices.

okanduzyel, 2019-10-14:
There is an example multiple sensor which has motion sensor and dht22 temperature & humidity. 

https://github.com/sakisdog/esp-homekit-demo/tree/master/examples/multiple_sensors

These sensors can work standalone and I can set up notification for that when motion occurs. I want to do samething for door-sensor. When door open, I want to get notification that  "door opened".

I don't want to buy Apple TV or an iPad now to manage HomeKit hub. When I connected my wifi, I wanna get door's information. Is there any way to work it standalone like example of multiple sensor?

maximkulkin, 2019-10-14:
No, that's not gonna work as far as I know, iPhone will not maintain persistent connection to devices even when you're on the same WiFi network: whenever iPhone is locked it will turn off WiFi (not immediately, but eventually), so it won't monitor your HomeKit devices.

HomeKit architecture does not assume devices pushing events to "a cloud", so you get notifications anywhere. Devices push changes in their state only to connected controllers (iPhones, iPads, etc), thus in order to have global notifications you need a hub.

okanduzyel, 2019-10-14:
This motion sensor https://github.com/sakisdog/esp-homekit-demo/tree/master/examples/multiple_sensors can work standalone and It can send notification for me when I am at home. I want to do same thing for door-sensor. 

When I am at home, I want to get notification. When I am outside I don't care it. 

So, buying Apple TV or new iPad is expensive for me. If I can change it as motion sensor's working style, It is okay for me. 

I am not sure that I could explain myself clearly. If didn't understand anything please ask me.

maximkulkin, 2019-10-14:
> I want to do same thing for door-sensor.

Ok, sure, go for it!

AchimPieters, 2019-10-15:
@okanduzyel What I understand from what you are asking [this](https://support.apple.com/en-us/HT208939) is the solution?

okanduzyel, 2019-10-15:
I have no any solution. I just asked to @maximkulkin is it technically possible or not? 

I have currently no any hardware to test it. Just, made ordered new esp8266 boards. 

I don't want to connect it to hub. I don't want to automate any other devices with door-sensor. 

I just want to see door's status, opened-closed, and wanna get notification on my phone or Mac. 

(Motion sensor example can do that without any hub requirement)

AchimPieters, 2019-10-15:
This is not how it works. this repo is not here to fulfill each and every your HomeKit accessory fantasy. Most examples that written is to show how FRAMEWORK works/should be used. The way it does work is: if you've used this framework to implement a very new and interesting accessory, you can send pull request to share it with the world, because someone might not have skill or time to customize existing example, is not the purpose of this repo.

okanduzyel, 2019-10-16:
Unfortunately, my esp-12e has been broken.I hHave taken off it on lolin v3 board and connected esp-01 via cable. 

I can program it. Moreover, I can run sonoff_basic example on it, however, I couldn't done door-sensor example yet.

When make file like that, 

```
PROGRAM = door-sensor

EXTRA_COMPONENTS = \
	extras/http-parser \
	$(abspath ../../components/esp-8266/cJSON) \
	$(abspath ../../components/common/wolfssl) \
	$(abspath ../../components/common/homekit)

REED_PIN ?= 2

FLASH_SIZE ?= 32

EXTRA_CFLAGS += -I../.. -DHOMEKIT_SHORT_APPLE_UUIDS -DREED_PIN=$(REED_PIN)

include $(SDK_PATH)/common.mk

monitor:
	$(FILTEROUTPUT) --port $(ESPPORT) --baud 9600 --elf $(PROGRAM_OUT)
```

It runs correctly but, after hard reset, esp's flash is vanishing. 

I changed it like bellow,

```
PROGRAM = door-sensor

EXTRA_COMPONENTS = \
	extras/http-parser \
	$(abspath ../../components/esp-8266/cJSON) \
	$(abspath ../../components/common/wolfssl) \
	$(abspath ../../components/common/homekit)

REED_PIN ?= 2

FLASH_SIZE ?= 8
FLASH_MODE ?= dout
FLASH_SPEED ?= 40
HOMEKIT_SPI_FLASH_BASE_ADDR ?= 0x7A000

EXTRA_CFLAGS += -I../.. -DHOMEKIT_SHORT_APPLE_UUIDS -DREED_PIN=$(REED_PIN)

include $(SDK_PATH)/common.mk

monitor:
	$(FILTEROUTPUT) --port $(ESPPORT) --baud 9600 --elf $(PROGRAM_OUT)
```

It runs but, when I try to add accessories, It doesn't pair and gives those errors. 

Failed to compact data: sector data read error
Failed to write pairing info to flash: max number of pairings

What do I miss?

maximkulkin, 2019-10-22:
@okanduzyel  It seems that you have hardware issues with your flash. Try using a different base address (e.g. 0x7B000 instead of 0x7A000).

okanduzyel, 2019-10-23:
> @okanduzyel It seems that you have hardware issues with your flash. Try using a different base address (e.g. 0x7B000 instead of 0x7A000). I also recommend erasing flash before flashing.

I tried all of those things and didn't work. Strange, right? I always find strange things 🙂

maximkulkin, 2019-10-23:
There is usually a simple reason for that: you probably do it wrong. :smile:

closed 2019-11-05 by maximkulkin