maximkulkin/esp-homekit-demo #261: problems with build

Kristian8606, 2019-10-13:
i built a homekit_demo with docker because i failed with the old method. I followed the instructions but when I try to make a flash I get this error

> make: *** No rule to make target `/project/examples/thermostat/thermostat.c', needed by `build/program//thermostat.o'.  Stop.

maximkulkin, 2019-10-14:
Before you flash (for the first time or after you make a change to source code), you need to build it. If your firmware is younger than all source files (which means it was last changed after last change to any source file) it will not try to rebuild it (and thus you won't get error like above).

Kristian8606, 2019-10-14:
I have no success with docker even though I follow all the instructions.
is there any other way to build without docker mac os catalina?

maximkulkin, 2019-10-14:
Nothing I am aware of. Docker works perfectly for me, see no reason to look for alternatives.

Kristian8606, 2019-10-14:
i managed i think my problem was in Makefile on a my acsesoaries. my path was set to - 
` include $ (abspath ../../sdk/esp-open-rtos/common.mk)`
after changing it to -
`include $ (SDK_PATH) /common.mk`

I was successful.

Kristian8606, 2019-10-14:
when i restart my computer do i have to do your dock instruction for docker every time?

maximkulkin, 2019-10-14:
no

AchimPieters, 2019-10-15:
@Kristian8606 I use macOS Catalina from the day it was released. I did a total clean install see [here](https://www.studiopieters.nl/macos-catalina-bootable-usb/).
Then I used the Docker script [esp-sdk-dockerfile](https://github.com/AchimPieters/HomeKit-Core/blob/master/esp-sdk-dockerfile) and [esp-rtos-dockerfile](https://github.com/AchimPieters/HomeKit-Core/blob/master/esp-rtos-dockerfile). and everything works just fine. Maybe [this ](https://www.studiopieters.nl/esp-homekit-sdk-full-installation/)will help?

closed 2019-11-05 by maximkulkin