maximkulkin/esp-homekit-demo #265: Sonoff_basic_toggle

okanduzyel, 2019-10-27:
I have temperature + motion sensor and some sonoff basic toggle on my lamps. 

I have also iPhone6 (iOS12) and iPhone7 (iOS13.2 beta) devices. The issue is that when I toggle any the lamp via physical button, iOS12 immediately shows its state on screen but iOS13 doesn't show sometimes.

Do somethings changed on iOS13? Is it OS bug or HomeKit doesn't exactly compatible with iOS13? Do you feel some issues in your home?

Addition: I was always like that on iOS13. 13, 13.1, 13.1.1 ... I thought it was os bug and had registered my phone to beta program for getting iOS13.2 beta but, nothing has been changed yet.

maximkulkin, 2019-11-05:
I haven't noticed any problems with that. Running on iOS 13.1.3, everything is updated promptly.

Kristian8606, 2019-11-05:
Sometimes it happened to me

closed 2019-12-18 by okanduzyel