maximkulkin/esp-homekit-demo #269: Guide to build accessories

renandw, 2019-11-07:
Hello.

In the past few months I've been building the circuits and using the examples in the library. And I was amazed, it's really good.
But now I am trying to build my own accessory merging some examples, for instance the Led and Button example.
The problem is that I am getting some errors and I am trying to search the answer.
Do you guys can give me a website or a PDF so I can keep learning

for example this is my code, but I am getting this errors
"main.c:21:13: warning: 'wifi_init' defined but not used [-Wunused-function]
 static void wifi_init() {
main.c:110:5: error: expected '}' at end of input
main.c:110:5: error: 'HOMEKIT_ACCESSORY' undeclared here (not in a function)
     HOMEKIT_ACCESSORY(
main.c:180:0: error: unterminated argument list invoking macro "HOMEKIT_ACCESSORY"
 }

#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"
#include "button.h"

#define BUTTON_PIN 0
#ifndef BUTTON_PIN
#error BUTTON_PIN is not specified
#endif

static void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
    sdk_wifi_station_connect();
}

const int led_gpio = 2;
bool led_on = false;

void led_write(bool on) {
    gpio_write(led_gpio, on ? 0 : 1);
}

void led_init() {
    gpio_enable(led_gpio, GPIO_OUTPUT);
    led_write(led_on);
}

void led_identify_task(void *_args) {
    for (int i=0; i<3; i++) {
        for (int j=0; j<2; j++) {
            led_write(true);
            vTaskDelay(100 / portTICK_PERIOD_MS);
            led_write(false);
            vTaskDelay(100 / portTICK_PERIOD_MS);
        }

        vTaskDelay(250 / portTICK_PERIOD_MS);
    }

    led_write(led_on);

    vTaskDelete(NULL);
}

void led_identify(homekit_value_t _value) {
    printf("LED identify\n");
    xTaskCreate(led_identify_task, "LED identify", 128, NULL, 2, NULL);
}

homekit_value_t led_on_get() {
    return HOMEKIT_BOOL(led_on);
}

void led_on_set(homekit_value_t value) {
    if (value.format != homekit_format_bool) {
        printf("Invalid value format: %d\n", value.format);
        return;
    }

    led_on = value.bool_value;
    led_write(led_on);
}

void button_identify(homekit_value_t _value) {
    printf("Button identify\n");
}

homekit_characteristic_t button_event = HOMEKIT_CHARACTERISTIC_(PROGRAMMABLE_SWITCH_EVENT, 0);

void button_callback(uint8_t gpio, button_event_t event) {
    switch (event) {
        case button_event_single_press:
            printf("single press\n");
            homekit_characteristic_notify(&button_event, HOMEKIT_UINT8(0));
            break;
        case button_event_double_press:
            printf("double press\n");
            homekit_characteristic_notify(&button_event, HOMEKIT_UINT8(1));
            break;
        case button_event_long_press:
            printf("long press\n");
            homekit_characteristic_notify(&button_event, HOMEKIT_UINT8(2));
            break;
        default:
            printf("unknown button event: %d\n", event);
    }
}

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(
        .id=1,
        .category=homekit_accessory_category_programmable_switch,
        .services=(homekit_service_t*[]) {
            HOMEKIT_SERVICE(
                ACCESSORY_INFORMATION,
                .characteristics=(homekit_characteristic_t*[]) {
                    HOMEKIT_CHARACTERISTIC(NAME, "Button"),
                    HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
                    HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0012345"),
                    HOMEKIT_CHARACTERISTIC(MODEL, "MyButton"),
                    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
                    HOMEKIT_CHARACTERISTIC(IDENTIFY, button_identify),
                    NULL
                },
            ),
            },
            HOMEKIT_SERVICE(
                STATELESS_PROGRAMMABLE_SWITCH,
                .primary=true,
                .characteristics=(homekit_characteristic_t*[]) {
                    HOMEKIT_CHARACTERISTIC(NAME, "Button"),
                    &button_event,
                    NULL
                },
            ),
            NULL
        },
        HOMEKIT_ACCESSORY(.id=2, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
            HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
                HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
                HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
                HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF19D"),
                HOMEKIT_CHARACTERISTIC(MODEL, "MyLED"),
                HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
                HOMEKIT_CHARACTERISTIC(IDENTIFY, led_identify),
                NULL
            }),
            HOMEKIT_SERVICE(LIGHTBULB, .primary=true, .characteristics=(homekit_characteristic_t*[]){
                HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
                HOMEKIT_CHARACTERISTIC(
                    ON, false,
                    .getter=led_on_get,
                    .setter=led_on_set
                ),
                NULL
            }),
            NULL        
    ),
    NULL
};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void user_init(void) {
    uart_set_baud(0, 115200);

    led_init();
    wifi_init();
    if (button_create(BUTTON_PIN, button_callback)) {
        printf("Failed to initialize button\n");
    }
    homekit_server_init(&config);
}

maximkulkin, 2019-11-07:
```
"main.c:21:13: warning: 'wifi_init' defined but not used [-Wunused-function]
static void wifi_init() {
```
It is what it says: your function is not used anywhere.

```
main.c:110:5: error: expected '}' at end of input
main.c:110:5: error: 'HOMEKIT_ACCESSORY' undeclared here (not in a function)
HOMEKIT_ACCESSORY(
main.c:180:0: error: unterminated argument list invoking macro "HOMEKIT_ACCESSORY"
}
```
There is a "}" missing somewhere.

[image]

Here you can see that same object types are found on different levels, some level closing constructs are missing.

renandw, 2019-11-08:
I've changed the code

`#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"
#include "button.h"

#define BUTTON_PIN 0
#ifndef BUTTON_PIN
#error BUTTON_PIN is not specified
#endif

static void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
    sdk_wifi_station_connect();
}

void button_identify(homekit_value_t _value) {
    printf("Button identify\n");
}

const int led_gpio = 2;
bool led_on = false;

void led_write(bool on) {
    gpio_write(led_gpio, on ? 0 : 1);
}

void led_init() {
    gpio_enable(led_gpio, GPIO_OUTPUT);
    led_write(led_on);
}

void led_identify_task(void *_args) {
    for (int i=0; i<3; i++) {
        for (int j=0; j<2; j++) {
            led_write(true);
            vTaskDelay(100 / portTICK_PERIOD_MS);
            led_write(false);
            vTaskDelay(100 / portTICK_PERIOD_MS);
        }

        vTaskDelay(250 / portTICK_PERIOD_MS);
    }

    led_write(led_on);

    vTaskDelete(NULL);
}

void led_identify(homekit_value_t _value) {
    printf("LED identify\n");
    xTaskCreate(led_identify_task, "LED identify", 128, NULL, 2, NULL);
}

homekit_value_t led_on_get() {
[... 96 lines cut ...]
    led_init();
    wifi_init();
    if (button_create(BUTTON_PIN, button_callback)) {
        printf("Failed to initialize button\n");
    }
    homekit_server_init(&config);
}
`

but now I get this errors:
`CC /home/esphomekit/projects/esp-homekit-demo/examples/button/main.c
/home/esphomekit/projects/esp-homekit-demo/examples/button/main.c:14:0: warning: "BUTTON_PIN" redefined [enabled by default]
 #define BUTTON_PIN 0
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/esphomekit/projects/esp-homekit-demo/components/common/homekit//include/homekit/homekit.h:4:0,
                 from /home/esphomekit/projects/esp-homekit-demo/examples/button/main.c:9:
/home/esphomekit/projects/esp-homekit-demo/components/common/homekit//include/homekit/types.h:233:5: error: invalid operands to binary & (have 'void *' and 'homekit_accessory_t')
     &(homekit_accessory_t) { \
     ^
/home/esphomekit/projects/esp-homekit-demo/examples/button/main.c:137:5: note: in expansion of macro 'HOMEKIT_ACCESSORY'
     HOMEKIT_ACCESSORY(.id=2, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
     ^
/home/esphomekit/projects/esp-open-rtos/common.mk:217: recipe for target 'build/program//main.o' failed
make: *** [build/program//main.o] Error 1`

maximkulkin, 2019-11-08:
It's hard to read, all formatting is messed up

renandw, 2019-11-08:
sorry about that. I've updated my reply.

maximkulkin, 2019-11-08:
I recommend you first start with some read (e.g. [this one](https://www.amazon.com/C-Dummies-Dan-Gookin/dp/0764570684/ref=asc_df_0764570684/))

renandw, 2019-11-10:
Thanks maxim, I really appreciate for your time. I will take a look into this.

closed 2020-02-13 by maximkulkin