maximkulkin/esp-homekit-demo #271: sonoff_dual_blinds

okanduzyel, 2019-11-09:
sonoff_dual_blinds example has a problem. Up-down buttons don't work correctly.

AchimPieters, 2019-11-10:
Uwe these buttons: https://github.com/maximkulkin/esp-button

okanduzyel, 2019-11-10:
I use exactly these buttons with that schematics (with pull up resistor)

AchimPieters, 2019-11-10:
What do they do, or not do?

okanduzyel, 2019-11-10:
up button = open
down button = close.

Down button never works, Up button works wrong. When curtain closed, up button doesn't response. When curtain opened, up button tries to open where opened curtain.

Gpio settings are same with example. Lolin v3 board are new. I don't think it has any problem. 

If you have esp device, try it, you will see what I mention.

AchimPieters, 2019-11-10:
well I have already: [https://github.com/AchimPieters/ESP8266-HomeKit-Blinds](https://github.com/AchimPieters/ESP8266-HomeKit-Blinds)
and it works just fine. Try my code?

okanduzyel, 2019-11-10:
Now, I am at business trip. Maybe next week, I will try your code @AchimPieters

okanduzyel, 2019-11-25:
> well I have already: https://github.com/AchimPieters/ESP8266-HomeKit-Blinds
> and it works just fine. Try my code?
> 
> btw: It's not a good idea to use pin nine, see my blog here:https://www.studiopieters.nl/esp8266-pinout/

Your example is definitely work! Thank you @AchimPieters

toshibochan, 2019-11-25:
@AchimPieters the button is active high or low?

AchimPieters, 2019-11-26:
@toshibochan  read this: https://github.com/maximkulkin/esp-button

closed 2019-12-18 by okanduzyel