maximkulkin/esp-homekit-demo #276: Any ideas in using it on S31 ?

pinalo, 2019-11-19:
Well basically the S20 and S31 have the same ESP and the same pin assignments , but when I load it it just sticks on the WiFi selection and don't connect to my Network it stuck in the AP mode,  im really not interested on the current monitor in this point, so any ideas what is happening ?

maximkulkin, 2019-11-19:
Need more info

pinalo, 2019-11-19:
Well, I check the serial monitor and its curious. Let me explain in the steps.
 I make the Sonoff basic all, after that I put in on my Sonoff and everything works perfect. After that I make the S20 and put in on my S31 and just stuck (checking the serial when I put my WiFi credentials, it just drops "No Wifi Configuration".
So I go back and flash another Sonoff Basic , and It just gets in the same error

closed 2019-11-21 by pinalo