maximkulkin/esp-homekit-demo #280: Window covering always on opening state

kiralikbeyin, 2019-11-25:
When i push the (up/down) button, homekit always show "opening" and never shows %xx open untill i set a new value from homekit screen.

````
position_state.value.int_value = POSITION_STATE_STOPPED; homekit_characteristic_notify(&position_state, position_state.value);
````
I think this code doesnt work for me, any one else?

toshibochan, 2019-11-25:
Same for me

kiralikbeyin, 2019-11-26:
@maximkulkin

maximkulkin, 2019-11-26:
@kiralikbeyin What do you expect from me? You didn't even mention which firmware it is.

maximkulkin, 2019-11-26:
Guys, guys, this is not my firmware, somebody contributed it. I do not know what is happening there. If you have ideas on how to improve it, you're welcome to contribute.

konifer44, 2019-12-21:
Im using it and everything its working. I modify code in some places to add stepper motor support and hall sensor to detect full opened window covering and ensure it will never lost position in case power loss.

toshibochan, 2019-12-22:
@konifer44 you can share your code?

konifer44, 2019-12-22:
What ESP are You using?

toshibochan, 2019-12-22:
I’m not sure yet. What you using? Can you take pictures of you setup?

konifer44, 2019-12-24:
Are you using ESP32 or ESP8266

toshibochan, 2019-12-24:
I don’t have yet. What is better ESP32 or ESP8266?

konifer44, 2019-12-24:
ESP32 is more powerful😊

closed 2020-02-13 by maximkulkin