maximkulkin/esp-homekit-demo #281: Unable to proceed pairing

sgetmanring, 2019-11-28:
Follow the manual. Flash `esp32 led` example to my esp32. And freeze on that step:
```
WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 0C:5B:3B:F7:ED:56
>>> HomeKit: Configuring mDNS
>>> homekit_setup_mdns: Accessory Setup ID = 1QJ9
>>> homekit_run_server: Staring HTTP server
```
Generate qr :
```
tools/gen_qrcode 5 111-11-111 1QJ9 qrcode.png
```
Home cannot find accessory. 
Appreaciate any help

maximkulkin, 2019-11-29:
Does your accessory category match code “5” (lightbulb) that you’re using?

sgetmanring, 2019-11-29:
Yes, so I picked up value from `led.c`:
```
HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
```
and
```
homekit_accessory_category_lightbulb = 5,
```

bagabont, 2019-12-01:
Same here too. I tried clean setup with the latest esp-idf (I did it on 16.11), then did a clean clone of esp-homekit-demo and make erase_flash flash monitor

`I (8133) tcpip_adapter: sta ip: 192.168.1.62, mask: 255.255.255.0, gw: 192.168.1.1
WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 58:4D:27:7D:7E:D6
>>> HomeKit: Configuring mDNS
>>> HomeKit: Got new client connection: 55
>>> HomeKit: [Client 55] Pair Setup Step 1/3
I (180633) wifi: state: run -> init (6c0)
I (180633) wifi: pm stop, total sleep time: 151687078 us / 174540208 us

I (180633) wifi: new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
!!! HomeKit: [Client 55] Error reading data from socket (code 113). Disconnecting
STA disconnected
>>> HomeKit: [Client 55] Closing client connection
I (180773) wifi: new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (180773) wifi: state: init -> auth (b0)
I (180773) wifi: state: auth -> assoc (0)
I (180783) wifi: state: assoc -> run (10)
I (180803) wifi: connected with Fi-Wi, aid = 57, channel 2, BW20, bssid = 78:29:ed:64:1f:eb, security type: 3, phy: bgn, rssi: -83
I (180803) wifi: pm start, type: 1

I (180823) wifi: AP's beacon interval = 102400 us, DTIM period = 5
I (182133) tcpip_adapter: sta ip: 192.168.1.62, mask: 255.255.255.0, gw: 192.168.1.1
WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 58:4D:27:7D:7E:D6
>>> HomeKit: Configuring mDNS
`

kemkriszt, 2019-12-14:
I had the same issue. After some struggling I tested with an other router and it kinda works (I think that's an other issue). Do you have any idea what router setting would prevent this from connecting? I have to mention I have Homekit accessories working on my network.

maximkulkin, 2020-02-13:
From log it looks like ESP32 looses WiFi connectivity. Was it resolved? Which ESP-IDF version are you running?

closed 2020-06-26 by maximkulkin