maximkulkin/esp-homekit-demo #284: anyone have Sonoff mini Homekit firmware?

toshibochan, 2019-12-05:
anyone have Sonoff mini Homekit firmware?

peros550, 2019-12-05:
It would be nice to know how to load firmware using Sonoff DIY firmware change tool

toshibochan, 2019-12-05:
I find this one but give me error to build the firmware. And is a lot difference of sonoff-basics

https://github.com/maccoylton/esp-homekit-sonoff-mini

maccoylton, 2019-12-05:
The build error is I suspect the way you have set up your build environment. Checkout the Wiki I just added for instructions to install pre-built firmware.

maccoylton, 2019-12-05:
@peros550 the mini uses a different boot loader, so to my knowledge no one has figure out how to replace that OTA with Sonoff DIY tools. I have successfully flashed three by hard wiring with my firmware and they work great

toshibochan, 2019-12-07:
@maccoylton thanks your sonoff mini is working great!
https://github.com/maccoylton/esp-homekit-sonoff-mini/wiki

closed 2019-12-07 by toshibochan