maximkulkin/esp-homekit-demo #288: Update build instructions so folders with spaces work

njdancer, 2019-12-17:
When following the build instructions from the wiki(which are generally great btw. Thanks!) the docker run command will fail if the working directory has any spaces in it and the following error is presented.

```
docker: invalid reference format: repository name must be lowercase.
```

This can be resolved by wrapping the path in double quotes so this,
```
docker run -it --rm -v `pwd`:/project -w /project esp-rtos make -C examples/sonoff_basic all
```
becomes
```
docker run -it --rm -v "`pwd`":/project -w /project esp-rtos make -C examples/sonoff_basic all
```
.

I would update this myself but it appears only contributors can edit the wiki.

njdancer, 2019-12-17:
So from what I can tell make doesn't handle folders with spaces in them at all so even after working around this I couldn't get it to work without ensuring the pwd contains no spaces. As such this wiki change is pointless but it may be worthwhile adding a note to use only folders without spaces as the docker error is not particularly helpful.

maximkulkin, 2020-01-27:
IMO using computers assumes some base level of competency and babysitting newbies is not the most enjoyful way to spend your personal time.
But I've fixed that particular case in documentation for you.

closed 2020-01-27 by maximkulkin