maximkulkin/esp-homekit-demo #289: Is it possible implement bridge service?

faronov, 2019-12-17:
Is it possible to implement bridge  service and add devices to this bridge later
as it implemented in Aqara Hub
I think esp32 will be powerful enough to serve multiple service and characteristics
but how bridge implementation must look like?

d4rkmen, 2019-12-24:
Bridge is very easy from the HAP side. First accessory object  (AID=1) is category BRIDGE, the rest (AID 2..100) are the bridged acessories. But for convinient work there should be identifiers cache implemented (same bridged devices should have same aid/iid every time)

Kristian8606, 2019-12-30:
I tried to add a bridge and statically add a few accessories but failed. Can anyone give an example code for how to make accessories?

d4rkmen, 2019-12-30:
Just make sure every accessory uses its own iid numbers (accessory info service should be always iid=1 and so on)

nonameplum, 2019-12-30:
You can take a look at my example https://github.com/nonameplum/esp-homekit-demo/blob/60437fa101ca7eabd1faf6cdd29b360c599ebe3e/examples/garage/garage.c#L344

Garage is the "main"/bridge device. Light switch is another accessory.

d4rkmen, 2019-12-30:
You need to change code this way: accessory id=1 category=bridge services=accinfo
id=2 garage
id=3 lightbulb

nonameplum, 2020-01-02:
@d4rkmen what is the advantage of having `id=1 bridge` and then:
```
id=2 garage
id=3 lightbulb
```

Instead of:
```
id=1 garage (becomes automatically a bridge)
id=2 lightbulb
```
?

d4rkmen, 2020-01-02:
@nonameplum well, you told it does not work with your sample. SO i thought may be thats the point. I never tried myself to make bridges with first accessory != bridge. If this works - why not.

nonameplum, 2020-01-02:
I don't see that I told anything about that it doesn't work for me 🙂 anyway home app automatically consider the first accessory as a bridge. Not sure though if making explicit one gives some additional benefits. I didn't try it yet.

RavenSystem, 2020-01-02:
If you declare more than 4 accessories, you must use an explicit bridge.

Kristian8606, 2020-01-02:
With your help I was able to create a bridge with three temperature sensors.  but how do i add them dynamically?

Kristian8606, 2020-01-03:
Is there any way to i start on the bridge some web server or through some protocol to work with Homekit to get temperature values on the bridge.

nonameplum, 2020-01-03:
If you just add another accessory (`HOMEKIT_ACCESSORY`) `homekit_accessory_category_thermostat` won't it work for you?
https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/temperature_sensor/temperature_sensor.c

Kristian8606, 2020-01-03:
Sorry for my bad English I obviously didn't ask my question correctly.  I would like to send temperature values ​​to my bridge.  in this way I will build wireless battery temperature sensors.

Kristian8606, 2020-01-03:
Can anyone give me an example of a udp client

closed 2020-01-27 by maximkulkin