maximkulkin/esp-homekit-demo #291: Turn off Serial Print

MGPhil, 2019-12-22:
Hi Maximkulkin,
I really appreciate your work and the great SW.
Until now I got the onboard LED example,
the 1-wire Lightstrip and the thermostat to work.

So I thought it is time for the next level and I bought a Beca wall thermostat that has an ESP8266 inside and I like to put your homekit and wifi manager inside to get „cheap“ homekit thermostats.

My problem is now that all communication from the ESP to the thermostat MCU is done via the serial port and if I send „wrong“ commands that could be problematic.

So my question is if there is a way of turning off all serial output from homekit and wifi manager and only have my commands send?

Thanks

Philipp

HomeACcessoryKid, 2019-12-23:
Hi Philipp

You can have a look at two of my repos that relate to your question.
The first is [UDPlogger](https://github.com/HomeACcessoryKid/UDPlogger) which diverts serial output to a UDP channel
The second is a [curtain controller](https://github.com/HomeACcessoryKid/AQARA-ZNCLDJ11LM) that uses the serial port to communicate to the curtain motor, so very similar objective as yours.
Having said that, I have never managed to suppress the initial boot message 100% but if your thermostat survives this just like my curtain, you should be fine to go...

Happy Hacking,
HacK

MGPhil, 2019-12-23:
Hi @HomeACcessoryKid,
thanks for the reply I will have a look in the two repos.
Just one question would it not be possible to tell the ESP with some system command that all printf go now to a virtual or unused port and with special printf command send/recieve only my commands on the right port which is used now by everything?
BR
Philipp

maximkulkin, 2020-01-27:
On esp-idf there is this thing: [esp_log_set_vprintf](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/log.html#_CPPv419esp_log_set_vprintf14vprintf_like_t). Also, you can [reduce bootloader log level](https://github.com/espressif/esp-idf/blob/master/components/bootloader/Kconfig.projbuild#L2).

On esp-open-rtos there is this thing: [set_write_stdout](https://github.com/SuperHouse/esp-open-rtos/blob/master/core/include/stdout_redirect.h#L30)

closed 2020-01-27 by maximkulkin