maximkulkin/esp-homekit-demo #293: Couldn't add fan-5B25

Alr2303, 2020-01-03:
I am using ESP8266 01 module and have followed all ur instructions the code complies and gets written on board but I can't connect to the accessory
Can anyone Help?

ERROR CODE:Wrote 430096 bytes (309729 compressed) at 0x00002000 in 27.3 seconds (effective 125.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
/home/alr/alr-pro/esp01/esp-open-rtos//utils/filteroutput.py --port /dev/ttyUSB1 --baud 115200 --elf ./build/fan.out
Traceback (most recent call last):
  File "/home/alr/alr-pro/esp01/esp-open-rtos//utils/filteroutput.py", line 10, in <module>
    import serial
ImportError: No module named serial
make: *** [/home/alr/alr-pro/esp01/esp-open-rtos/common.mk:261: test] Error 1
make: Leaving directory '/home/alr/alr-pro/dockerprojects/esp-homekit-demo/examples/fan'

maximkulkin, 2020-01-03:
`pip install pyserial`

Alr2303, 2020-01-03:
I tried pip3 install pyserial in my ubuntu PC

Golovkov, 2020-01-04:
Hi @maximkulkin! 
I have the same problem, the device does not add.
Used command 
`pip install pyserial`

but there is no result, everything still does not work: the error is this:
`/Users/IG/ESP/esp-open-rtos//utils/filteroutput.py --port /dev/tty.usbserial-A50285BI --baud 115200 --elf ./build/led.out
Traceback (most recent call last):
  File "/Users/IG/ESP/esp-open-rtos//utils/filteroutput.py", line 10, in <module>
    import serial
ImportError: No module named serial
make: *** [monitor] Error 1`

what else could be the problem?

maximkulkin, 2020-01-06:
Since I have toolchain set up in container, filteroutput address decoding does not work for me, so I have it modified like this:
```diff
diff --git a/utils/filteroutput.py b/utils/filteroutput.py
index 5e9b1cf..c318774 100755
--- a/utils/filteroutput.py
+++ b/utils/filteroutput.py
@@ -1,4 +1,4 @@
-#!/usr/bin/env python
+#!/usr/bin/env python2
 #
 # A thin Python wrapper around addr2line, can monitor esp-open-rtos
 # output and uses gdb to convert any suitable looking hex numbers
@@ -74,13 +74,13 @@ def main():
         print("Reading from stdin...")
         port = sys.stdin
         # disable echo
-	try:
-            old_attr = termios.tcgetattr(sys.stdin.fileno())
-	    attr = termios.tcgetattr(sys.stdin.fileno())
-	    attr[3] = attr[3] & ~termios.ECHO
-	    termios.tcsetattr(sys.stdin.fileno(), termios.TCSADRAIN, attr)
-	except termios.error:
-	     pass
+    try:
+        old_attr = termios.tcgetattr(sys.stdin.fileno())
+        attr = termios.tcgetattr(sys.stdin.fileno())
+        attr[3] = attr[3] & ~termios.ECHO
+        termios.tcsetattr(sys.stdin.fileno(), termios.TCSADRAIN, attr)
+    except termios.error:
+         pass
 
     try:
         while True:
@@ -88,14 +88,14 @@ def main():
             if line == '':
                 break
             print(line.strip())
-            for match in re.finditer(RE_EXECADDR, line, re.IGNORECASE):
-                addr = match.group(0)
-                if not addr.startswith("0x"):
-                    addr = "0x"+addr
-                    # keeping addr2line and feeding it addresses on stdin didn't seem to work smoothly
-                addr2line = subprocess.check_output(["xtensa-lx106-elf-addr2line","-pfia","-e","%s" % args.elf, addr], cwd=".").strip()
-                if not addr2line.endswith(": ?? ??:0"):
-                    print("\n%s\n" % addr2line.strip())
+            # for match in re.finditer(RE_EXECADDR, line, re.IGNORECASE):
+            #     addr = match.group(0)
+            #     if not addr.startswith("0x"):
+            #         addr = "0x"+addr
+            #         # keeping addr2line and feeding it addresses on stdin didn't seem to work smoothly
+            #     addr2line = subprocess.check_output(["docker", "run", "-v", "%s:/project" % os.getcwd(), "-w", "/project", "esp-rtos", "xtensa-lx106-elf-addr2line","-pfia","-e","%s" % args.elf, addr], cwd=".").strip()
+            #     if not addr2line.endswith(": ?? ??:0"):
+            #         print("\n%s\n" % addr2line.strip())
     finally:
         if args.port is None:
             # restore echo
```

closed 2020-02-13 by maximkulkin