maximkulkin/esp-homekit-demo #294: Remote access ESP8266

d4rkmen, 2020-01-07:
The demo accessory (tried lightbulb and thermostat demo) seems unreachable using remote access (through homekit center like iPad/AppleTV)
Same time other my accessories works fine (using avahi-client for advertise)
Once the controller (iPhone) registering in local network - all works again
Any hints please?

maximkulkin, 2020-01-08:
This works perfectly fine for me. Make sure your AppleTV/iPad is enabled as HomeKit hub and is in the same WiFi network as HomeKit accessories.

d4rkmen, 2020-01-08:
Great to hear the thing should work. Now need to understand why it doesnt work for me.
May be I have to set hostname or anything else?

> AppleTV/iPad is enabled as HomeKit

Yes, my other accessories available. Only this ESP-based is not

maccoylton, 2020-01-09:
FYI I run 20 or so ESP based device of various types and remote access works flawlessly

d4rkmen, 2020-01-13:
ESP8266_RTOS example tested and remote access works! Still have pairing problems: 3-4 false tries to pair

maccoylton, 2020-01-13:
@d4rkmen I have seen this also recently, we think it may be related to some memory management issue in recent version of esp-open-rtos, still investigating

d4rkmen, 2020-01-13:
@maccoylton this log is by the latest ESP8266_RTOS_SDK build. But I had same expiriense by esp-open-rtos

closed 2020-01-27 by maximkulkin