maximkulkin/esp-homekit-demo #295: Bridge

Kristian8606, 2020-01-10:
Hello!
I need some help, I created a bridge thanks to you, but after a few minutes the esp freezes.

I use the library for the LoRa module through which I would like to receive data from battery temperature sensors,
the example is still incomplete and is running only one task that receives data from the LoRa module, but after a few minutes the esp freezes without even getting an error.
I will be grateful for any advice.
here is the library.

https://github.com/aldwinhermanudin/esp8266-freertos-lora/tree/master/extras/lora

'
 #define DEVICE_MANUFACTURER "Kristian Dimitrov"
#define DEVICE_NAME "HomeKit-thermostat"
#define DEVICE_MODEL "LoRa Bridge"
#define DEVICE_SERIAL "12345678"
#define FW_VERSION "0.5.0"

#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <espressif/esp_common.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <wifi_config.h>
#include <dht/dht.h>
#include "button.h"
#include <string.h>
#include <stdlib.h>
#include "esp/uart.h"
#include "lora.h"

//const int SENSOR_PIN = 14;

// The GPIO pin that is connected to the relay on the Sonoff Basic.
const int relay_gpio = 2;
// The GPIO pin that is connected to the LED on the Sonoff Basic.
const int led_gpio = 2;
// The GPIO pin that is oconnected to the button on the Sonoff Basic.
const int button_gpio = 0;

void relay_write(bool on) {
    gpio_write(relay_gpio, on ? 1 : 0);
}

void led_write(bool on) {
    gpio_write(led_gpio, on ? 0 : 1);
}

void update_state();

void on_update(homekit_characteristic_t *ch, homekit_value_t value, void *context) {
    update_state();
}

void switch_on_callback(homekit_characteristic_t *_ch, homekit_value_t on, void *context);
void button_callback(uint8_t gpio, button_event_t event);

homekit_characteristic_t current_temperature = HOMEKIT_CHARACTERISTIC_(CURRENT_TEMPERATURE, 0);
homekit_characteristic_t target_temperature = HOMEKIT_CHARACTERISTIC_(TARGET_TEMPERATURE, 20, .callback = HOMEKIT_CHARACTERISTIC_CALLBACK(on_update));
homekit_characteristic_t units = HOMEKIT_CHARACTERISTIC_(TEMPERATURE_DISPLAY_UNITS, 0);
homekit_characteristic_t current_state = HOMEKIT_CHARACTERISTIC_(CURRENT_HEATING_COOLING_STATE, 0);
homekit_characteristic_t target_state = HOMEKIT_CHARACTERISTIC_(TARGET_HEATING_COOLING_STATE, 1, .callback = HOMEKIT_CHARACTERISTIC_CALLBACK(on_update));
homekit_characteristic_t cooling_threshold = HOMEKIT_CHARACTERISTIC_(COOLING_THRESHOLD_TEMPERATURE, 25, .callback = HOMEKIT_CHARACTERISTIC_CALLBACK(on_update));
homekit_characteristic_t heating_threshold = HOMEKIT_CHARACTERISTIC_(HEATING_THRESHOLD_TEMPERATURE, 15, .callback = HOMEKIT_CHARACTERISTIC_CALLBACK(on_update));
homekit_characteristic_t current_humidity = HOMEKIT_CHARACTERISTIC_(CURRENT_RELATIVE_HUMIDITY, 0);

[... 289 lines cut ...]
             macaddr[3], macaddr[4], macaddr[5]);
    
    name.value = HOMEKIT_STRING(name_value);
}

void user_init(void) {
    uart_set_baud(0, 115200);
   
    create_accessory_name();
  
    lora_init();
	
	xTaskCreate(lora_recive, "Recive", 256, NULL, 2, NULL);

    wifi_config_init("Home Bridge", NULL, on_wifi_ready);
    gpio_init();
	

    if (button_create(button_gpio, 0, 10000, button_callback)) {
        printf("Failed to initialize button\n");
    }
    
    
}
'

maximkulkin, 2020-01-10:
When you guys learn how to format code in Github comments?! It's not that hard

Kristian8606, 2020-01-10:
Sorry, I corrected it.

closed 2020-02-01 by Kristian8606