maximkulkin/esp-homekit-demo #297: esp32/led linker error

loicloic, 2020-01-11:
I've installed esp-idf and can build espressif examples with idf.py. However when I try to compile the esp32/led samples I get a linker error with an undefined reference in no-rrit/crt.o.

See below form the error and surrounding outputs:

``macmini:esp-homekit-demo$ make -C examples/esp32/led/ all``

``Toolchain path: /Users/user/Devel/Platform/esp/esp32/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc``
``Toolchain version: esp-2019r2``
``Compiler version: 8.2.0``
``Python requirements from /Users/user/Devel/Platform/esp/esp-idf/requirements.txt are satisfied.``

``LD build/bootloader/bootloader.elf
/Users/user/Devel/Platform/esp/esp32/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/user/Devel/Platform/esp/esp32/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/crt0.o:(.literal+0x0): undefined reference to `main'
collect2: error: ld returned 1 exit status``

``make[1]: *** [/Users/user/Devel/Platform/esp/esp-homekit-demo/examples/esp32/led/build/bootloader/bootloader.elf] Error 1
make: *** [/Users/user/Devel/Platform/esp/esp-homekit-demo/examples/esp32/led/build/bootloader/bootloader.bin] Error 2``

Any idea what I might be missing ?

maximkulkin, 2020-01-11:
`unset LDFLAGS`

loicloic, 2020-01-11:
Thanks @maximkulkin !

closed 2020-01-11 by loicloic