maximkulkin/esp-homekit-demo #299: Homekit Example sending infrared remote control?

marcelopalermo, 2020-01-13:
Hi,
First of all, thanks a lot for this homekit github. Simply fantastic!
Question: do you have any esp-homekit-demo example by using infrared remote control sending? I have already implemented it via Arduino Uno and infrared LED with proper infrared Arduino libraries. I'm facing difficulties to implement a similar solution under esp-homekit-demo. I don't know whether there is any Infrared library I can include.

Any sample code will be very helpful.

Thanks
Regards
Marcelo

maximkulkin, 2020-01-13:
This is my implementation of AC controller: https://github.com/maximkulkin/esp-ir-thermostat. I wrote my own IR library, which works pretty well for my purposes.

marcelopalermo, 2020-01-13:
Thanks so much. I will download it. Best regards!

closed 2020-01-13 by maximkulkin