maximkulkin/esp-homekit-demo #300: Bluetooth

Kristian8606, 2020-01-15:
Hi maximkulkin is it possible to add an example with bluetooth on esp32
for example, one of these

https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh

maximkulkin, 2020-01-15:
esp-homekit does not work over bluetooth

closed 2020-01-15 by maximkulkin

Kristian8606, 2020-01-16:
I had in mind whether it was possible to receive temperature data via bluetooth instead of using the DHT library.

Kristian8606, 2020-01-16:
https://www.youtube.com/watch?v=Wef_bWNC2rY

Kristian8606, 2020-01-16:
https://youtu.be/Wef_bWNC2rY

lrusnac, 2020-01-19:
@Kristian8606 try searching for a ble library for freeRTOS, but as far as I remember the esp32 had issues in using both the bluetooth and wifi at the same time, they share the same radio, so it's one at a time, this might create some annoyances but should be possible if you use some semaphores and keep it always on wifi mode and once in a while switch to bluetooth to get the temperature and go back to wifi.
it might be a bit of work to get it right, so good luck :) but share with us if you manage to do it