maximkulkin/esp-homekit-demo #301: Two flashed ESP8266 in conflict within same Wi-Fi network?

marcelopalermo, 2020-01-19:
Hi Colleagues,

When I have two flashed ESP8266 within the same WIFi network, HomeKit enters in conflict when adding second one. Due to my house physical reasons, I can't add all the accessories I need under one ESP8226, only.

Details below:

I've got two ESP8266 NodeMCU chips flashed with esp-homekit-demo under the same WiFi network.

- Apple HomeKit detects ESP8266 #1 successfully and registers "Luz da Sala" as the hub, and adds two accessories correctly.
- When plugging ESP8266 #2 into same WiFi, then it can't be found by Apple HomeKit and the following "No Response" (in red Portuguese "Sem Resposta") appears for all ESP8266 #1 accessories 1 or 2 seconds later:

[image]

It sounds like the two ESP8266 have the very same identity, from Apple HomeKit standpoint. I believe that if HomeKit understands the two ESP8266 chips as different accessories/identities then Apple HomeKit would understand then as independent and would allow adding both ESP8266 #1 and #2. Is this assumption correct?

ESP8266 #1 contains the following code -- before that please notice the following:
- .id=1 has primary=true and .id=2 has primary=false
-  name and manufacturer are different for each of them

````C
homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Luz da Sala"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Marcelo1"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "030A2BABF19D"),
            HOMEKIT_CHARACTERISTIC(MODEL, "MyLight"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, led_identify),
            NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Luz da Sala"),
            HOMEKIT_CHARACTERISTIC(
                ON, false,
                .getter=led_on_get,
                .setter=led_on_set
            ),
            NULL
        }),
        NULL
    }),
    
    HOMEKIT_ACCESSORY(.id=2, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Motor de Passo"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Marcelo2"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "030A2BABF19E"),
            HOMEKIT_CHARACTERISTIC(MODEL, "MyStepMotor"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, led3_identify),
            NULL
        }),
        HOMEKIT_SERVICE(SWITCH, .primary=false, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Motor de Passo"),
            HOMEKIT_CHARACTERISTIC(
                ON, false,
                .getter=led3_on_get,
                .setter=led3_on_set
            ),
            NULL
        }),
        NULL
    }),
    
    NULL
};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "222-22-222"
};

void user_init(void) {
   // uart_set_baud(0, 115200);
    uart_set_baud(0, 9600);

    wifi_init();
    led_init();
    homekit_server_init(&config);
}
````

ESP98266 #2 has the following code (below at end of this thread), with one accessory only, and .id=3 and primary=false:

- getter and setter have the same method names but it is a different program
- my understanding of .id=3 is due to all is under the same WiFi and hub ("Luz da Sala") is already installed via ESP8266 #1 (above explained).

I've also tested ESP8266 #2 with .id=1 and .primary=true but issue persists. I've tried other sequences like below (all of them leading to the same issue again):
- plugging ESP8266 #1 and configure it in HomeKit and then plugging in ESP8266 #2 
- plugging both ESP8266 #1 and #2 at the same time over the same WiFi and then configure ESP8266 #1 

I'm pretty sure I am missing something on the process and/or architecture. Any help/hints would be much appreciated.

Thanks in advance
Regards
Marcelo 

````C
homekit_accessory_t *accessories[] = {

    
    HOMEKIT_ACCESSORY(.id=3, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Motor"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Marcelo3"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "022A2BABF19A"),
            HOMEKIT_CHARACTERISTIC(MODEL, "Motor1"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, led3_identify),
            NULL
        }),
        HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Motor"),
            HOMEKIT_CHARACTERISTIC(
                ON, false,
                .getter=led3_on_get,
                .setter=led3_on_set
            ),
            NULL
        }),
        NULL
    }),
    
    NULL
};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "222-22-222"
};

void user_init(void) {
   // uart_set_baud(0, 115200);
    uart_set_baud(0, 9600);

    wifi_init();
    led_init();
    homekit_server_init(&config);
}
````

peros550, 2020-01-19:
I believe id=3 is not right. The Id numbers are independent between devices. Therefore, in your second accessory try id=1 and set it as primary since it’s the only accessory for this device.

marcelopalermo, 2020-01-21:
Thanks for your response @peros550 . I believe that I have already tried this combination but I will redo it and will post results here.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

marcelopalermo, 2020-01-23:
I've tested with id=1 and primary and confirmed the issue persists. Thanks 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

maximkulkin, 2020-01-23:
First, I recommend using a single accessory definition and just grouping all services inside of it. Second, stop talking about `.primary=true`, it has nothing to do with it.

maximkulkin, 2020-01-23:
While having both of them connected, can you ping each of them? What does `dns-sd -B _hap._tcp local` (command is for macOS) say?

closed 2020-02-13 by maximkulkin