maximkulkin/esp-homekit-demo #302: Unable to build example for ESP32 - component-http-parser-build

jwktje, 2020-01-23:
After following the guide on the wiki I attempted to build the example for the ESP32.
`make -C examples/esp32/led all`

This fails with the following error;
`make: *** No rule to make target `component-http-parser-build', needed by `component-homekit-build'.  Stop.`

I'm on OSX 10.15.2 (19C57) and trying to flash a Wemos Lolin32. I really hope someone can tell me how to fix this error.

jwktje, 2020-01-23:
Tried reinstalling ESP-IDF and the error changed; 

```
jwktje@Jwktje esp-homekit-demo % make -C examples/esp32/led all         
Toolchain path: /Users/jwktje/Development/Esp-idf/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/jwktje/Development/Esp-idf/esp/esp-idf/requirements.txt are satisfied.
make[2]: *** No rule to make target `/Users/jwktje/Development/Esp-idf/esp-idf/components/log/log.c', needed by `log.o'.  Stop.
make[1]: *** [component-log-build] Error 2
make: *** [/Users/jwktje/Development/Esp-idf/esp-homekit-demo/examples/esp32/led/build/bootloader/bootloader.bin] Error 2
```

maximkulkin, 2020-01-23:
You need to
1. check out git submodules (`git submodule update --init --recursive`)
2. make sure you clean project (`make -C examples/your_example clean`) if you've built it before updating

jwktje, 2020-01-23:
I did run `git submodule update --init --recursive` but weirdly it didn't show any output after this command. This confused me so that's why I tried reinstalling everything.

This didn't change it for the better. For now I'm attempting to run ESP-IDF in a Ubuntu virtualbox on my mac just so I know for sure the environment is clean. 

This results in the same error though. It builds for a while and then it gets stuck on: 
```
make: *** No rule to make target 'component-http-parser-build', needed by 'component-homekit-build'.  Stop.
```

Are you able to build the examples/esp32/led example?

maximkulkin, 2020-01-23:
Oh, that one was a broken link to component directory. Pushed a fix.

jwktje, 2020-01-23:
Yes! You are the best. That was the problem. I am now able to build without issue without VirtualBox too. Thanks for clearing this up!

closed 2020-01-23 by jwktje